platformio/platform-ststm32 #633: New Version: asio.hpp: No such file or directory

Sarah26-10, 2022-06-17:
Hello there,

today i tried to work on my Portenta project, which compiled and worked some weeks ago. Until now i had to reinstall PIO.
When i now build the project the following error occurs in every portenta projects and configurations:
**fatal error: asio.hpp: No such file or directory**

I found the solution in degrading the platform version to: 

> platform = ststm32@15.3.0

So my question: Is there an error in the new Version with this **asio.hpp**?

valeros, 2022-07-05:
Hi @Sarah26-10 ! Seems like you have an old broken package, try to force PlatformIO redownload the framework package by deleting the folder `c:\Users\YOUR_USER\.platformio\packages\framework-arduino-mbed` (`/home/YOUR_USER/.platformio/packages/framework-arduino-mbed` on Linux) and compile your project again.

Sarah26-10, 2022-07-05:
No, problem still occurs.

valeros, 2022-07-13:
@Sarah26-10 Could you please put together a simple example to reproduce the issue?

valeros, 2023-06-14:
Closing this issue because it needs more information and has not had recent activity.

closed 2023-06-14 by valeros