platformio/platform-ststm32 #641: ST-LINKv2 / STM32L0 / Unable to connect to the target

JeroenIoT, 2022-07-28:
* Platform: Platformio
* Processor: STM32L082
* Upload protocol: stlink
* Uploader: Original ST-Linkv2 (5 wires connected)

- Problem: Not able to connect and upload bin file to stm32 when using platformio.  
-- When using STM32CubeProgrammer the connect and upload works without a problem.

See also: https://community.platformio.org/t/dual-bank-stm32l082/26059/18

Below the "Verbose Upload" output:

```
AVAILABLE: mbed, stlink
CURRENT: upload_protocol = stlink
openocd -d2 -s C:\Users\jeroen\.platformio\packages\tool-openocd/scripts -f board/JENG_STM32L082KZ.cfg -c "program {.pio\build\STM32L082_bank1\firmware.elf}  verify reset; shutdown;"
Info : STLINK V2J34S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 1.762846
Error: init mode failed (unable to connect to the target)
in procedure 'program'
** OpenOCD init failed **
shutdown command invoked

*** [upload] Error 1
================================================================ [FAILED] Took 18.31 seconds ================================================================

Environment      Status    Duration
---------------  --------  ------------
STM32L082_bank1  FAILED    00:00:18.306
STM32L082_bank2  IGNORED
=========================================================== 1 failed, 0 succeeded in 00:00:18.306 ===========================================================

```

JeroenIoT, 2022-07-29:
UPDATE: The connect issued was solved using suggestion in following thread:

[STM32 + VS Code + mbed upload issue: Error: libusb_open() failed with LIBUSB_ERROR_ACCESS Error: open failed - #12 by maxgerhardt](https://community.platformio.org/t/stm32-vs-code-mbed-upload-issue-error-libusb-open-failed-with-libusb-error-access-error-open-failed/10650/12)?

Following reset config solved the connect for me

`reset_config srst_only srst_nogate connect_assert_srst`

Open is that after programming the MCU the reset is not completed.
Most likey an openOCD related config issue. Will update this issue once I found the solution.

```
AVAILABLE: mbed, stlink
CURRENT: upload_protocol = stlink
openocd -d2 -s C:\Users\jeroen\.platformio\packages\tool-openocd/scripts -f board/JENG_STM32L082KZ.cfg -c "program {.pio\build\STM32L082_bank1\firmware.elf}  verify reset; shutdown;"
xPack OpenOCD x86_64 Open On-Chip Debugger 0.11.0+dev (2021-10-16-21:19)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
srst_only separate srst_nogate srst_open_drain connect_assert_srst

Info : clock speed 300 kHz
Info : STLINK V2J34S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 1.764427
Info : STM32L082KZTx.cpu: Cortex-M0+ r0p1 processor detected
Info : STM32L082KZTx.cpu: target has 4 breakpoints, 2 watchpoints
Info : starting gdb server for STM32L082KZTx.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x08002da8 msp: 0x20005000
STM32L0: Enabling HSI16
Info : Unable to match requested speed 2500 kHz, using 1800 kHz
Info : Unable to match requested speed 2500 kHz, using 1800 kHz
** Programming Started **
Info : Device: STM32L0xx (Cat.5)
Info : STM32L flash has dual banks. Bank (0) size is 96kb, base address is 0x8000000
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
in procedure 'program'
*** [upload] Error 1
```

JeroenIoT, 2022-08-10:
Update: Found out that the upload does work BUT the reset in the openOCD does not work.

_C:\Users\jeroen\.platformio\packages\tool-openocd\bin\openocd -d2 -s C:\Users\jeroen\.platformio\packages\tool-openocd/scripts -f board/JENG_STM32L082KZ.cfg -c "program {.pio\build\STM32L082_bank1\firmware.elf}  verify; **reset**; shutdown;"_

When removing the _reset_ in the above command there are no errors. Of course in this case I have to reset the MCU manually.
So I am still looking for a way to solve this. However afaik this is not related to platformio.

Therefore I will close this issue in the coming weeks (in case I find a solution beforehand I will share the info here)

ethyljoybadiang, 2024-01-08:
Does this also work for MAX7800x?