platformio/platform-ststm32 #643: Update tool-jlink

maxgerhardt, 2022-08-11:
Per https://community.platformio.org/t/please-update-jlink-to-v770a/29184. 

V7.70a is released at the 11.08.2022 (https://www.segger.com/downloads/jlink/), PIO still is at v7.50 (https://registry.platformio.org/tools/platformio/tool-jlink/versions).

maxgerhardt, 2024-10-03:
Push! Segger has release 8.10a of the software. ([community post](https://community.platformio.org/t/can-the-segger-j-link-software-be-updatet/43588)).

McKayRansom, 2024-12-06:
I also need the PlatformIO JLink package updated for some of the new device support that has been added since v7.5.

functionpointer, 2026-01-01:
JLink is now on V8.96. Can it be updated in platformio?

functionpointer, 2026-01-04:
So apparently you can cheat it by replacing all the files in platformio's package. That's in `~/.platformio/packages/tool-jlink`. Just make sure to leave the package.json in place.