platformio/platform-ststm32 #645: STM32_F4VE - dfu-util: No DFU capable USB device available (additional hwid for this board)

alandpearson, 2022-08-22:
Hello,

I have a STM32_F4VE dev board and platformIO refuses to flash using DFU giving the dreaded 

 dfu-util: No DFU capable USB device available

The issue in this case turned out to be this board has the following HWIDS:
   "0x0483",  "0xdf11"
   
   Could these IDs be added to the genericSTM32F407VET6.json   board description please ?

douardda, 2023-04-22:
using the following `platformio.ini` entry :

```
[env:black_f407ve]
platform = ststm32
framework = stm32cube
board = black_f407ve
build_flags = -DF4
upload_protocol = dfu
```

in the example `stm32cube-hal-blink` works just fine here (on a chinese clone of the F4VE board); just had to change the definition of LED in the main.c to match the board.

```
stm32cube-hal-blink  pio run -e black_f407ve -t upload 
Processing black_f407ve (platform: ststm32; framework: stm32cube; board: black_f407ve)
------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/black_f407ve.html
PLATFORM: ST STM32 (15.6.0) > Black STM32F407VE
HARDWARE: STM32F407VET6 168MHz, 128KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-stm32cubef4 @ 1.26.2 
 - tool-dfuutil @ 1.11.0 
 - tool-ldscripts-ststm32 @ 0.2.0 
 - tool-openocd @ 2.1100.211028 (11.0) 
 - tool-stm32duino @ 1.0.1 
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 52 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/black_f407ve/src/main.o
Linking .pio/build/black_f407ve/firmware.elf
Checking size .pio/build/black_f407ve/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.0% (used 44 bytes from 131072 bytes)
Flash: [          ]   0.2% (used 1232 bytes from 524288 bytes)
Building .pio/build/black_f407ve/firmware.bin
Adding dfu suffix to firmware.bin
dfu-suffix (dfu-util) 0.11

Copyright 2011-2012 Stefan Schmidt, 2013-2020 Tormod Volden
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Suffix successfully added to file
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, dfu, jlink, serial, stlink
CURRENT: upload_protocol = dfu
Uploading .pio/build/black_f407ve/firmware.bin
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Opening DFU capable USB device...
Device ID 0483:df11
Device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(10) = dfuERROR, status(10) = Device's firmware is corrupt. It cannot return to run-time (non-DFU) operations
Clearing status
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading element to address = 0x08000000, size = 1632

Erase   	[                         ]   0%            0 bytes
Erase   	[                         ]   0%            0 bytes
Erase   	[=========================] 100%         1632 bytes
Erase    done.

Download	[                         ]   0%            0 bytes
Download	[=========================] 100%         1632 bytes
Download done.
File downloaded successfully
Submitting leave request...
Transitioning to dfuMANIFEST state
================================================ [SUCCESS] Took 1.88 seconds ================================================

Environment    Status    Duration
-------------  --------  ------------
black_f407ve   SUCCESS   00:00:01.875
================================================ 1 succeeded in 00:00:01.875 ================================================
 
```

Using platformio 6.1.6

valeros, 2023-06-14:
Only the first pair of HWIDs is used generating DFU image. The default HWID values can be overridden via an extra script as described here https://docs.platformio.org/en/latest/scripting/examples/override_board_configuration.html

closed 2023-06-14 by valeros