platformio/platform-ststm32 #655: Bug with IntelliSense and paths

RasmusAPedersen, 2022-11-03:
Reported here on Github as per request from the forums
https://community.platformio.org/t/bug-with-intellisense-and-paths/30269

------------------------------------------------------------------

### Configuration

**Operating system**: Ubuntu 22.04.1 LTS x86_64, kernel 5.15.0-52-generic 

**PlatformIO Version** (`platformio --version`): 
[image]

### Description of problem
I have been getting red squiggly lines from IntelliSense in VS Code on my includes (even of Arduino.h) with PIO on Ubuntu 22, but I have also found the solution.

```cpp
cannot open source file "Arduino.h"C/C++(1696)
```

This is the exact error I am getting:

```json
Failed to parse 
"/home/ ... /.vscode/c_cpp_properties.json": Unexpected token ( in JSON at position 85
```

When I go and have a look in c_cpp_properties.json it is very clear that the error is reported correctly.
[image]

By removing the backslashes, the error is resolved, and the red squiggly lines go away. However, c_cpp_properties.json is autogenerated, and the file explicitly advises against manual edits, so it is eventually just re-generated, and the error returns.

What can I do to fix this permanently?

Let me know if I need to supply more information.

ivankravets, 2022-11-03:
@fpistm , how is it difficult to remove SHELL reserved symbols from the folder names? See https://github.com/stm32duino/Arduino_Core_STM32/tree/main/variants/STM32G4xx

The `",(, etc.` are reserved symbols and the shell treats them as a code. In this case, we escape Arduino STM32 symbols and later other integration components (such as VSCode) fail. We could potentially add some workarounds to the PIO Core to fix specific STM32 issues but it is better to fix the root of problem.

fpistm, 2022-11-03:
@ivankravets 
I will not remove it as it is supported well by intellisense. It seems you add "\" before eaupported well by intellisense. It seems you add "\" before each "()" so do not add them.
I use daily VSCode on Linux and Windows with many plugin and have met no issue even wch "()" so do not add them.
I use daily VSCode on Linux and Windows with many plugin and have met no issue even with cli tools.

ivankravets, 2022-11-03:
@valeros did we add escaping on the dev-platform side? If we remove it, does it break SCons?

> I will not remove

@fpistm, thanks. Please don't remove it 👍

runger1101001, 2022-11-03:
I can confirm that the same problem occurs on Mac.

It's a "real" problem because the STM32 framework for Arduino actually has a bunch of files with "(" in their names:
.platformio/packages/framework-arduinoststm32/variants/STM32G4xx/G473R(B-C-E)T_G474R(B-C-E)T_G483RET_G484RET

So the moment you try to create a project based on STM32 MCUs, IntelliSense fails and you get spurious errors highlighted in your code's includes...

```
[env:nucleo_g474re]
platform = ststm32
board = nucleo_g474re
framework = arduino

lib_archive = false
```

PlatformIO: 2.5.5 (2022-10-31)

VSCode: 
  Version: 1.73.0
  Commit: 8fa188b2b301d36553cbc9ce1b0a146ccb93351f
  Date: 2022-11-01T15:38:50.881Z (2 days ago)
  Electron: 19.0.17
  Chromium: 102.0.5005.167
  Node.js: 16.14.2
  V8: 10.2.154.15-electron.0
  OS: Darwin x64 21.4.0
  Sandboxed: No

IntelliCode v1.2.29

valeros, 2022-11-05:
Hi @RasmusAPedersen ! Thanks for reporting. I've pushed an updated package with a fix, please pull the changes via `pio platform update ststm32`.

closed 2022-11-05 by valeros

RasmusAPedersen, 2022-11-05:
Yes it seems to be fixed now. Thank you!

4ydan, 2023-07-13:
I have this problem on a freshly installed debian bookworm

```
/.vscode/c_cpp_properties.json": Bad escaped character in JSON at position 82
```

```
$ platformio --version
PlatformIO Core, version 6.1.9
```

I tried `pio platform update ststm32`

```
adi at pc in ~/NucleoF411RE on main [!?]
$ pio platform update stm32

WARNING: This command is deprecated and will be removed in the next releases. 
Please use `pio pkg update` instead.

UnknownPackageError: Could not find the package with 'stm32' requirements for your system 'linux_x86_64'

adi at pc in ~/NucleoF411RE on main [!?]
$ pio pkg update
Resolving Nucleo_F411RE_USB_RTOS dependencies...
Already up-to-date.
Resolving test dependencies...
Already up-to-date.
```

I do not have any other c++ extensions beside the microsoft extension pack

ivankravets, 2023-07-13:
Please remove `/home/adi/.platformio/packages` and `/home/adi/.platformio/platforms` folders. It looks like you have custom-broken packages.

4ydan, 2023-07-13:
Thanks for the fast reply

I removed the folders, deleted the json file and pressed reload window in vscode. The file got generated again with the same error

```
"/home/adi/.platformio/packages/framework-arduinoststm32/variants/STM32F4xx/F411R\(C-E\)T",
```

Maybe it has something to do that I worked before on windows?

ivankravets, 2023-07-13:
Please navigate to the project folder and type `pio pkg update`.

4ydan, 2023-07-13:
It only happens with the first env, the second one is fine

```
[env:Nucleo_F411RE_USB_RTOS]
platform = ststm32
board = nucleo_f411re
framework = arduino
monitor_port = COM[3]
build_flags = 
	-Iinclude
	-std=gnu++11
	-DSERIAL_RX_BUFFER_SIZE=256
	-DSERIAL_TX_BUFFER_SIZE=256
	-DENABLE_RTOS
	-DUSE_DISPLAY
	-DDEBUG
	; -DSECOND_DISPLAY_PAGE
build_type = debug ;https://docs.platformio.org/en/latest/projectconf/build_configurations.html
board_build.core = stm32 
test_build_src = true
lib_deps = STM32duino FreeRTOS
upload_protocol = stlink

[env:test]
platform = ststm32
board = nucleo_f411re
framework = arduino
monitor_port = COM[7]
build_flags = 
	-std=gnu++11
	-DSERIAL_RX_BUFFER_SIZE=256
	-DSERIAL_TX_BUFFER_SIZE=256
test_ignore =
	test_blink
	test_lfs
	; test_eeprom_reboot
	; test_filter
	; test_modbus
	; test_temp
test_build_src = true
build_type = test
upload_protocol = stlink
```

ivankravets, 2023-07-13:
Try to remove `.vscode` folder from the project and restart VSCode.

4ydan, 2023-07-13:
Still persists.

valeros, 2023-07-13:
Hi @4ydan, could you please run `pio pkg update -p ststm32` in the IDE terminal, it should install the latest Arduino package that hopefully will fix the issue.

4ydan, 2023-07-13:
Hello @valeros 
It fixed the issue. Thank you both for the fast help 🚀 . Have a nice day.

michael-neros, 2025-07-21:
Hi, sorry to open this back up but I'm having the same issue and it's not being resolved with any of the fixes listed in this thread.

```shell
Ubuntu 22.04

pio --version
PlatformIO Core, version 6.1.18
```

I performed the following commands in and out of the pio terminal on vscode with no luck
```
pio pkg updateResolving DIY_900_TX_STM32_SX1272_via_STLINK dependencies...
Already up-to-date.
Resolving DIY_900_RX_BluePill_via_STLINK dependencies...
Already up-to-date.
... etc...
```

```
pio pkg update -p stm32
Resolving native dependencies...
UnknownPackageError: Could not find the package with 'None' requirements for your system 'linux_x86_64'
```

The only thing that fixes it is removing the `\` from the error string in c_cpp_properties.json

michael-neros, 2025-07-22:
@valeros yea sure,
```shell
pio pkg list -g --only-platforms
Platforms
├── native @ 1.2.1 (required: platformio/native)
└── ststm32 @ 15.1.0 (required: platformio/ststm32)
```

valeros, 2025-07-22:
@michael-neros v15.1.0 is three years old, you need to update your platform to the latest.

michael-neros, 2025-07-22:
@valeros how would I go about doing that?

valeros, 2025-07-22:
@michael-neros can you share here your `platformio.ini` file ?

michael-neros, 2025-07-22:
@valeros platform.ini

```ini
; ExpressLRS PlatformIO Project Configuration File

[platformio]
extra_configs =
	# keep common first
	targets/common.ini
	# defined one by one to maintain the order
	targets/unified.ini
	targets/diy_900.ini
	targets/diy_2400.ini
	targets/frsky.ini
	targets/happymodel_900.ini
	targets/happymodel_2400.ini
	targets/imrc.ini
	targets/namimnorc_900.ini
	targets/namimnorc_2400.ini
	targets/siyi.ini

# ------------------------- TARGET ENV DEFINITIONS -----------------

[env:native]
platform = native
framework =
test_ignore = test_embedded
lib_ignore = BUTTON, DAC, LQCALC, LBT, SPIEx, PWM, WIFI, TCPSOCKET, LR1121Driver
build_src_filter = ${common_env_data.build_src_filter} -<ESP32*.*> -<STM32*.*> -<ESP8*.*> -<tx_*.cpp> -<rx_*.cpp> -<common.*> -<config.*>
build_flags =
	-std=c++11
	-Iinclude
	-D PROGMEM=""
	-D UNIT_TEST=1
	-D TARGET_NATIVE
	-D CRSF_RX_MODULE
	-D CRSF_TX_MODULE

```

valeros, 2025-08-04:
@michael-neros The behavior is fixed in the latest version of the [build script](https://github.com/stm32duino/Arduino_Core_STM32/blob/main/tools/platformio/platformio-build.py), try adapting it manually for your package.