platformio/platform-ststm32 #674: Applications uploaded to the LoRa-E5 mini does not boot

larswd, 2023-01-24:
@jerabaul29 and I have been trying to do some work using the Seeed studio board LoRa-E5 mini using STM32WLE5JC.
After struggling a bit with removing the included write protection using the STM Programmer and a ST-Link, we managed to upload a script to the board. While we can, indeed see that the script has been uploaded, see picture attached below, the board seems to be entirely unresponsive. We can neither detect any activity on the connected serial ports, nor does the builtin LED flash (Code included below). We have also tried this on two boards, making it more unlikely to be caused by faulty hardware. The only response we have been able to get is that the RX LED flashes briefly if we send messages to it using the Arduino IDE (v2) serial monitor.
 Has anyone encountered similar problems to ours? 

[image]

```c++
#include <Arduino.h>

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  delay(100);
  Serial.println(F("booted"));

  pinMode(LED_BUILTIN, OUTPUT); //We have tried using the exact pin name as well
  digitalWrite(LED_BUILTIN, HIGH);
}

void loop() {
  delay(2000);
  Serial.println(F("loop"));
  // put your main code here, to run repeatedly:
}
```

jerabaul29, 2023-01-24:
(I cannot see the screenshot of the memory explorer showing that the flash contains the strings)

jerabaul29, 2023-01-24:
I wonder if this means somehow that the board does not boot; is it possible that there are some specific "things" to set up to make sure that the bootloader is correctly configured / the user code gets correctly started?

larswd, 2023-01-24:
> (I cannot see the screenshot of the memory explorer showing that the flash contains the strings)

I forgot to add it! Should be there now!

jerabaul29, 2023-01-24:
Thanks for the udpate :) . A small note: we feel confident that the firmware has been uploaded because we can find in the firmware the strings ```booted``` and ```loop``` that are used in the ```Serial.print``` commands (though this is not shown at the moment when I write this in the exact screenshot included above).

larswd, 2023-03-31:
I managed to make it work. A guide with relevant programs can be found here: 

https://github.com/larswd/STM32WL-Seeed-LoRa-E5-mini-Playground