platformio/platform-ststm32 #683: Updated version of tool-openocd package

orvalman, 2023-03-08:
Hello,

We are developing a custom framework for our line of evaluation kits.
We have run into two separate issues that could be solved by updating the tool-openocd package to xPack OpenOCD v0.11.0-5.
1. https://community.platformio.org/t/nucleo-u575-upload-debug-support/32084/3
2. Issue running OpenOCD on Mac Mini with M2 chip (Bad CPU type in executable)

Would it be possible to get the tool-openocd package updated in the platformIO registry?
Thank you!

tookitogo, 2023-07-01:
@valeros 
Surely this should be labeled "package update", since this isn't a request for a new feature?!? The old OpenOCD version is causing problems with existing features (like, you know, being able to upload a sketch) to already-supported boards. 

As it is right now, the PlatformIO board registry lists the Nucleo U575 board as supported, but with the outdated OpenOCD, it is not usable whatsoever.

valeros, 2023-07-03:
[v16.0](https://github.com/platformio/platform-ststm32/releases/tag/v16.0.0) of this platform uses the latest OpenOCD v0.12

closed 2023-07-03 by valeros