platformio/platform-ststm32 #690: Code Upload successful, but not being executed

joelsa, 2023-03-29:
I have troubles trying to turn on a simple led and I think this might be related to a bug in platformio.

I can turn the led on using several other framworks (such as stm32-rs, modm-io or using the ST-HAL without platformio), so the wiring is correct, event though this is a custom board.

General information:
Host OS: Ubuntu 22.04.1 with Kernel 5.19.0-35
PlatformIO Version: PlatformIO Core, version 6.1.7a4
st-stm32 version: 15.6.0
Microcontroller: STM32F446RE
Upload interface: J-LINK V11
OpenOCD version: 0.11.0
stm32duino version: 1.0.1
VSCode version: 1.76.2

Here is my main.cpp:

<details>
  <summary>main.cpp</summary>
  
  ```cpp
  #include <Arduino.h>

void setup() {
  // put your setup code here, to run once:
  pinMode(PB0, OUTPUT); // LED_RED
  pinMode(PB1, OUTPUT); // LED_GREEN
}

void loop() {
  // put your main code here, to run repeatedly:
  digitalWrite(PB0, LOW); // Active LOW
  digitalWrite(PB1, LOW); // Active LOW
  delay(1000);
}
  ```
  
</details>

Here is my platformio.ini:

<details>
  <summary>platformio.ini</summary>
  
  ```ini
[env:genericSTM32F446RE]
platform = ststm32
board = genericSTM32F446RE
framework = arduino
debug_tool = jlink

; SWD interface
upload_protocol = jlink
  ```
  
</details>

Here is the `pio run -t upload --verbose` output:

<details>
  <summary>verbose output</summary>

```text
Processing genericSTM32F446RE (platform: ststm32; board: genericSTM32F446RE; framework: arduino; debug_tool: jlink; board_upload.maximum_ram_size: 98304; upload_protocol: jlink; upload_resetmethod: none separate)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F446RE.html
PLATFORM: ST STM32 (15.6.0) > STM32F446RE (128k RAM. 512k Flash)
HARDWARE: STM32F446RET6 180MHz, 96KB RAM, 512KB Flash
DEBUG: Current (jlink) External (blackmagic, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 @ 4.20400.0 (2.4.0) 
 - framework-cmsis @ 2.50700.210515 (5.7.0) 
 - tool-dfuutil @ 1.11.0 
 - tool-jlink @ 1.77001.0 (7.70.1) 
 - tool-openocd @ 2.1100.211028 (11.0) 
 - tool-stm32duino @ 1.0.1 
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 13 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
MethodWrapper(["checkprogsize"], [".pio/build/genericSTM32F446RE/firmware.elf"])
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.9% (used 920 bytes from 98304 bytes)
Flash: [          ]   2.7% (used 14324 bytes from 524288 bytes)
.pio/build/genericSTM32F446RE/firmware.elf  :
section              size        addr
.isr_vector           452   134217728
.text               12712   134218180
.rodata              1456   134230892
.ARM.extab              0   134232348
.ARM                    8   134232348
.preinit_array          0   134232356
.init_array            16   134232356
.fini_array             8   134232372
.data                 156   536870912
.bss                  764   536871068
.noinit                 0   536871832
._user_heap_stack    1536   536871832
.ARM.attributes        48           0
.comment              102           0
.debug_frame         1360           0
Total               18618
<lambda>(["upload"], [".pio/build/genericSTM32F446RE/firmware.bin"])
AVAILABLE: blackmagic, dfu, jlink, serial, stlink
CURRENT: upload_protocol = jlink
JLinkExe -device STM32F446RE -speed 4000 -if swd -autoconnect 1 -NoGui 1 -CommanderScript "/home/jsa/Documents/PlatformIO/Projects/tof-sensor-code--am-pio/.pio/build/genericSTM32F446RE/upload.jlink"
SEGGER J-Link Commander V7.70a (Compiled Aug 10 2022 16:32:44)
DLL version V7.70a, compiled Aug 10 2022 16:32:29

J-Link Command File read successfully.
Processing script file...
J-Link>h
J-Link connection not established yet but required for command.
Connecting to J-Link via USB...O.K.
Firmware: J-Link V11 compiled Mar 15 2023 11:20:48
Hardware version: V11.00
J-Link uptime (since boot): 0d 00h 00m 11s
S/N: 261013303
License(s): FlashBP, GDB
OEM: SEGGER-EDU
USB speed mode: High speed (480 MBit/s)
VTref=4.895V
Target connection not established yet but required for command.
Device "STM32F446RE" selected.

Connecting to target via SWD
InitTarget() start
InitTarget() end
Found SW-DP with ID 0x2BA01477
DPIDR: 0x2BA01477
CoreSight SoC-400 or earlier
Scanning AP map to find all available APs
AP[1]: Stopped AP scan as end of AP map has been reached
AP[0]: AHB-AP (IDR: 0x24770011)
Iterating through AP map to find AHB-AP to use
AP[0]: Core found
AP[0]: AHB-AP ROM base: 0xE00FF000
CPUID register: 0x410FC241. Implementer code: 0x41 (ARM)
Found Cortex-M4 r0p1, Little endian.
FPUnit: 6 code (BP) slots and 2 literal slots
CoreSight components:
ROMTbl[0] @ E00FF000
[0][0]: E000E000 CID B105E00D PID 000BB00C SCS-M7
[0][1]: E0001000 CID B105E00D PID 003BB002 DWT
[0][2]: E0002000 CID B105E00D PID 002BB003 FPB
[0][3]: E0000000 CID B105E00D PID 003BB001 ITM
[0][4]: E0040000 CID B105900D PID 000BB9A1 TPIU
[0][5]: E0041000 CID B105900D PID 000BB925 ETM
Cortex-M4 identified.
PC = 10001626, CycleCnt = 80504B2A
R0 = 00300200, R1 = 00000000, R2 = BF00E78A, R3 = 00000000
R4 = 00300200, R5 = 00000005, R6 = 00000000, R7 = 00000000
R8 = 00001388, R9 = 00000000, R10= 00000000, R11= 00000000
R12= 00000000
SP(R13)= 2001FEF0, MSP= 2001FEF0, PSP= 00000000, R14(LR) = 1000002F
XPSR = 41000003: APSR = nZcvq, EPSR = 01000000, IPSR = 003 (HardFault)
CFBP = 00000000, CONTROL = 00, FAULTMASK = 00, BASEPRI = 00, PRIMASK = 00

FPS0 = 00000000, FPS1 = 00000000, FPS2 = 00000000, FPS3 = 00000000
FPS4 = 00000000, FPS5 = 00000000, FPS6 = 00000000, FPS7 = 00000000
FPS8 = 00000000, FPS9 = 00000000, FPS10= 00000000, FPS11= 00000000
FPS12= 00000000, FPS13= 00000000, FPS14= 00000000, FPS15= 00000000
FPS16= 00000000, FPS17= 00000000, FPS18= 00000000, FPS19= 00000000
FPS20= 00000000, FPS21= 00000000, FPS22= 00000000, FPS23= 00000000
FPS24= 00000000, FPS25= 00000000, FPS26= 00000000, FPS27= 00000000
FPS28= 00000000, FPS29= 00000000, FPS30= 00000000, FPS31= 00000000
FPSCR= 00000000
J-Link>loadbin .pio/build/genericSTM32F446RE/firmware.bin, 0x08000000
'loadbin': Performing implicit reset & halt of MCU.
Reset: Halt core after reset via DEMCR.VC_CORERESET.
Reset: Reset device via AIRCR.SYSRESETREQ.
Downloading file [.pio/build/genericSTM32F446RE/firmware.bin]...
Comparing flash   [100%] Done.
Erasing flash     [100%] Done.
Programming flash [100%] Done.
J-Link: Flash download: Bank 0 @ 0x08000000: 1 range affected (16384 bytes)
J-Link: Flash download: Total: 0.436s (Prepare: 0.026s, Compare: 0.001s, Erase: 0.314s, Program & Verify: 0.080s, Restore: 0.014s)
J-Link: Flash download: Program & Verify speed: 199 KB/s
O.K.
J-Link>r
Reset delay: 0 ms
Reset type NORMAL: Resets core & peripherals via SYSRESETREQ & VECTRESET bit.
Reset: Halt core after reset via DEMCR.VC_CORERESET.
Reset: Reset device via AIRCR.SYSRESETREQ.
J-Link>q

Script processing completed.

OnDisconnectTarget() start
OnDisconnectTarget() end
=============================================================================== [SUCCESS] Took 3.38 seconds ===============================================================================
```
  
</details>

Compiling and uploading works well, but nothing happens.
I can also flash it with `openocd -f interface/jlink.cfg -c "transport select swd" -f target/stm32f4x.cfg -c "program .pio/build/genericSTM32F446RE/firmware.elf"`.

Is there an obvious problem or a hint where the problem may be?

valeros, 2023-03-30:
Hi @joelsa ! I don't see where any of the configured pins is toggled to HIGH `DigitalWrite(PB0, HIGH);`

joelsa, 2023-03-30:
> Hi @joelsa ! I don't see where any of the configured pins is toggled to HIGH `DigitalWrite(PB0, HIGH);`

Hi, the LEDs are active low, so turning the GPIO to LOW turns the LED on.

valeros, 2023-03-30:
I see you already have a configured debugger in place. Have you tried running a debug session to figure out where the board gets stuck?

joelsa, 2023-03-31:
Hi, looks like the code is getting stuck at line `tickstart = HAL_GetTick();` inside the function `HAL_RCCEx_PeriphCLKConfig` in file `stm32f4xx_hal_rcc_ex.c`, which is being called from `SystemClock_Config()`

joelsa, 2023-03-31:
I have added a Clock configuration like this:

<details>
  <summary>Clock configuration</summary>
```
extern "C" void SystemClock_Config(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct = {0};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {0};

  /** Configure the main internal regulator output voltage
  */
  __HAL_RCC_PWR_CLK_ENABLE();
  __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE3);

  /** Initializes the RCC Oscillators according to the specified parameters
  * in the RCC_OscInitTypeDef structure.
  */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSI;
  RCC_OscInitStruct.HSIState = RCC_HSI_ON;
  RCC_OscInitStruct.HSICalibrationValue = RCC_HSICALIBRATION_DEFAULT;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_NONE;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
  {
    Error_Handler();
  }

  /** Initializes the CPU, AHB and APB buses clocks
  */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK|RCC_CLOCKTYPE_SYSCLK
                              |RCC_CLOCKTYPE_PCLK1|RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_HSI;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV1;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV1;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_0) != HAL_OK)
  {
    Error_Handler();
  }
}
```
</details>

This works well. I still think this is a bug, because in my opinion the generic STM32F446 target should be configured to use the internal oscillator.

The error might be here: https://github.com/stm32duino/Arduino_Core_STM32/blob/17fd01f9ff569cc55de975da6a5742a8aab6d004/variants/Generic_F446Rx/variant.cpp

Should I close this issue and open an issue there, since this issue is obviously not with PlatformIO?

fpistm, 2023-03-31:
Submit an issue. I thought I've removed all those kind of clock config.  Do not hesitate to provide a PR.

joelsa, 2023-03-31:
Okay, will do!

closed 2023-03-31 by joelsa

joelsa, 2023-04-03:
I think my core is 2.4.0, for some reason I just opened an old commit in GitHub, so the link above is wrong. Sorry for the confusion.

I have made some progress debugging this further and it works on a Nucleo Board perfectly, it just doesn't work on my hardware.

I think the PLL might not be stable and/or the brownout detection is triggering.

I will probe some more with an oscilloscope, but I am 99.9% sure this isn't a software bug.

joelsa, 2023-04-05:
I have identified and solved the problem, Vcap was not connected and the PLL power source was unstable. Adding 10u solved all problems completely, and I can use the external HSE.

The problem with the initial clock config was simply, that I could not use the PLL at that frequencies but with Vcap (at 4u7 low ESR per Datasheet) in place, it is working perfectly.