platformio/platform-ststm32 #691: Add STM32CubeWL support

maxgerhardt, 2023-03-30:
STM32WLxx series chips should be supported by the `stm32cube` framework through a `framework-stm32cubewl` package. However, this doesn't exist yet and the board JSONs are also not marked to have support for it.

It can be manually added per https://community.platformio.org/t/how-to-use-stm32wl55jc-under-stm32cube-framework/32926, but we need official support.

Strooom, 2023-11-03:
That would be great as it simplifies porting projects from STM32CubeIDE to PlatformIO.

Strooom, 2023-11-26:
One more reason why I need this: after manually installing the package, I can do a local build, but my github action script builds fail, because there this package is unavailable...

Strooom, 2023-11-26:
Right, let me try that : another 'checkout' step, including the submodules and into the right path in the Platformio installation. Thanks for the hint.

Strooom, 2023-11-27:
Max, I've tried but did not yet get it working.
In order to share the knowhow with the community, I've made a forum topic : https://community.platformio.org/t/installing-stm32wl-package-for-github-actions/36950
Could you please take a look, maybe there's a quick fix.

tve, 2024-05-04:
Any way to make this happen? Also for framework = cmsis?

Strooom, 2024-05-04:
@tve I've got it working now for some time.
The package is not yet included by PlatformIO, but I've added it manually, as well as in the CI scripts.
I can proceed with this setup until it becomes part of the standard installation.

P

Strooom, 2024-05-04:
Take a look at testbuildrelease.yml in my repository mumo-v2-node-sw