platformio/platform-ststm32 #693: stm32 C0 Family - Missing Board

tgrotowski, 2023-04-05:
Hello,
I can't find nucleo board stm32 C031C6. https://www.st.com/en/evaluation-tools/nucleo-c031c6.html 

C0 it's quite new stm32 family with Cortex M0+ but becomes popular. There are more and more each day, it would be really useful to programming it on PlatformIO.

minumn, 2023-06-19:
I would like to use this board as well and together with zephyr. Can anyone help me with how I could implement this?

RudolphRiedel, 2023-06-28:
The Arduino core v2.5.0 added support for C031C6 back in end of march but the last update of the Arduino core was to v2.4.0 back in 15.5.0.
So at least for Arduino this is looking promising.

RudolphRiedel, 2023-07-21:
I made a board definition for the nucleo-c031c6:
[nucleo_c031c6.zip](https://github.com/platformio/platform-ststm32/files/12130689/nucleo_c031c6.zip)

It kind of works but not entirely as expected or I would have opened a pull-request for this.
First of all stlink upload does not work, I set it to mbed and that works.
And then I only used it to compile an Arduino test-project which works in general but for some reason this is awfully slow, at least with SPI, I measure huge pauses between SPI.transfer(byte) calls of almost 6µs.
Is this perhaps running on Arduino-mbed instead of Arduino-core-STM32?

I am using the development version of platform-ststm32 now but that also makes no difference.

RudolphRiedel, 2023-10-01:
I just tried this again in order to perhaps put in a pull request but now my board definition does not work anymore:
Resolving STM32C0C31 dependencies...
KeyError: Traceback (most recent call last):
  File "C:\Users\Ich\.platformio\penv\lib\site-packages\platformio\__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Users\Ich\.platformio\penv\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\Ich\.platformio\penv\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Users\Ich\.platformio\penv\lib\site-packages\platformio\cli.py", line 85, in invoke
    return super().invoke(ctx)
  File "C:\Users\Ich\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\Ich\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\Ich\.platformio\penv\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\Ich\.platformio\penv\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\Ich\.platformio\penv\lib\site-packages\platformio\package\commands\list.py", line 56, in package_list_cmd
    list_project_packages(options)
  File "C:\Users\Ich\.platformio\penv\lib\site-packages\platformio\package\commands\list.py", line 179, in list_project_packages
    _found = print_project_env_platform_packages(env, options)
  File "C:\Users\Ich\.platformio\penv\lib\site-packages\platformio\package\commands\list.py", line 192, in print_project_env_platform_packages
    p = PlatformFactory.from_env(project_env)
  File "C:\Users\Ich\.platformio\penv\lib\site-packages\platformio\platform\factory.py", line 102, in from_env
    p.configure_project_packages(env, targets)
  File "C:\Users\Ich\.platformio\penv\lib\site-packages\platformio\platform\base.py", line 184, in configure_project_packages
    self.configure_default_packages(options, targets or [])
  File "C:\Users\Ich\.platformio\platforms\ststm32\platform.py", line 110, in configure_default_packages
    return PlatformBase.configure_default_packages(self, variables,
  File "C:\Users\Ich\.platformio\penv\lib\site-packages\platformio\platform\base.py", line 196, in configure_default_packages
    self.packages[_pkg_name]["optional"] = False
KeyError: 'framework-stm32cubec0'

Looks like there was a change to PlatformIO itself that removed framework-stm32cubec0???

RudolphRiedel, 2024-01-07:
I tried this again adding my nucleo_c031c6.json to .platformio\platforms\ststm32\boards after installing 17.2.0 and I can not get this to work anymore.

It really would be nice if someone could take a look at this who has some understanding of what is going on there in regards of PlatformIO.

Overall I have to say that PlatformIO really is nice as long there already is support for the board you want to use and when the board definition supports the framework you like to use.
But beyond that PlatformIO tends to be a major pain for those like me who are unfamiliar with the inner workings of PlatformIO itself.
Do I want to know how to fix these things? Yes and No.
Yes, I am interested in fixing these things myself and therefore I would apreciate it if beeing pointed to a tutorial on how to add new boards.
But no in the sense that what I am actually trying to do is to compile software for the board I have right before me.

ftab, 2024-08-15:
I got the same error after making a custom board definition for the STM32C0116-DK. (It works in Arduino, but I can't use Arduino due to the license)

What do we need to do to get it to recognize c0 for stm32cube?

JikHuijberts, 2024-11-18:
This is still a problem with the current PlatformIO version. My suggestion is just like this issue https://github.com/platformio/platform-ststm32/issues/686 is adding the C0 chip range package: https://github.com/STMicroelectronics/STM32CubeC0 to the PlatformIO registry.

closed 2026-08-28 by valeros

jhmaloney, 2026-08-28:
Does "completed" mean that after I update  I can now use the env entry posted by Max (or similar) directly without patching my Platformio installation?

maxgerhardt, 2026-08-28:
No, the commits update the Arduino core and Zephyr and add these boards

[image]

So there's not a `genericSTM32C071KB` board definition yet, you still need those. The Arduino core supports far more STM32C0 chips that what was added into the ststm32 platform just now.

valeros, 2026-08-28:
Hi @jhmaloney, is `nucleo_c071rb` a suitable replacement for that DUELink board or you need that specific STM32C071KB MCU?

jhmaloney, 2026-08-28:
I think nucleo_c071rb is fine. I've been using that in the Arduino IDE.

jhmaloney, 2026-08-28:
So should the env entry be something like:
```
[env:DUELink]
platform = ststm32
framework = arduino
board = nucleo_g071rb
upload_protocol = dfu
...
```

maxgerhardt, 2026-08-28:
As of right now (since no new tag for the platform has been created yet), the best you can do is
```ini
[env:DUELink]
platform = https://github.com/platformio/platform-ststm32/archive/7c08b0c1a481983a56403e0680117d00aee7da1a.zip
framework = arduino
board = nucleo_c071rb
upload_protocol = dfu
```

jhmaloney, 2026-08-28:
I get "fatal error: stm32c0xx.h: No such file or directory"
That file was from the Arduino version of the framework. What is the equivalent file?

maxgerhardt, 2026-08-28:
Works fine for me. Is your config correct?

[image]

jhmaloney, 2026-08-28:
You're right, my config was wrong. Thanks!

jhmaloney, 2026-08-28:
I can't figure out how to get it to default to using USB as the Serial device. For Arduino, I was using the build flags:
```
-D ARDUINO_USB_CDC_ON_BOOT -D USBD_USE_CDC
```
It PlatformIO it looks like I should still use "-D USBD_USE_CDC" but I what is the equivalent of ARDUINO_USB_CDC_ON_BOOT? I don't see that flag defined or used anywhere in the framework.

Symptom is that the board does not appear as a USB CDC (serial) device in the /dev folder even after power cycling it.

maxgerhardt, 2026-08-28:
As per [docs](https://docs.platformio.org/en/latest/platforms/ststm32.html#configuration), setting
```ini
build_flags =
    -DPIO_FRAMEWORK_ARDUINO_ENABLE_CDC
```
in the `platformio.ini` is sufficient. The [builder script](https://github.com/stm32duino/Arduino_Core_STM32/blob/main/tools/platformio/platformio-build.py#L87) will derive the rest of the needed defines.

jhmaloney, 2026-08-28:
That did the trick! MicroBlocks is up and running on DUELink boards!

Also, thanks for the useful docs link...

One more question: is the "-D__CORTEX_SC=0" flag needed? I found that in the "boards.txt" file entry for the "NUCLEO_C071RB board". I'm guessing it is not needed in platformio...

Thank you for all your help!

maxgerhardt, 2026-08-28:
Great to hear!

PlatformIO still gives that flag, the same way the Arduino IDE does.

https://github.com/platformio/platform-ststm32/blob/7c08b0c1a481983a56403e0680117d00aee7da1a/boards/nucleo_c071rb.json#L7-L9

However, it is likely not needed anymore. A very old version of CMSIS 4 needed this defined, otherwise it would emit a warning when doing an `#if` on an undefined macro ([reference](https://github.com/ARM-software/CMSIS_4/blob/b94b8ff1cc5aa7e054bdc4b1bf76a2069fb1fde4/CMSIS/Include/core_cmInstr.h#L221)). The new STM32Duino core 3.0.0 uses CMSIS 6 and does not need that at all. In fact, `__CORTEX_SC = 0` is the identification macro for the Secure Core SC000 ([see reference](https://arm-software.github.io/CMSIS_6/main/Core/group__version__control__gr.html#gaeaaf66c86e5ae02a0e1fe542cb7f4d8c)). But nothing uses it, so it's a dead definition. It's fine to keep it, like the Arduino core does.. Or remove it with `build_unflags = -D__CORTEX_SC=0` in the `platformio.ini`. Absolutely nothing should change. I'd just leave it alone.