platformio/platform-ststm32 #694: Portenta H7: broken pipe (uploading with dfu)

sigurd4, 2023-04-26:
When i upload i get the following:

```
Processing portenta_h7 (platform: ststm32; board: portenta_h7_m7; framework: arduino)
--------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/portenta_h7_m7.html
PLATFORM: ST STM32 (15.6.0) > Arduino Portenta H7 (M7 core)
HARDWARE: STM32H747XIH6 480MHz, 511.35KB RAM, 768KB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduino-mbed @ 4.0.2 
 - tool-dfuutil @ 1.11.0 
 - tool-openocd @ 2.1100.211028 (11.0) 
 - tool-stm32duino @ 1.0.1 
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
Converting main.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/portenta_h7/src/main.ino.cpp.o
Checking size .pio/build/portenta_h7/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  11.8% (used 61560 bytes from 523624 bytes)
Flash: [==        ]  19.8% (used 156016 bytes from 786432 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, dfu, jlink, mbed, stlink
CURRENT: upload_protocol = dfu
Looking for upload port...
Using manually specified: /dev/ttyACM0
*** [upload] Broken pipe
===== [FAILED] Took 2.89 seconds =====

Environment    Status    Duration
-------------  --------  ------------
portenta_h7    FAILED    00:00:02.893
 1 failed, 0 succeeded in 00:00:02.893
```

I am running Ubuntu-20.04 on WSL2 for Windows 10 using usbip for attaching the usb-port to WSL.

This is my `platformio.ini` file:

```
[platformio]
default_envs = portenta_h7

[env:portenta_h7]
platform = ststm32
board = portenta_h7_m7
upload_protocol = dfu
framework = arduino
build_src_filter = +<*>
```

If i upload using Arduino IDE it works, both from Windows and WSL2 Ubuntu. I have updated the bootloader of the microcontroller. When uploading it's set to bootloader-mode, with the green LED blinking.

This deleted post on stackoverflow appears to describe the same problem: https://web.archive.org/web/20220602165523/https://stackoverflow.com/questions/72362799/broken-pipe-error-when-uploading-to-arduino-portenta-h7-with-platformio-on-lin

How come this doesn't work?

MarcelRobitaille, 2023-05-09:
@sigurd4 I'm also getting this. Did you find a solution?

sigurd4, 2023-05-09:
No. A temporary (non-ideal) solution is to lint and write the code in VSCode, and flash the device with Arduino IDE. If you use the newest version of Arduino IDE it's possible to make your workspace compatible with both IDE's, by adding an .ino file in the project folder with #include "main.hpp" and add a header file for main.cpp with setup and loop functions visible.

MarcelRobitaille, 2023-05-10:
I've had success pressing the reset button a single time before flashing instead of 2.

maxgerhardt, 2023-05-11:
```
Using manually specified: /dev/ttyACM0
```
Weird -- wouldn't DFU use a USB device, not a serial port? What happens when you delete the `upload_port = ..` section of your `platformio.ini`?

MarcelRobitaille, 2023-05-11:
I don't have any `upload_port` in my `platformio.ini` and I'm still getting this. 

```
Configuring upload protocol...
AVAILABLE: cmsis-dap, dfu, jlink, mbed, stlink
CURRENT: upload_protocol = dfu
Looking for upload port...
Auto-detected: /dev/ttyACM0
```

maxgerhardt, 2023-05-11:
Ah yes, it tries to find a serial port so that it can reset the board into bootloader mode using a 1200 baud open serial port procedure.

https://github.com/platformio/platform-ststm32/blob/9da8f1bafcfd8b2a59061962dc1e298f545b52a4/boards/portenta_h7_m7.json#L44-L46

Do you get any new messages on `sudo dmesg` after after the broken pipe error?

sigurd4, 2023-06-05:
@maxgerhardt I no longer have access to the Portenta H7 in question, because the project i was on is now over. Good luck

mathemaphysics, 2024-07-08:
Same thing here. One day I double click into blinking green light bootloader mode and flash and it gives broken pipe.

martincerven, 2024-10-09:
Any updates to this? Arduino IDE uploads sketches with no problems while platformio has:
```
CURRENT: upload_protocol = dfu
Looking for upload port...
Auto-detected: /dev/ttyACM0
*** [upload] Broken pipe
========================================================================= [FAILED] Took 6.13 seconds =========================================================================
```

I checked udev, rebooted,...

maxgerhardt, 2024-10-09:
One thing you could do is
1. Do a verbose upload ([CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) -> `pio run -v -t upload`) in PlatformIO to compare it with the verbose upload logs in the Arduino IDE (File -> Preferences -> Verbose Upload); this would show if the arguments for `dfu-util` are the same, or if the Arduino IDE does any additional steps, like resetting the board via a 1200bps touch
2. Equalize the dfu-util version between PlatformIO and the Arduino IDE: Find the path to the dfuutil binary that the Arduino IDE is using and use them to replace the files in `~/.platformio/packages/tool-dfuutil`

martincerven, 2024-10-17:
I think the problem is I wanted to build & upload it from Aarch 64 Jetson device.
```
Resolving portenta_h7_m7 dependencies...
Tool Manager: Installing platformio/toolchain-gccarmnoneeabi @ ~1.70201.0
UnknownPackageError: Could not find the package with 'platformio/toolchain-gccarmnoneeabi @ ~1.70201.0' requirements for your system 'linux_aarch64'
```

This version build & uploads, but results in probably **not functioning firmware**?
```
platform_packages = 
	toolchain-gccarmnoneeabi@~1.90301.0
```
As I got same [solid green light ](https://forum.arduino.cc/t/solid-green-light-on-portenta-h7-lite-after-firmware-upload/1145649)on Portenta H7.

I finally put to use my stlinkv3 which I never used before, and I could upload from Jetson to Portenta by stlink (and breakout board), but again: **solid green light** and no communication.

Then I tried using x86 PC, and there if I was in broken state from Jetson, even double clicking button to get into bootloader wouln'd upload with dfu.

With stlink it works with both normal and bootloader modes.
On x86:
```
PACKAGES:
 - framework-arduino-mbed @ 4.1.5
 - tool-dfuutil-arduino @ 1.11.0
 - tool-openocd @ 3.1200.0 (12.0)
 - tool-stm32duino @ 1.0.1
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)

```

So I think on Aarch64 Jetson (and probably other Aarch64) it doesn't work because of different toolchain-gccarmnoneeabi which is not released for Aarch64 with platformio.

But confusing parts is that on Jetson+Arduino IDE 1.8 **it works**, and also the bootloader double clicking doesn't work on platformio reliably while it works all the time on Arduino IDE.