platformio/platform-ststm32 #702: Add board Arduino Giga R1 Wifi

n13ldo, 2023-06-07:
Please add board 'Arduino Giga R1 Wifi' to the boards support list:
MCU: STM32H747XIH6, Cortex M7 480mhz

Mucho appreciated.

Haschtl, 2023-07-14:
I've created a test project for the Arduino GIGA [here](https://github.com/Haschtl/pio-arduinogiga/tree/main) 

Thanks to @maxgerhardt for the test-project for the [Portenta X8](https://github.com/maxgerhardt/pio-portentax8), which was my starting point. 

This may help until it's officially supported.

dberlin, 2023-09-22:
@Haschtl 
The fix_link.py is unnecessary if you change builder/frameworks/arduino/mbed-core/arduino-core-mbed.py to recognize the giga the same way it recognizes the portenta h7.

The issue with SerialRPC is due to the default ldf mode being not good enough.
If you set lib_ldf_mode to deep+, it will not be needed.
Patch attached

[link_giga.patch.txt](https://github.com/platformio/platform-ststm32/files/12696612/link_giga.patch.txt)

AndrewCapon, 2023-10-12:
Hi Guys,

I have put in two PRs to add the board:

https://github.com/platformio/platform-ststm32/pull/735

https://github.com/platformio/builder-framework-arduino-core-mbed/pull/9

Cheers

Andy

bdudelsack, 2023-11-19:
What is actually the status? I see the PRs were closed but I think without being merged.

AndrewCapon, 2023-11-19:
It is all a bit on hold.

The Arduino guys are currently working on a different way of differentiating between various STM32h7 boards, once this is done I will have another look at getting it into platformio.

dav322, 2023-12-28:
strange how it was working with 3.3.1 ide with 12/14/23 development version
https://github.com/platformio/platformio-core/archive/develop.zip

now with 3.3.2 and today's development version it is no longer present as a board type
Is there a daily archive of the development version so I can download the 12/14 version?

Maybe I will just buy a portenta lite since it has a smaller foot print and has a better connector for 160 port pins of the stm32h747 and it is only a few $ more

AndrewCapon, 2023-12-29:
> It is all a bit on hold.
> 
> The Arduino guys are currently working on a different way of differentiating between various STM32h7 boards, once this is done I will have another look at getting it into platformio.

Two new PRs to add the board in:

https://github.com/platformio/builder-framework-arduino-core-mbed/pull/10

https://github.com/platformio/platform-ststm32/pull/747

AndrewCapon, 2023-12-29:
Here is a zip with the changes is as well, basically the files go into the same location in the .platformio folder.

Probably best to back up the .py files before copying over them!

To check you are updating the correct version look in `/.platformio/packages/framework-arduino-mbed/variants`

You should see `GENERIC_STM32H747_M4` and `GIGA`. If you don't then this will not work!

[giga.zip](https://github.com/platformio/platform-ststm32/files/13792895/giga.zip)

dav322, 2023-12-29:
It would not work if I used my existing project.  It worked fine when I created a new project.  Thanks for the files.  I just found the same files in .platformio/platforms/ststm32 and overwrote them with files from your zip file.

only compile problem I had was HSE_VALUE was different in 
C:\Users\david\.platformio\packages\framework-arduino-mbed\variants\GIGA\mbed_config.h and 
C:\Users\david\.platformio\packages\framework-arduino-mbed\cores\arduino\mbed\targets\TARGET_STM\TARGET_STM32H7\STM32Cube_FW\stm32h7xx_hal_conf.h

I changed the 2nd file to match the first.  I assume this is the issue they are trying to fix as you mentioned above.

here are compiler warnings:
In file included from C:\Users\david\.platformio\packages\framework-arduino-mbed\variants\GIGA/pinmode_arduino.h:23:0,
                 from C:\Users\david\.platformio\packages\framework-arduino-mbed\cores\arduino/Arduino.h:26,
                 from C:\Users\david\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed.h:4,
                 from src\zsharedheader/os.h:16,
                 from src\init\StartUsbTask.c++:9:
C:\Users\david\.platformio\packages\framework-arduino-mbed\variants\GIGA/mbed_config.h:56:0: warning: "HSE_VALUE" redefined
 #define HSE_VALUE                                                               16000000                                  
                                                       // set by target:GIGA

In file included from C:\Users\david\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/STM32H7xx_HAL_Driver/stm32h7xx_hal.h:30:0,
                 from src\zsharedheader/mytypes.h:13,
                 from src\init\StartUsbTask.c++:2:
C:\Users\david\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/stm32h7xx_hal_conf.h:109:0: note: this is the location of the previous definition
 #define HSE_VALUE    25000000 //(25000000UL) /*!< Value of the External oscillator in Hz */

after it compiled when I clicked on debug and then run button.  It compiled without error but it never loaded the program with stlink.

debug console was blank and terminal had gdb> prompt and vscode hung with bargraph continuing to run on the debug window.  I let it run for a couple of minutes.

I closed visual studio and tried debugging again and then it loaded my program and stopped at a breakpoint in C:\Users\david\.platformio\packages\framework-arduino-mbed\cores\arduino\main.cpp
I could then step through the initialization.

So its almost there
Let be know if you want me to do more tests or want some log files

AndrewCapon, 2023-12-29:
So if you create a new project and build it it gives you those errors with the default main.cpp?

m7 or m4?

dav322, 2023-12-29:
it's for the m7.

I renamed the .platformio directory
repatched it

running simple hello world application
I am back to debugger not starting

this is log from terminal
 *  Executing task: C:\Users\david\.platformio\penv\Scripts\platformio.exe debug 

Preparing firmware for debugging...
Processing giga_r1_m7 (platform: ststm32; board: giga_r1_m7; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/giga_r1_m7.html
PLATFORM: ST STM32 (17.2.0) > Arduino Giga R1 (M7 core)
HARDWARE: STM32H747XIH6 480MHz, 511.35KB RAM, 768KB Flash
DEBUG: Current (stlink) External (cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduino-mbed @ 4.0.10 
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
No dependencies
Building in debug mode

bunch of compiles main.cpp and library

Linking .pio\build\giga_r1_m7\firmware.elf
Checking size .pio\build\giga_r1_m7\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   8.7% (used 45400 bytes from 523624 bytes)
Flash: [=         ]  12.1% (used 95456 bytes from 786432 bytes)
Building .pio\build\giga_r1_m7\firmware.bin
=================================================================== [SUCCESS] Took 11.39 seconds ===================================================================
Reading symbols from C:\Users\david\Documents\adevel\private\giga\platformio\hello world\.pio\build\giga_r1_m7\firmware.elf...done.
(gdb)

Then it just hangs

-------------------------------------------------------------------
I fixed the HSE_VALUE warning by including mbed.h before including stm32h7xx_hal.h"

Also when I call __HAL_RCC_GPIOA_CLK_ENABLE(); I get a warning

In file included from C:\Users\david\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/STM32H7xx_HAL_Driver/stm32h7xx_hal_rcc.h:29:0,
                 from C:\Users\david\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/stm32h7xx_hal_conf.h:398,
                 from C:\Users\david\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/STM32H7xx_HAL_Driver/stm32h7xx_hal.h:30,
                 from src\zsharedheader/mytypes.h:13,
                 from src\main.cpp:2:
src\main.cpp: In function 'void MX_GPIO_Init()':
C:\Users\david\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/STM32H7xx_HAL_Driver/stm32h7xx_hal_rcc.h:1430:48: warning: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'        
                                         UNUSED(tmpreg); \
C:\Users\david\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/STM32H7xx_HAL_Driver/stm32h7xx_hal_def.h:70:27: note: in definition of macro 'UNUSED'
 #define UNUSED(x) ((void)(x))
                           ^
src\main.cpp:147:3: note: in expansion of macro '__HAL_RCC_GPIOA_CLK_ENABLE'
   __HAL_RCC_GPIOA_CLK_ENABLE();

---------------------------

I suggest changing C:\Users\david\.platformio\packages\framework-arduino-mbed\cores\arduino\mbed\targets\TARGET_STM\TARGET_STM32H7\STM32Cube_FW\STM32H7xx_HAL_Driver\stm32h7xx_hal_rcc.h
#define __HAL_RCC_GPIOA_CLK_ENABLE()   do { \
                                        __IO uint32_t tmpreg; \
                                        SET_BIT(RCC->AHB4ENR, RCC_AHB4ENR_GPIOAEN);\
                                        /* Delay after an RCC peripheral clock enabling */ \
                                        tmpreg = READ_BIT(RCC->AHB4ENR, RCC_AHB4ENR_GPIOAEN);\
                                        UNUSED(tmpreg); \
                                       } while(0)

to

#define __HAL_RCC_GPIOA_CLK_ENABLE()   do { \
                                        SET_BIT(RCC->AHB4ENR, RCC_AHB4ENR_GPIOAEN);\
                                        SET_BIT(RCC->AHB4ENR, RCC_AHB4ENR_GPIOAEN);\
                                       } while(0)

which provides the delay and does not generate a warning.

AndrewCapon, 2023-12-30:
Hi @dav322 

I just installed a new .platformio, patched the files, created a ne giga m7 project.

Everything builds fine.

```
Processing giga_r1_m7 (platform: ststm32; board: giga_r1_m7; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/giga_r1_m7.html
PLATFORM: ST STM32 (17.2.0) > Arduino Giga R1 (M7 core)
HARDWARE: STM32H747XIH6 480MHz, 511.35KB RAM, 768KB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduino-mbed @ 4.0.10 
 - tool-dfuutil-arduino @ 1.11.0 
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/giga_r1_m7/src/main.cpp.o
Generating LD script .pio/build/giga_r1_m7/cpp.linker_script.ld
Compiling .pio/build/giga_r1_m7/FrameworkArduinoVariant/pure_analog_pins.cpp.o
Compiling .pio/build/giga_r1_m7/FrameworkArduinoVariant/variant.cpp.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/Interrupts.cpp.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/Serial.cpp.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/Tone.cpp.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/USB/PluggableUSBDevice.cpp.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/USB/USBCDC.cpp.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/USB/USBSerial.cpp.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/WMath.cpp.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/abi.cpp.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/api/CanMsg.cpp.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/api/CanMsgRingbuffer.cpp.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/api/Common.cpp.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/api/IPAddress.cpp.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/api/PluggableUSB.cpp.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/api/Print.cpp.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/api/Stream.cpp.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/api/String.cpp.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/arm_hal_random.c.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/as_mbed_library/variant.cpp.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/itoa.c.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/main.cpp.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/mbed/platform/cxxsupport/mstd_mutex.cpp.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/pinToIndex.cpp.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/random_seed.cpp.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/timer.cpp.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/wiring.cpp.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/wiring_analog.cpp.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/wiring_digital.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:8:7: warning: unused variable 'result' [-Wunused-variable]
   int result = myFunction(2, 3);
       ^~~~~~
Compiling .pio/build/giga_r1_m7/FrameworkArduino/wiring_pulse.cpp.o
Compiling .pio/build/giga_r1_m7/FrameworkArduino/wiring_shift.cpp.o
Archiving .pio/build/giga_r1_m7/libFrameworkArduinoVariant.a
Indexing .pio/build/giga_r1_m7/libFrameworkArduinoVariant.a
Archiving .pio/build/giga_r1_m7/libFrameworkArduino.a
Indexing .pio/build/giga_r1_m7/libFrameworkArduino.a
Linking .pio/build/giga_r1_m7/firmware.elf
Checking size .pio/build/giga_r1_m7/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   8.7% (used 45400 bytes from 523624 bytes)
Flash: [=         ]  12.0% (used 94056 bytes from 786432 bytes)
Building .pio/build/giga_r1_m7/firmware.bin
================================================================================================================================ [SUCCESS] Took 5.84 seconds ================================================================================================================================
 *  Terminal will be reused by tasks, press any key to close it. 
```

With jlink:

Debugging with PIO debugger works fine.

Debugging with cortex-debug works fine.

Can you attach the main.cpp here you are using.

dav322, 2024-01-05:
I rebooted computer and it is starting the debugger again with stlink.   Not sure why that had any effect.

AndrewCapon, 2024-01-06:
When I first got the GIGA I had problems with my STLINK V2 and platformio, it worked fine from the STM IDE though.

So I switched to a jlink and that seemed to work much more reliably.

I also found using cortex-debug to be much more reliable than the platformio debugger as well, especially when debugging the M4.

DIGITDUDE, 2024-01-16:
so far it work great, except the uploading to my Giga R1 controller doesnt work.

it goes to the how compiling proces, find the right com port - puts the GIGA R1 in bootloader mode than this:

Building in release mode
Checking size .pio\build\giga_r1_m7\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  19.7% (used 103264 bytes from 523624 bytes)
Flash: [====      ]  36.6% (used 287728 bytes from 786432 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, dfu, jlink, mbed, stlink
CURRENT: upload_protocol = dfu
Looking for upload port...
Auto-detected: COM3
Forcing reset using 1200bps open/close on port COM3
Waiting for the new upload port...
Uploading .pio\build\giga_r1_m7\firmware.bin
dfu-util 0.11-arduino4

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

No DFU capable USB device available
*** [upload] Error 74
========================================================================== [FAILED] Took 8.10 seconds ===

DIGITDUDE, 2024-01-16:
that is the weird thing, Platformio seems to activate dfu mode it self, after that it tries to find the port again and doesn't find it again. even when i first put the giga in dfu mode myself by double clicking the resetbutton it shows the exact same behaviour.
but it seems like il have to learn how to use jtag.

Paolino625, 2024-01-16:
I ran into the same problem when uploading. Figured it out by adding this line in the platformio.ini under the environment settings.

`upload_command = arduino-cli upload --port=/dev/cu.usbmodem2101 -b arduino:mbed_giga:giga -i .pio/build/ArduinoMaster/firmware.bin`

 I can't remember where I saw this trick; you might have to play around with the port to get it working.

I hope it helps.

DIGITDUDE, 2024-01-16:
thanks that worked indeed, had to change 

from:
upload_command = arduino-cli upload --port=/dev/cu.usbmodem2101 -b arduino:mbed_giga:giga -i .pio/build/ArduinoMaster/firmware.bin 

to

upload_command = arduino-cli upload --port=/dev/cu.usbmodem2101 -b arduino:mbed_giga:giga -i .pio/build/giga_r1_m7/firmware.bin

because of the new files used by the zip file from andrew.

AndrewCapon, 2024-01-16:
Does the port change then when it is in DFU mode?

cyberhelmer, 2024-03-12:
I'm really trying to get this to work but it's a little messy following this thread, and I'm not sure what to do really. Would anyone be so kind to post a little step by step guide to how to make this work? 

I'm on a mac btw.

AndrewCapon, 2024-03-12:
> I'm really trying to get this to work but it's a little messy following this thread, and I'm not sure what to do really. Would anyone be so kind to post a little step by step guide to how to make this work?
> 
> I'm on a mac btw.

Have a look here: https://github.com/platformio/platform-ststm32/issues/702#issuecomment-1871959957

cyberhelmer, 2024-03-12:
> Have a look here: #702 (comment)

I did manage to install so I had the two folders GENERIC_STM32H747_M4 and
GIGA in in /.platformio/packages/framework-arduino-mbed/variants

And I replaced the ststm32 folder with the one from your zip file.

But not sure what board to select when creating a platform.io project and
what the platformio.ini file should be. For selecting a board neither the
Giga or STM32H747XI shows up?

My platformio.ini file looks like this:
platform = ststm32
board = giga_r1_m4
framework = arduino

But I get squiggly lines under ststm32 saying:
File contains no section headers.
file: '/Users/rubymainstage/projects/Arduino
Projects/GigaTest2/platformio.ini', line: 12
'platform = ststm32\n'

Thank you very much for your help!

AndrewCapon, 2024-03-12:
When you create a project you should be able to select the giga M7 or M4, does that not work?

[image]

cyberhelmer, 2024-03-12:
I reinstalled platform.io and did everything again and now the boards show, but there seem to be a problem when creating a new project with any of the Giga boards selected. I get this message: 

Could not initialize project
PIO Core Call Error: "The following files/directories have been created in /Users/rubymainstage/projects/Arduino Projects/Test\ninclude - Put project header files here\nlib - Put project specific (private) libraries here\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\nResolving giga_r1_m4 dependencies...\nAlready up-to-date.\nUpdating metadata for the vscode IDE...\n\n\nUserSideException: Processing giga_r1_m4 (platform: ststm32; board: giga_r1_m4; framework: arduino)\n--------------------------------------------------------------------------------\nVerbose mode can be enabled via `-v, --verbose` option\nCONFIGURATION: https://docs.platformio.org/page/boards/ststm32/giga_r1_m4.html\nPLATFORM: ST STM32 (17.3.0) > Arduino Giga R1 (M4 core)\nHARDWARE: STM32H747XIH6 480MHz, 287.35KB RAM, 1MB Flash\nDEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, stlink)\nPACKAGES: \n - framework-arduino-mbed @ 4.1.1 \n - tool-dfuutil-arduino @ 1.11.0 \n - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)\nLDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf\nLDF Modes: Finder ~ chain, Compatibility ~ soft\nFound 42 compatible libraries\nScanning dependencies...\nNo dependencies\nBuilding in release mode\nKeyError: 'tool-dfuutil':\n File \"/Users/rubymainstage/.platformio/penv/lib/python3.11/site-packages/platformio/builder/main.py\", line 173:\n env.SConscript(\"$BUILD_SCRIPT\")\n File \"/Users/rubymainstage/.platformio/packages/tool-scons/scons-local-4.6.0/SCons/Script/SConscript.py\", line 609:\n return _SConscript(self.fs, *files, **subst_kw)\n File \"/Users/rubymainstage/.platformio/packages/tool-scons/scons-local-4.6.0/SCons/Script/SConscript.py\", line 279:\n exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)\n File \"/Users/rubymainstage/.platformio/platforms/ststm32/builder/main.py\", line 221:\n _upload_tool = '\"%s\"' % join(platform.get_package_dir(\n File \"/Users/rubymainstage/.platformio/penv/lib/python3.11/site-packages/platformio/platform/_packages.py\", line 32:\n pkg = self.get_package(name)\n File \"/Users/rubymainstage/.platformio/penv/lib/python3.11/site-packages/platformio/platform/_packages.py\", line 29:\n return self.pm.get_package(spec or self.get_package_spec(name))\n File \"/Users/rubymainstage/.platformio/penv/lib/python3.11/site-packages/platformio/platform/_packages.py\", line 21:\n owner=self.packages[name].get(\"owner\"),\n========================== [FAILED] Took 1.39 seconds =========================="

cyberhelmer, 2024-03-12:
This is my platformio.ini: 
[env:giga_r1_m7]
platform = ststm32
board = giga_r1_m7
framework = arduino
platform_packages = platformio/framework-arduino-mbed@^4.1.1

This is what I get when trying to build: 
Resolving giga_r1_m7 dependencies...
Already up-to-date.
Updating metadata for the vscode IDE...
UserSideException: Processing giga_r1_m7 (platform: ststm32; board: giga_r1_m7; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/giga_r1_m7.html
PLATFORM: ST STM32 (17.3.0) > Arduino Giga R1 (M7 core)
HARDWARE: STM32H747XIH6 480MHz, 511.35KB RAM, 768KB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduino-mbed @ 4.1.1 
 - tool-dfuutil-arduino @ 1.11.0 
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
KeyError: 'tool-dfuutil':
  File "/Users/rubymainstage/.platformio/penv/lib/python3.11/site-packages/platformio/builder/main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "/Users/rubymainstage/.platformio/packages/tool-scons/scons-local-4.6.0/SCons/Script/SConscript.py", line 609:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/rubymainstage/.platformio/packages/tool-scons/scons-local-4.6.0/SCons/Script/SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/rubymainstage/.platformio/platforms/ststm32/builder/main.py", line 221:
    _upload_tool = '"%s"' % join(platform.get_package_dir(
  File "/Users/rubymainstage/.platformio/penv/lib/python3.11/site-packages/platformio/platform/_packages.py", line 32:
    pkg = self.get_package(name)
  File "/Users/rubymainstage/.platformio/penv/lib/python3.11/site-packages/platformio/platform/_packages.py", line 29:
    return self.pm.get_package(spec or self.get_package_spec(name))
  File "/Users/rubymainstage/.platformio/penv/lib/python3.11/site-packages/platformio/platform/_packages.py", line 21:
    owner=self.packages[name].get("owner"),
========================== [FAILED] Took 0.99 seconds ==========================

dberlin, 2024-03-12:
I'm not sure there is a native tool-dfuutil for mac, or the dependencies are now wrong.

Can you try adding:

platform_packages = 
    tool-dfuutil@~1.9

to the [env...] you are using and see if that fixes it?

dberlin, 2024-03-12:
Actually, I misread your config - unless you have locally patched in some of the platformio changes, your env is not enough.
In particular, it does not have the patch to use tool-dfu-arduino for giga on arduino.

If you instead change platform = stmst32
to
platform = https://github.com/dberlin/platform-ststm32.git#develop

I would bet it will just work.

cyberhelmer, 2024-03-12:
Holy cow it worked after changing the platform = https://github.com/dberlin/platform-ststm32.git#develop and adding platform_packages = tool-dfuutil@~1.9!!! 

Thousand thanks!! I been trying to get this to work for 15 hours straight, I was 10 minutues from jumping from the balcony.

dberlin, 2024-03-12:
> Holy cow it worked after changing the platform = https://github.com/dberlin/platform-ststm32.git#develop and adding platform_packages = tool-dfuutil@~1.9!!!
> 
> Thousand thanks!! I been trying to get this to work for 15 hours straight, I was 10 minutues from jumping from the balcony.

Happy to help.

Do you actually need the tool-dfuutil? If so, it means they still havent' gotten tool-dfuutil out for native mac, which would be weird.

cyberhelmer, 2024-03-12:
No you're right, it works with that line removed to.

AndrewCapon, 2024-03-13:
Hi Guys, I'm a bit confused.

Those changes are already in the zip file provided.

dberlin, 2024-03-13:
> Hi Guys, I'm a bit confused.
> 
> Those changes are already in the zip file provided?

Overwriting with the zip file only lasts until the next time it decides to update things :)
Using a fork works forever.

AndrewCapon, 2024-03-13:
Ah right, got you now :)

Virusssssss21, 2024-03-17:
Hey guys, would love to use my Arduino giga with platfromIO. Could someone explain the procedure i need to follow to acomplisch this. Thanks a lot.

dberlin, 2024-03-18:
Use: 

```
[env:giga_r1_m7]
platform = https://github.com/dberlin/platform-ststm32.git#develop
board = giga_r1_m7
framework = arduino

```

dansiviter, 2024-05-13:
@dberlin To get this to work I had to add the following or it wouldn't compile:
```
...
lib_ignore =
	GSM
	Ethernet
```
Note sure how the filter these from the mBed core for a seamless build.

Couple of other things:
* I noticed the M4 has `wifi` enabled which according to the [docs](https://docs.arduino.cc/tutorials/giga-r1-wifi/giga-dual-core/#m4-support), that core doesn't support it,
* I added "blackmagic" probe on [my fork](//github.com/dansiviter/platform-ststm32#develop) and it works.

ahak-Yem, 2024-06-26:
> Here is a zip with the changes is as well, basically the files go into the same location in the `.platformio` folder.
>
> Probably best to back up the .py files before copying over them!
>
> To check you [...]

So the same location means: "~\.platformio\platforms\ststm32"
And then paste the main.py there and the content of the other two folders in the corresponding locations

AndrewCapon, 2024-06-26:
Yep all the files in the `ststm32 folder` in the zip file go into the `ststm32` folder in `~/.platformio/platforms`

ahak-Yem, 2024-06-26:
Thanks mate. It works good. I hope it will get merged and will be provided by default when a new PlatformIO update comes up.

tbattisti, 2024-07-17:
Will there eventually be an official solution for this issue?

AndrewCapon, 2024-07-30:
Hi guys, 

Both the PRs needed have been merged now.

So I'm guessing the next PIO release will have this all in.

Cheers

Andy

valeros, 2024-08-08:
Resolved in https://github.com/platformio/platform-ststm32/pull/747

closed 2024-08-08 by valeros