platformio/platform-ststm32 #713: Syntax error "(" unexpected

4ydan, 2023-07-13:
After running `pio pkg update -p ststm32` because I had this issue #655 I get syntax errors when I want to build my project.
```
Processing Nucleo_F411RE_USB_RTOS (platform: ststm32; board: nucleo_f411re; framework: arduino)
-----------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_f411re.html
PLATFORM: ST STM32 (16.0.0) > ST Nucleo F411RE
HARDWARE: STM32F411RET6 100MHz, 128KB RAM, 512KB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, cmsis-dap, jlink)
PACKAGES: 
 - framework-arduinoststm32 @ 4.20500.230713 (2.5.0) 
 - framework-cmsis @ 2.50700.210515 (5.7.0) 
 - tool-dfuutil @ 1.11.0 
 - tool-openocd @ 3.1200.0 (12.0) 
 - tool-stm32duino @ 1.0.1 
 - toolchain-gccarmnoneeabi @ 1.100301.220327 (10.3.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 22 compatible libraries
Scanning dependencies...
Dependency Graph
|-- STM32duino FreeRTOS @ 10.3.2
|-- modbusino_FreeRTOS
|-- Adafruit SSD1306 @ 2.5.7
|-- SHT21-Arduino-Library
|-- Wire @ 1.0.0
|-- littlefs-2.5.1
|-- MPU6050 @ 0.6.0
Building in debug mode
Compiling .pio/build/Nucleo_F411RE_USB_RTOS/FrameworkArduinoVariant/PeripheralPins.c.o
sh: 1: Syntax error: "(" unexpected
Compiling .pio/build/Nucleo_F411RE_USB_RTOS/FrameworkArduinoVariant/PeripheralPins_PYBSTICK26_STD.c.o
Compiling .pio/build/Nucleo_F411RE_USB_RTOS/FrameworkArduinoVariant/generic_clock.c.o
sh: 1: Syntax error: "(" unexpected
Compiling .pio/build/Nucleo_F411RE_USB_RTOS/FrameworkArduinoVariant/variant_generic.cpp.o
sh: 1: Syntax error: "(" unexpected
sh: 1: Syntax error: "(" unexpected
Compiling .pio/build/Nucleo_F411RE_USB_RTOS/FrameworkArduinoVariant/variant_PYBSTICK26_STD.cpp.o
sh: 1: Syntax error: "(" unexpected
Compiling .pio/build/Nucleo_F411RE_USB_RTOS/FrameworkArduinoVariant/variant_NUCLEO_F411RE.cpp.o
Compiling .pio/build/Nucleo_F411RE_USB_RTOS/SrcWrapper/src/HAL/stm32yyxx_hal_adc.c.o
sh: 1: Syntax error: "(" unexpected
Compiling .pio/build/Nucleo_F411RE_USB_RTOS/SrcWrapper/src/HAL/stm32yyxx_hal.c.o
sh: 1: Syntax error: "(" unexpected
sh: 1: Syntax error: "(" unexpected
Compiling .pio/build/Nucleo_F411RE_USB_RTOS/SrcWrapper/src/HAL/stm32yyxx_hal_adc_ex.c.o
sh: 1: Syntax error: "(" unexpected
Compiling .pio/build/Nucleo_F411RE_USB_RTOS/SrcWrapper/src/HAL/stm32yyxx_hal_can.c.o
sh: 1: Syntax error: "(" unexpected
Compiling .pio/build/Nucleo_F411RE_USB_RTOS/SrcWrapper/src/HAL/stm32yyxx_hal_cec.c.o
sh: 1: Syntax error: "(" unexpected
*** [.pio/build/Nucleo_F411RE_USB_RTOS/FrameworkArduinoVariant/PeripheralPins.c.o] Error 2
*** [.pio/build/Nucleo_F411RE_USB_RTOS/FrameworkArduinoVariant/generic_clock.c.o] Error 2
*** [.pio/build/Nucleo_F411RE_USB_RTOS/FrameworkArduinoVariant/variant_generic.cpp.o] Error 2
*** [.pio/build/Nucleo_F411RE_USB_RTOS/FrameworkArduinoVariant/variant_PYBSTICK26_STD.cpp.o] Error 2
*** [.pio/build/Nucleo_F411RE_USB_RTOS/FrameworkArduinoVariant/PeripheralPins_PYBSTICK26_STD.c.o] Error 2
*** [.pio/build/Nucleo_F411RE_USB_RTOS/SrcWrapper/src/HAL/stm32yyxx_hal_adc.c.o] Error 2
*** [.pio/build/Nucleo_F411RE_USB_RTOS/FrameworkArduinoVariant/variant_NUCLEO_F411RE.cpp.o] Error 2
*** [.pio/build/Nucleo_F411RE_USB_RTOS/SrcWrapper/src/HAL/stm32yyxx_hal.c.o] Error 2
*** [.pio/build/Nucleo_F411RE_USB_RTOS/SrcWrapper/src/HAL/stm32yyxx_hal_adc_ex.c.o] Error 2
*** [.pio/build/Nucleo_F411RE_USB_RTOS/SrcWrapper/src/HAL/stm32yyxx_hal_can.c.o] Error 2
Compiling .pio/build/Nucleo_F411RE_USB_RTOS/SrcWrapper/src/HAL/stm32yyxx_hal_comp.c.o
*** [.pio/build/Nucleo_F411RE_USB_RTOS/SrcWrapper/src/HAL/stm32yyxx_hal_cec.c.o] Error 2
sh: 1: Syntax error: "(" unexpected
*** [.pio/build/Nucleo_F411RE_USB_RTOS/SrcWrapper/src/HAL/stm32yyxx_hal_comp.c.o] Error 2
```

valeros, 2023-07-13:
Sorry my bad, it should be fixed in the latest package, please pull the update via `pio pkg update -p ststm32`.

4ydan, 2023-07-13:
I somehow can't pull the latest package, it says already up-to-date. I also tried deleting all and installing them fresh, problem persists.

valeros, 2023-07-13:
Maybe something is cached, have you tried deleting the `~/.platformio/.cache` folder?

4ydan, 2023-07-13:
Awesome, everything works again.
Thx

closed 2023-07-13 by 4ydan