platformio/platform-ststm32 #715: Missleading support for STM32H723

yamaan93, 2023-07-29:
The STM32H723 series (and specifically the [ST Nucleo H723ZG](https://docs.platformio.org/en/stable/boards/ststm32/nucleo_h723zg.html#id1)) is listed as a supported board with support for Arduino, STM32Cube, and CMSIS frameworks, however the [CMSIS package used for the H7 boards ](https://registry.platformio.org/tools/platformio/framework-cmsis-stm32h7/installation) is on release 1.8.0 which is 1 version before support for the H723 series chips was added. This makes it so you can't actually build anything for those boards, which suggests they aren't actually supported.

valeros, 2023-07-31:
Thanks for pointing it out, fixed in the dev branch.

closed 2023-07-31 by valeros

valeros, 2023-07-31:
I probably misinterpreted your question in the first place. However, enabled CMSIS support for STM32H723 was indeed misleading, so I resolved the issue in the most straightforward way I could. Please note that while we would love to keep all packages up to date, we face certain limitations, particularly when it comes to integrating with projects from vendors outside of our official partnerships. Our resources and capacity are dedicated to maintaining compatibility with hardware that belongs to silicon vendors we actively collaborate with. Nevertheless, your feedback has not gone unnoticed, and we will take it into account when planning future improvements and integrations.

commit a6a0cd52 (2023-07-31): Update ST Nucleo H723ZG config

- Removed `cmsis` from the list of supported frameworks

Resolves #715