platformio/platform-ststm32 #718: Update to Arduino Core for STM32 2.6.0

vshymanskyy, 2023-08-09:
I'm interested in LORA_E5_MINI (STM32WLE5JC) board support which was added in:
https://github.com/stm32duino/Arduino_Core_STM32/commit/aed76aae10d274b4ee86c9c43dc3ed23fead5565

@valeros this would also close #581 #624 #596

RudolphRiedel, 2023-08-12:
That would also add STM32H5 support and I have a Nucleo-H503RB that I would like to see working with PlatformIO.

sstaub, 2023-08-13:
Nucleo H563ZI board is also missing.

closed 2023-09-19 by valeros

vshymanskyy, 2023-09-19:
Thank you @valeros !
Timing is perfect

vshymanskyy, 2023-10-16:
It's still unclear on how to actually select this board in Platformio...
It's STM32WLE5JC, In Arduino IDE it's under `stm32:LoRa`, than you need to select `Tools -> Board Part Number -> LoRa-E5 mini`

vshymanskyy, 2023-10-16:
I was able to use LoRa-E5 mini:
```ini
[env:lora-e5]
platform = ststm32@17.0.0
board = lora_e5_mini
framework = arduino
debug_tool = stlink
upload_protocol = stlink
```

ypearson-bdai, 2024-10-02:
Any updates to support nucleo_h563zi?

I'm getting `: Unknown board ID 'nucleo_h563zi'`

commit 1c5bffe7 (2023-09-19): Update Arduino core to v2.6.0

Resolves #718, resolves #728