platformio/platform-ststm32 #720: Mismatched GCC version for STM32Duino

maxgerhardt, 2023-08-16:
PlatformIO uses arm-none-eabi-gcc 10.3.1 for all regular STM32 + Arduino projects

https://github.com/platformio/platform-ststm32/blob/a6a0cd52968b5a209aa56f91a9e5e803fd5a769a/platform.py#L49-L52

however, the package declaration <https://raw.githubusercontent.com/stm32duino/BoardManagerFiles/main/package_stmicroelectronics_index.json> declares that since version 2.5.0, the `12.2.1-1.2` version of the toolchain must be used.

Not doing so causes failures in e.g. the STM32FreeRTOS library beceause the older toolchain version was compiled with different Newlib library and newlib options (no `_REENT_BACKWARD_BINARY_COMPAT`) which shrinks down the size of `struct _reent` from over 1000 bytes to only 288 bytes which decreases the FreeRTOS `TCB_t` task control block structure accordingly. This basically means that with PlatofrmIO, each created FreeRTOS task has a *signifacant* higher heap memory usage and out-of-memory occurs ealier or instantly. See <https://community.platformio.org/t/stm32duino-freertos-problems/35163>.

valeros, 2023-08-17:
Thanks for pointing it out, updated.

maxgerhardt, 2023-08-17:
Could you verify that the debugger works in this toolchain version? The user is just stuck on it [here](https://community.platformio.org/t/stm32duino-freertos-problems/35163/31?u=maxgerhardt)

valeros, 2023-08-17:
Indeed there is a problem, GDB v12 is not able to launch a debug server in pipe mode. I reverted the toolchain update to avoid breaking already working projects. As of now I'd probably recommend switching to GCC v12 directly in `platformio.ini`:

```ini
[env:nucleo_f401re]
platform = ststm32
framework = arduino
board = nucleo_f401re
debug_tool = custom
platform_packages = 
   toolchain-gccarmnoneeabi @ ~1.120201.0
```
and launching OpenOCD in a separate process, for example

```bash
"C:/Users/YOUR_USER/.platformio/packages/tool-openocd/bin/openocd.exe" -c "tcl_port disabled; telnet_port disabled" -f "C:/Users/YOUR_USER/.platformio/packages/tool-openocd/openocd/scripts/board/st_nucleo_f4.cfg"
```

maxgerhardt, 2023-08-17:
Very weird thing. Maybe https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/releases behaves differently? :/

valeros, 2023-08-17:
We use that exact packages.

closed 2023-09-22 by valeros

maxgerhardt, 2023-09-22:
I assume GCC/GDB 12.3.1 also resolve the "GDB can't start OpenOCD in pipe mode" issue? Does this also resolve https://github.com/platformio/platformio-core/issues/4741 on Linux?

commit d280708e (2023-08-17): Update toolchain package to v12.2.1 for Arduino integration

Resolves #720
commit 78196250 (2023-09-22): Update toolchain package to v12.3.1 for Arduino integration

Resolves #720