platformio/platform-ststm32 #722: Github troll placing ""malware"" text in documentation repos

maxgerhardt, 2023-08-18:
Currently, four boards point to "WeActTC" github repos

https://github.com/search?q=repo%3Aplatformio%2Fplatform-ststm32%20https%3A%2F%2Fgithub.com%2FWeActTC&type=code

https://github.com/platformio/platform-ststm32/blob/7bf9172b067b38fba167e95ec764c9bfaf3b10b0/boards/blackpill_f411ce.json#L46-L47

However, the original user account (https://github.com/WeActTC) was deleted and they seem to have moved over to the chinese gitee.com website, e.g., https://gitee.com/WeAct-TC/WeActStudio.MiniSTM32F4x1.

Another user by a different username was somehow able to get a redirect to their github repo (https://github.com/modauthgssapi/MiniSTM32F4x1) and place a "this is malware" file on it, which is just the text 

>This would be an executable file that the user would run and thus you would have RCE. Obviously this is a benign txt file for PoC purposes.

maxgerhardt, 2023-08-18:
In a boarder sense, I think when users in the VSCode PIO Home or on docs.platformio.org/ website click on an external link that's read from the board's or package's metadata, there needs to be a huge "You're about to open an external link, PlatformIO has no control over its content" disclaimer. Or, somehow archive / mirror a picture of the website at the time of referencing that can be safely referenced. CC @ivankravets

valeros, 2023-08-18:
Thanks for the report, I've updated the urls. As for the external links warning, it would be better to open a proper a feature requests in PlatformIO Home / Docs repos.

commit 6e0a89b8 (2023-08-18): Update links to WeAct Studio boards

Resolve #722
closed 2023-08-18 by valeros