platformio/platform-ststm32 #729: Uploading to Arduino Opta fails: libusb-1.0.0.dylib not loaded?

papakpmartin, 2023-09-18:
On MacBook Pro M2 (Mac OS 13.3.1) running PlatformIO IDE v3.3.1 in VS Code 1.82.2

Main error: `dyld[2890]: Library not loaded: /usr/local/opt/libusb/lib/libusb-1.0.0.dylib`

Uploading from Arduino IDE works fine. The device is actually at `/dev/cu.usbmodem1101`.

`platformio.ini` contains only:

```
[env:opta]
platform = ststm32
board = opta
framework = arduino
```

Attempting to upload and run the "Testing with Blink Sketch" code gets me the following:

```
 *  Executing task: platformio run --target upload --upload-port /dev/cu.usbmodem1101 

Processing opta (platform: ststm32; board: opta; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/opta.html
PLATFORM: ST STM32 (16.1.0) > Arduino Opta
HARDWARE: STM32H747XIH6 480MHz, 511.35KB RAM, 768KB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduino-mbed @ 4.0.2 
 - tool-dfuutil @ 1.11.0 
 - tool-openocd @ 3.1200.0 (12.0) 
 - tool-stm32duino @ 1.0.1 
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio/build/opta/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  10.9% (used 57152 bytes from 523624 bytes)
Flash: [=         ]  14.0% (used 109760 bytes from 786432 bytes)
Building .pio/build/opta/firmware.bin
Adding dfu suffix to firmware.bin
dyld[2890]: Library not loaded: /usr/local/opt/libusb/lib/libusb-1.0.0.dylib
  Referenced from: <2C64C116-83B7-3F20-B5BF-951356492177> /Users/ken.martin/.platformio/packages/tool-dfuutil/bin/dfu-suffix
  Reason: tried: '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/lib/libusb-1.0.0.dylib' (no such file), '/usr/lib/libusb-1.0.0.dylib' (no such file, not in dyld cache)
*** [.pio/build/opta/firmware.bin] Error -6
========================================================================== [FAILED] Took 0.51 seconds ==========================================================================

 *  The terminal process "platformio 'run', '--target', 'upload', '--upload-port', '/dev/cu.usbmodem1101'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

I did try `brew install libusb` but it made no difference.

The sketch is simply...

```cpp
#include <Arduino.h>

/**
  Getting Started with Opta™
  Name: LED_Blink_Opta
  Purpose: Blink STATUS LEDs on Opta™.

  @author Arduino
*/

void setup() {
  pinMode(LED_D0, OUTPUT);
  pinMode(LED_D1, OUTPUT);
  pinMode(LED_D2, OUTPUT);
  pinMode(LED_D3, OUTPUT);
}

void loop() {
  digitalWrite(LED_D0, HIGH);
  delay(100);
  digitalWrite(LED_D0, LOW);
  delay(100);
  
  digitalWrite(LED_D1, HIGH);
  delay(100);
  digitalWrite(LED_D1, LOW);
  delay(100);
  
  digitalWrite(LED_D2, HIGH);
  delay(100);
  digitalWrite(LED_D2, LOW);
  delay(100);
  
  digitalWrite(LED_D3, HIGH);
  delay(100);
  digitalWrite(LED_D3, LOW);
  delay(500);
}
```

Thoughts?

papakpmartin, 2023-09-18:
```bash
opta-test % pio pkg install -g --tool "platformio/tool-dfuutil@^1.11.0"
Tool Manager: tool-dfuutil@1.11.0 is already installed
```

papakpmartin, 2023-09-18:
On a whim...

```sh
opta-test % pio pkg install -g --tool "platformio/tool-libusb"
Tool Manager: Installing platformio/tool-libusb
UnknownPackageError: Could not find the package with 'platformio/tool-libusb' requirements for your system 'darwin_arm64'
```

papakpmartin, 2023-09-18:
From "Verbose Upload"...

```
...a bunch of buildy stuff, then...
MethodWrapper(["checkprogsize"], [".pio/build/opta/firmware.elf"])
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  10.9% (used 57152 bytes from 523624 bytes)
Flash: [=         ]  14.0% (used 109760 bytes from 786432 bytes)
.pio/build/opta/firmware.elf  :
section             size        addr
.text             105648   134479872
.ARM.exidx             8   134585520
.data               4104   603979776
.uninitialized        24   603983880
.bss               53048   603983904
.heap             466088   604036952
.lwip_sec         278528   805306368
.ARM.attributes       44           0
.comment             156           0
.debug_frame        4188           0
.stab                 60           0
.stabstr             118           0
Total             912014
arm-none-eabi-objcopy -O binary .pio/build/opta/firmware.elf .pio/build/opta/firmware.bin
"/Users/me/.platformio/packages/tool-dfuutil/bin/dfu-suffix" -v 0x2341 -p 0x0364 -d 0xffff -a .pio/build/opta/firmware.bin
dyld[7101]: Library not loaded: /usr/local/opt/libusb/lib/libusb-1.0.0.dylib
  Referenced from: <2C64C116-83B7-3F20-B5BF-951356492177> /Users/ken.martin/.platformio/packages/tool-dfuutil/bin/dfu-suffix
  Reason: tried: '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/lib/libusb-1.0.0.dylib' (no such file), '/usr/lib/libusb-1.0.0.dylib' (no such file, not in dyld cache)
*** [.pio/build/opta/firmware.bin] Error -6
=================================================================================== [FAILED] Took 5.49 seconds ===================================================================================
```

papakpmartin, 2023-09-19:
Tried manually config/make/installing libusb and I get...

```
'/usr/local/lib/libusb-1.0.0.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')
```

papakpmartin, 2023-09-19:
Also tried coping the `libusb` files created by homebrew into the expected `/usr/local/lib/` location... same result as above.

That's about all the thrashing I can think of right now. :)

valeros, 2023-09-21:
Hi @papakpmartin, I've recently pushed a possible fix for your issue. Could you please the upstream version of the development platform (a Git client is required) ?

```ini
[env:opta]
platform = https://github.com/platformio/platform-ststm32.git
board = opta
framework = arduino
```

papakpmartin, 2023-09-21:
Amazing... yes I will try this today

papakpmartin, 2023-09-21:
It worked!

I had tried going upstream at one point, so I did Uninstall the production and the upstream versions, then did a Full Clean (just to be sure). Then it built and uploaded!

FWIW, I realized this morning that the error is in the build, not upload, process. So presumably this is detectable/testable even if you don't have a Opta on hand.

Thanks so much for getting this figured out, @valeros!

I will not close this because I don't know how you want to handle that, project-wise.

noisymime, 2023-10-12:
@valeros Is this something that needs to be applied to other boards as well? I'm seeing exactly the same errors with `black_f407ve` and using `platform = https://github.com/platformio/platform-ststm32.git` in the env for that doesn't resolve the issue. 

It looks like `tool-dfuutil` is still providing x86_64 binaries on M1/M2 systems, which cannot load the arm64 version of `libusb-1.0.0.dylib` that is installed with brew.

adgbu, 2023-10-12:
I believe I have the same issue, except with the RP2040 microcontroller. My combination is:
macOS arm64 (13.5.2), VSCode (1.83.1), PlatformIO (Core 6.1.11, Home 3.4.4), Arduino framework, RP2040.
I'm trying to program the Raspberry Pi Pico W using the Raspberry Pi Debug Probe (picoprobe).

It appears as if the compilation works, but on Upload I get same error message as OP:
```
dyld[51634]: Library not loaded: /usr/local/opt/libusb/lib/libusb-1.0.0.dylib
```

Config in platformio.ini
```
[env:pico-debug-tool]
framework = arduino
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = rpipicow
board_build.core = earlephilhower
; Upload and debug over picoprobe
upload_protocol = picoprobe
;alt. upload_protocol = cmsis-dap
debug_tool = picoprobe
;alt. debug_tool = cmsis-dap
build_type = debug
```

I'm a bit slow/confused about what the solution was for @papakpmartin. Could it be similar for the platform in my case?

noisymime, 2023-10-12:
> I believe I have the same issue, except with the RP2040 microcontroller. 

This issue is specific to the stm32 platform using tool-dfuutil. You'll need to raise an issue for the raspberrypi platform: https://github.com/platformio/platform-raspberrypi/issues