platformio/platform-ststm32 #730: missing board Nucleo STM32H563ZI

dev-passion76, 2023-09-22:
hello

i need to work on vscode platformio with the board STM32H563ZI , but this board is missing.
do you think is possible to have this board or an update is it programmed

thank for your help

sstaub, 2023-09-29:
This board is also a part of the latest STM32Duino core

dev-passion76, 2023-10-06:
Thank for you your answer but I do not understand how to setup VSCode and have this board in Platformio available 
have a good day

sstaub, 2023-10-06:
The board should part of the latest STM32Duino core but it is missing. The support of third party boards and cores in PIo is horrible.

dev-passion76, 2023-10-06:
ok thank i wait and see

mrrotzi, 2023-10-09:
support for the NUCLEO-H563ZI board in PlatformIO would be great!

dev-passion76, 2023-10-09:
hello thanks 
need an update ??

[image]

no board nucleo H563ZI available

have a good day

MikeAelbrecht, 2024-02-21:
By chance any update on this?

PR #800 (2025-02-17): Add support to nucleo h563zi
Fix #730
commit e0bb3737 (2025-02-17): Add support for Nucleo H563ZI (#800)

Also added SVD files for H503, H562, H563, H573

Resolves #730
closed 2025-02-17 by valeros