platformio/platform-ststm32 #731: Cannot build using STM32 platform on aarch64

vskh, 2023-10-01:
Hello team,

First of all, thank you for the amazing tool and ecosystem for embedded programming, I enjoy it a lot.

However, I spotted the following problem: trying to build a simple example program for STM32-based board on an ARM-based SOC  fails.

**Target board:** MxChip AZ3166
**Build host:** Rock Pi 4C+ `Linux 6.1.11-rockchip64 aarch64`
**Cause:** As far as I understand the contents of 'platform.json', the version of _toolchain-gccarmnoneeabi_ it depends by default (`"version": ">=1.60301.0,<1.80000.0"` resolves to '1.60301') does not seem to be compiled for linux_aarch64.

With the following setup in platformio.ini:
```
[env:mxchip_az3166]
platform = ststm32
board = mxchip_az3166
framework = arduino
```
configuration of the project fails with
```
Processing mxchip_az3166 (platform: ststm32; board: mxchip_az3166; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing platformio/toolchain-gccarmnoneeabi @ ~1.60301.0
UnknownPackageError: Could not find the package with 'platformio/toolchain-gccarmnoneeabi @ ~1.60301.0' requirements for your system 'linux_aarch64'
```

If I try to explicitly change it to the version of package which I experimentally found to be available for linux_aarch64 with
```
 platform_packages =
     toolchain-gccarmnoneeabi@~1.120301.0
```
it installs, however, the build of the project fails because of many compilation errors, e.g.:
```
In file included from .platformio/packages/framework-arduinostm32mxchip/system/azure-iot-sdk-c/deps/azure-macro-utils-c/inc/azure_macro_utils/macro_utils.h:14,
                 from .platformio/packages/framework-arduinostm32mxchip/system/azure-iot-sdk-c/deps/umock-c/inc/umock_c/umock_c_prod.h:4,
                 from .platformio/packages/framework-arduinostm32mxchip/system/azure-iot-sdk-c/c-utility/inc/azure_c_shared_utility/strings.h:13,
                 from .platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/string.h:24,
                 from .platformio/packages/framework-arduinostm32mxchip/cores/arduino/Arduino.h:26,
                 from src/main.cpp:1:
.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/12.3.1/cstring:77:11: error: 'memchr' has not been declared in '::'
   77 |   using ::memchr;
      |           ^~~~~~
.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/12.3.1/cstring:78:11: error: 'memcmp' has not been declared in '::'
   78 |   using ::memcmp;
      |           ^~~~~~
.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/12.3.1/cstring:79:11: error: 'memcpy' has not been declared in '::'
   79 |   using ::memcpy;

```

I am not sure if this is a problem with the toolchain package or some environment/board/configuration issue. Could someone advice?

Maybe, I somehow can use the native toolchain available on the system instead?

valeros, 2023-10-02:
Hi @vskh, the problem is in the Arduino framework package for this hardware. I believe it's simply not compatible with the latest GCC packages and there were no ARM64 toolchains for this board when it was released. I also assume that the vendor of this hardware has no intentions of maintaining the software stack for this board as their repository has been archived https://github.com/microsoft/devkit-sdk

closed 2023-10-02 by valeros

vskh, 2023-10-08:
hi @valeros ! Thank you for the explanation. I have tried to use stm32cube instead and it seems to work.