platformio/platform-ststm32 #737: Missing support for STM32H723VGT6 board (including custom json attempt)

maarten1C96, 2023-11-06:
While PlatformIO currently supports the WeAct STM32F411CEU6 (aka 'black pill'), the WeAct STM32H723VGT6 is not being supported yet. The first comes with a 100 MHz CPU, but since the second comes with a 550 MHz CPU my curiosity recently got drawn to the latter. The F411 is working works pretty well in PlatformIO, but would it not be good to support the H723 as well?

Although not being experienced in the following, I tried to learn from the STM32duino-, STM32CubeIDE- and Zephyr–, and the `blackpill_f411ce.json`, `genericSTM32F411CC.json` and `weact_mini_h743vitx.json` files. The result is the following, but yet not successful:

```
{
  "build": {
    "arduino": {
      "variant_h": "variant_generic.h"
    },
    "core": "stm32",
    "cpu": "cortex-m7",
    "extra_flags": "-Dstm32h7xx -Dstm32h723xx",
    "f_cpu": "550000000L",
    "framework_extra_flags": {
      "arduino": "-DCUSTOM_PERIPHERAL_PINS"
    },
    "mcu": "stm32h723zgt6",
    "product_line": "STM32H723xx",
    "variant": "STM32H7xx/H723V(E-G)(H-T)_H730VB(H-T)_H733VG(H-T)",
    "zephyr": {
      "variant": "stm32h723xx"
    }
  },
  "debug": {
    "jlink_device": "STM32H723VG",
    "openocd_target": "stm32f7x",
    "svd_path": "STM32F723.svd"
  },
  "frameworks": [
    "arduino",
    "cmsis",
    "stm32cube",
    "libopencm3",
    "zephyr"
  ],
  "name": "WeAct Studio STM32H723VGT6",
  "upload": {
    "maximum_ram_size": 577536,
    "maximum_size": 1048576,
    "protocol": "stlink",
    "protocols": [
      "jlink",
      "cmsis-dap",
      "stlink",
      "blackmagic",
      "serial",
      "dfu"
    ]
  },
  "url": "https://github.com/WeActStudio/WeActStudio.MiniSTM32H723",
  "vendor": "WeAct Studio"
}

```

gdiciocco, 2024-01-03:
another bump more

vivichrist, 2024-01-04:
I think `"openocd_target": "stm32f7x",` should be `"openocd_target": "stm32h7x",`.

IcaroMendes00, 2024-03-22:
Can you share the full project? or at least the .ini archive?

IcaroMendes00, 2024-03-24:
How you made the platfomio recognize the board archive? I followed the guides, but doesnt work

vivichrist, 2024-04-17:
I did some OpenOCD jiggery pokery that I can't quite remember. It compiled but I couldn't verify if the uploaded program actually worked.