platformio/platform-ststm32 #742: Escaping pharentesis cause error at using Arduino_Core_STM32 2.7.0

asukiaaa, 2023-11-23:
I want to use version 2.7.0 of Arduino_Core_STM32.that was released on Nov 17, 2023.
I tried the following configuration but the path of variant  on .vscode/c_cpp_properties.json is escaped so I cannot build with 2.7.0.

Is there any way to resolve this error?

platformio.ini
```
[env:nucleo_f429zi]
board = nucleo_f429zi
platform = https://github.com/platformio/platform-ststm32.git
platform_packages =
  framework-arduinoststm32 @ https://github.com/stm32duino/Arduino_Core_STM32.git#2.7.0
```

The path of variant on `.vscode/c_cpp_properties.json`.
 ```
                "/home/asuki/.platformio/packages/framework-arduinoststm32@src-b50b2b5a3a566d56a33fa1fea8083459/variants/STM32F4xx/F427Z\(G-I\)T_F429ZET_F429Z\(G-I\)\(T-Y\)_F437Z\(G-I\)T_F439Z\(G-I\)\(T-Y\)",
```

I could building with replasing `\(` as `\` and `\)` ad `)` on the path.

```
                "/home/asuki/.platformio/packages/framework-arduinoststm32@src-b50b2b5a3a566d56a33fa1fea8083459/variants/STM32F4xx/F427Z(G-I)T_F429ZET_F429Z(G-I)(T-Y)_F437Z(G-I)T_F439Z(G-I)(T-Y)",
```

The following change on Arduino_Core_STM32 seems cause this error.
https://github.com/stm32duino/Arduino_Core_STM32/commit/feba4321e0b6b566cf9c1ba5d4765d28babef357#diff-4cc34091d0e66b4fc7efa0c22346a26f99eec780038a13a6c5b694b0b02219c6R58

My environment
OS: ubuntu22.04
PlatformIO: 6.1.11

valeros, 2023-11-23:
Hi @asukiaaa, v2.7.0 is not officially supported yet, but I guess you can copy the build script from the PlatformIO package v2.6.0 that has this issue fixed.

asukiaaa, 2023-11-23:
Thank you for the response.

> v2.7.0 is not officially supported yet,

Do you know when will it be?

> I guess you can copy the build script from the PlatformIO package v2.6.0 that has this issue fixed.

I see.
I use my own branch of arduino core which change the file until this repository supports v2.7.0.

asukiaaa, 2023-12-07:
I could use Arduino_Core_STM32 v2.7.1 with using platformio-ststm32 v17.0.1.
Thank you.

asukiaaa, 2023-12-07:
Sorry, I could build but stil same issue for .vscode/c_cpp_properties.json

commit e505a30e (2023-12-18): Update Arduino core to v2.7.1

Resolves #742
closed 2023-12-18 by valeros