platformio/platform-ststm32 #754: Missing several chips/boards

Levi--G, 2024-01-28:
Hi,

I encountered a small issue with trying to use the stm32g030 series (cortex M0+ mainstream line) which seem to not be present in platformio? They are supported by cube and stm32duino as far as i know but i think they lack the definition. while i was looking for it i noticed alot of different chips are also missing like G4 some F3 series and the entire L series? is that correct or am i mistaken? Its kind of odd since i think they just lack the board definition but i have no experience how to set that up. I know how the chips and the cores work, but platformio internals are a mystery for me. ;)

Is it possible to add at least the G030 series please? Or point me to a guide for absolute dummies how to add it myself (but chances are i won't be able to)

Thank you

Levi--G, 2024-01-30:
I have found this topic, is this still a valid board json? https://community.platformio.org/t/stm32-g030f6-board-config/24560/10

positron96, 2024-05-13:
Hi. For my project that uses STM32G030 I've added a board to my project itself (added `boards` folder and a board definitions file inside it),

<details><summary>boards\generic_stm32g030f6px.json</summary>
<p>

```
{
    "build": {
      "core": "stm32",
      "cpu": "cortex-m0",
      "extra_flags": "-DSTM32G0xx -DSTM32G030xx",
      "f_cpu": "64000000L",
      "framework_extra_flags": {
        "arduino": "-D__CORTEX_SC=0"
      },
      "mcu": "stm32g030f6",
      "product_line": "STM32G030xx",
      "variant": "STM32G0xx/G030F6P"
    },
    "debug": {
      "default_tools": [
        "stlink"
      ],
      "jlink_device": "STM32G030F6",
      "onboard_tools": [
        "stlink"
      ],
      "openocd_target": "stm32g0x",
      "svd_path": "STM32G030.svd"
    },
    "frameworks": [
      "arduino",
      "cmsis",
      "stm32cube"
    ],
    "name": "Generic STM32G030F6P6",
    "upload": {
      "maximum_ram_size": 8192,
      "maximum_size": 32768,
      "protocol": "stlink",
      "protocols": [
        "blackmagic",
        "cmsis-dap",
        "dfu",
        "jlink",
        "serial",
        "stlink",
        "mbed"
      ]
    },
    "url": "https://www.aliexpress.com/item/4000012148114.html",
    "vendor": "ST"
  }
```

</p>
</details> 

Than I added settings to platformio.ini. Not sure if all of that is needed, it could be that only `board=` line is needed, and the rest is due to me using LL instead of HAL.

<details><summary>platformio.ini</summary>
<p>

```
[platformio]
default_envs = devboard

[env]
platform = ststm32@~17
framework = stm32cube

build_flags =
  -DUSE_FULL_LL_DRIVER
  -DHSE_VALUE=8000000
  -DHSE_STARTUP_TIMEOUT=100
  -DLSE_STARTUP_TIMEOUT=5000
  -DLSE_VALUE=32768
  -DEXTERNAL_CLOCK_VALUE=12288000
  -DHSI_VALUE=16000000
  -DLSI_VALUE=32000

[env:devboard]
board = generic_stm32g030f6px

```

</p>
</details>

Levi--G, 2024-05-14:
Thanks for the info, but i need a g030c8t6 board file somehow, maybe ill try adapting yours. If i eventually get it to work ill post back. :)