platformio/platform-ststm32 #756: After update the ststm32 to the newest, something was wrong  about VECT_TAB_OFFSET?

KiraVerSace, 2024-02-27:
Hello, I use bootloader + application in my stm32l476rg, so 120KB was used for bootloader, and the reset used for application.

At last, I changed 2 places:
1. Modified the `ldscript.ld` file
    Add `LD_FLASH_OFFSET = 0x00020000;`
   and modified

```
/* Memories definition */
MEMORY
{
  RAM     (xrw)     : ORIGIN = 0x20000000,   LENGTH = LD_MAX_DATA_SIZE
  RAM2    (xrw)     : ORIGIN = 0x10000000,   LENGTH = 32K
  FLASH   (rx)      : ORIGIN = 0x8000000 + LD_FLASH_OFFSET, LENGTH = LD_MAX_SIZE - LD_FLASH_OFFSET
}
```
2. Modified the `platform.ini` file 
Add (this for the interrupt offset)

```
build_flags =
     -DVECT_TAB_OFFSET=0x00020000
```
and (this for jlink download)

`board_upload.offset_address = 0x08020000`

And all the system worked well, after I update the PIO, when I complie the program, too many warnings like:
[image]
ignore the warnings and the application can not start from the bootloader.

I have no idea, where should I change the value of VECT_TAB_OFFSET,
At last I delete the 

```
build_flags =
     -DVECT_TAB_OFFSET=0x00020000
```
and add 
`#define VECT_TAB_OFFSET 0x00020000` in the `system_stm32l4xx.c`  file

there also a warning, but works well
[image]

Could you help me?  Thank you !

valeros, 2024-02-27:
Hi @KiraVerSace, starting from Arduino core v2.7.0 the `VECT_TAB_OFFSET` macro is a default flag. You can adjust its value directly in your platformio.ini:
```
board_build.flash_offset = 0xYOURVALUE
```

This value will be assigned to `VECT_TAB_OFFSET` and `-Wl,--defsym=LD_FLASH_OFFSET` flags.

fpistm, 2024-02-27:
Note: `VECT_TAB_OFFSET` can be redefined since a while. It is redefined thanks a flags mainly when a bootloader is used. Nothing related with the 2.7.0 version.

KiraVerSace, 2024-02-27:
> Hi @KiraVerSace, starting from Arduino core v2.7.0 the `VECT_TAB_OFFSET` macro is a default flag. You can adjust its value directly in your platformio.ini:
> 
> ```
> board_build.flash_offset = 0xYOURVALUE
> ```
> 
> This value will be assigned to `VECT_TAB_OFFSET` and `-Wl,--defsym=LD_FLASH_OFFSET` flags.

Thank you for your help, I will try use board_build.flash_offset tomorrow,

but I don’t know what do you mean about the - -defsym
, what the relationship between them?

KiraVerSace, 2024-02-27:
> Note: `VECT_TAB_OFFSET` can be redefined since a while. It is redefined thanks a flags mainly when a bootloader is used. Nothing related with the 2.7.0 version.

Thank you for  your tips
But I really don’t know what do you mean, I use bootloader written by myself.
In my application I use the VECT_TAB have nothing to do with the bootloader,  they are independent

KiraVerSace, 2024-02-27:
> Hi @KiraVerSace, starting from Arduino core v2.7.0 the `VECT_TAB_OFFSET` macro is a default flag. You can adjust its value directly in your platformio.ini:
> 
> ```
> board_build.flash_offset = 0xYOURVALUE
> ```
> 
> This value will be assigned to `VECT_TAB_OFFSET` and `-Wl,--defsym=LD_FLASH_OFFSET` flags.

I also want to know how do you know this, where is the help documentation  which mentioned it?

RedEchidnaUK, 2024-03-30:
I can't seem to get `VECT_TAB_OFFSET` to work at all in the STMduino environment. This is my platform.ini

```
[env:bluepill_f103c8_128k]
platform = ststm32
board = bluepill_f103c8_128k
framework = arduino
upload_protocol = stlink
debug_tool = stlink
upload_flags = -c set CPUTAPID 0x2ba01477 ; ID for F103 clone
build_flags = 
 	-D VECT_TAB_OFFSET=0x08003000 ; Set Vector Tab offset, doesn't work
board_upload.offset_address = 0x08003000 ; Set flash offset
```
I've also tried adding,

```board_build.flash_offset = 0x0800300```

I'm also not sure if I'm supposed to put in the full memory location or just the hex offset, but I've tired both with no success. 

For now, the only way I've managed to get it to work is to manually edit the offset in`platformio-build.py`, but then you get lots of warnings.

fpistm, 2024-03-30:
You talk about offset. So I guess it should be:
```
build_flags = 
board_upload.offset_address = 0x00003000
```

RedEchidnaUK, 2024-03-30:
Unfortunately that doesn't work either, or at least it doesn't run. Did you mean `board_upload.offset_address` or `board_build.flash_offset`? Although, I've tried both and neither work.

fpistm, 2024-03-30:
I don't know. I do not use pio. Only comment to point that here it talk about offset not base address + offset. 😉

xiezhoubin, 2024-06-06:
> Hello, I use bootloader + application in my stm32l476rg, so 120KB was used for bootloader, and the reset used for application.
>
> At last, I changed 2 places:
>
> 1. Modified the `ldscript.ld` file [...]

你好，我也在使用arduino软件框架开发stm32，在Platformio中编写程序。目前在编写IAP程序时遇到了一点问题，bootloader程序正常运行，但是app程序始终无法跳转运行成功(用STM32CUBE IDE编译的app程序就没问题)。方便留个微信联系方式交流下吗？我的微信是：XZB1436056045