platformio/platform-ststm32 #757: Broken config for ST Nucleo L552ZE-Q? - can't create project with it, unknown board

CZdigger146, 2024-02-27:
Hello, I have two nucleo boards available - Nucleo L4R5ZI, which works fine with platformio, and Nucleo L552ZE-Q.
When creating a project, i can select both boards just fine but only L55 throws an error and refuses to create a project. Only way forward was creating the project for the L4 and modifying stuff.

how to reproduce issue: create a project using the Nucleo L552ZE-Q on stm32cube framework (arduino might not work either)

I eventually got the L55 working, but i had to do a lot of workarounds, most useful was modifying platformio.ini to update openOCD and stmCube: 
`[env:nucleo_l552ze_q]
platform = ststm32
board = nucleo_l552ze_q
board_build.mcu = stm32l552zet6
board_build.f_cpu = 80000000L
framework = stm32cube
platform_packages = platformio/framework-stm32cubel5@^1.4.0, platformio/tool-openocd@^3.1200.0`

After all that I don't remember if i downloaded anything extra that is also required to get it working, maybe I found a L55 json somewhere on this github to also make it work? I don't remember exactly my steps, sorry.

Also noteworthy is that this is on a basically fresh install of platformio, so I'm confident I didn't change anything that would break the L55.

I'd appreciate if anyone who understands platformio would take a look at this and possibly fix the L55 so it works out of the box when creating a project like with the L4, and newbies like me don't have to give up on platformio with this board, thank you.

valeros, 2024-02-27:
Hi @CZdigger146, what was the error exactly?

CZdigger146, 2024-02-27:
So, I'm gonna try and create a new project on another machine where i didn't do any of the things I did:

1. try and create a project like this:
[image]

2. an error pops up:
[image]

`PIO Core Call Error: "The following files/directories have been created in C:\\Users\\misas\\Documents\\PlatformIO\\Projects\\nucleo L5 test\r\ninclude - Put project header files here\r\nlib - Put project specific (private) libraries here\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\nResolving nucleo_l552ze_q dependencies...\r\n\n\nUnknownBoard: Unknown board ID 'nucleo_l552ze_q'"`

no idea what to do with that...

valeros, 2024-02-28:
You're using an outdated version of the `ststm32` development platform, just update to the latest and it will work.

closed 2024-02-28 by valeros

CZdigger146, 2024-03-01:
The thing is i already updated everything in the troubleshooting process and it didn't help. So i tried complete reinstall of STM32 platform and it now works fine. No clue what went wrong on the first install especially as two separate computers had the same issue but yeah, it works fine now on both of them...