platformio/platform-ststm32 #758: STM32H5 Support?

samspencer5991, 2024-03-04:
Unless I've missed something, there is no mention of support for STM32H5 series chips. Is there a reason for this or has it simply not been a priority? I've got a couple of boards using STM32H5 chips I'd like to get started with the bring up for. If not officially supported, can someone point me in the best direction to integrate these chips with a new platformio project?

samspencer5991, 2024-03-06:
I have downloaded the firmware package from ST, duplicated an existing framework folder (STM32G4 for testing), renamed everything accordingly and copied the H5 files and drivers in place of the G4 ones, and immediately errors were thrown when trying to overwrite the framework and explicitly use my custom H5 (referenced as a GitHub link as per the PIO reference docs).

samspencer5991, 2024-03-08:
@valeros are you able to point me in the right direction to create my own STM32H5 if required? I'd really like to get up and running with this chip.

samspencer5991, 2024-03-08:
@valeros hmm ok, I'll give that a crack. Is there a particular reason that MCU series hasn't been added to PIO? Happy to get it working on a fork and look at a PR to help others out. Just seems like a strange omission for what is a very powerful and cost effective STM32 series.

valeros, 2024-03-08:
@samspencer5991 It's simply not a priority, the team has plenty of other tasks to deal with.

samspencer5991, 2024-03-10:
@valeros I've followed everything as best as I can, running into a specific issue. Here are the steps taken:

1. Downloaded the STM32H5 repo from ST, ensured that the folder structure matches an existing package file (I used the STM32G4 as a reference as I have mature, known working projects on it using PIO/STM32Cube).

2. Added a package.json file to the folder identical to the STM32G4 but with the correct naming changes (I bumped the version to 1.5.1 in case that was required):
`{
  "description": "STM32Cube is a set of tools and embedded software bricks available free of charge to enable fast and easy development on the STM32 platform (STM32CubeH5 MCU Firmware Package)",
  "keywords": [
    "framework",
    "hal",
    "stm32",
    "st"
  ],
  "name": "framework-stm32cubeh5",
  "version": "1.5.1",
  "homepage": "https://github.com/STMicroelectronics/STM32CubeH5"
}`

3. I added this folder local folder:  .platformio\packages (running Windows 10)

4. In .platformio\platforms\ststm32 I modified the platform.json file to include the relevant entry:
`"framework-cmsis-stm32h5": {
      "type": "framework",
      "optional": true,
      "owner": "platformio",
      "version": "~1.5.1"
    }`
    
5. I created a custom board file (I already have done this for specific G0 and G4 variants, which both work correctly, so I utilised the template from them). I added this in the project folder under the 'boards' folder, and also tried creating this file under .platformio\platforms\ststm32\boards with exactly the same result when trying to build the project.

The only thing I haven't done yet is handling the svd and ld script files, however, the issue I'm facing seems to be occuring before that. The error output from the terminal is:
`KeyError: Traceback (most recent call last):
  File "C:\Users\samsp\.platformio\penv\Lib\site-packages\platformio\__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
    ^^^^^
  File "C:\Users\samsp\.platformio\penv\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\samsp\.platformio\penv\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\samsp\.platformio\penv\Lib\site-packages\platformio\cli.py", line 85, in invoke
    return super().invoke(ctx)
           ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\samsp\.platformio\penv\Lib\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\samsp\.platformio\penv\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\samsp\.platformio\penv\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\samsp\.platformio\penv\Lib\site-packages\click\decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\samsp\.platformio\penv\Lib\site-packages\platformio\run\cli.py", line 145, in cli
    process_env(
  File "C:\Users\samsp\.platformio\penv\Lib\site-packages\platformio\run\cli.py", line 208, in process_env
    ).process()
      ^^^^^^^^^
  File "C:\Users\samsp\.platformio\penv\Lib\site-packages\platformio\run\processor.py", line 81, in process
    install_project_env_dependencies(
  File "C:\Users\samsp\.platformio\penv\Lib\site-packages\platformio\package\commands\install.py", line 132, in install_project_env_dependencies
    _install_project_env_platform(project_env, options),
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\samsp\.platformio\penv\Lib\site-packages\platformio\package\commands\install.py", line 149, in _install_project_env_platform
    PlatformPackageManager().install(
  File "C:\Users\samsp\.platformio\penv\Lib\site-packages\platformio\package\manager\platform.py", line 60, in install
    p.configure_project_packages(project_env, project_targets)
  File "C:\Users\samsp\.platformio\penv\Lib\site-packages\platformio\platform\base.py", line 185, in configure_project_packages
    self.configure_default_packages(options, targets or [])
  File "C:\Users\samsp\.platformio\platforms\ststm32\platform.py", line 111, in configure_default_packages
    return PlatformBase.configure_default_packages(self, variables,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\samsp\.platformio\penv\Lib\site-packages\platformio\platform\base.py", line 197, in configure_default_packages
    self.packages[_pkg_name]["optional"] = False
    ~~~~~~~~~~~~~^^^^^^^^^^^
KeyError: 'framework-stm32cubeH5'`

I have created a public repo with the framework-stm32cubeh5 folder both to try referencing it in my platformio.ini file using platform_packages and also as a reference here: https://github.com/samspencer5991/framework-stm32cubeh5/blob/main/README.md

Happy to do the legwork to get it running, but the cross-referencing of various documentation without much in the way of previous examples to go off is tricky. I have made all these changes on my local build rather than forking as a test.

valeros, 2024-03-11:
You're working with the `framework-stm32cubeh5` package, but in `platform.json` you've added `framework-cmsis-stm32h5` which is a different one.

> The only thing I haven't done yet is handling the svd and ld script files

The SVD file in located in the `misc` folder in this repo, so just add necessary files to this folder in your fork. Linker script files live in a separate package `tool-ldscripts-ststm32`. You can add your files there locally, but to make your project portable, I'd recommend keeping an appropriate linker script in your project folder and just specify it directly in your `platformio.ini` via the `board_build.ldscript = example.ld` option.

samspencer5991, 2024-03-12:
@valeros thank you for your help. I found some of the steps to be a little tricky to work out but got there in the end with a successful compilation. One problem I found is that I couldn't add the files locally, it would throw up the following error: `platformio requirements for your system 'windows_amd64'`
However, with the exact same folder structure for the package, referencing my GH repo for it worked well using the `platform_packages` option. It's not a deal breaker and does make it more portable, but is there anything in particular that needs to be done to add the package files locally?

Having successfully gotten compilation, what can I do to provide everything needed for you to easily merge this into the repo so others can get up and running with the STM32H5 platform?

valeros, 2024-03-12:
> platformio requirements for your system 'windows_amd64'

Since you added all packages manually, you also need the `.piopm` file (with appropriate changes) near `platform.json` (note that .piopm shouldn't be in your repository, as this file is handled by PlatformIO automatically when a new package is installed)

> platform_packages

You don't need the `platform_packages` option if you set a URL to your Git repo directly in the `platform.json` file:

```json
"framework-cmsis-stm32h5": {
      "type": "framework",
      "optional": true,
      "owner": "platformio",
      "version": "https://github.com/samspencer5991/framework-stm32cubeh5.git#3fc9e3e"
}
```
Where `3fc9e3e` is the hash of the revision of your repository you want to install.

> Having successfully gotten compilation, what can I do to provide everything needed for you to easily merge this into the repo so others can get up and running with the STM32H5 platform?

It's a bit more complex mater than just a PR. But still, you can share here an URL to your fork, so anyone who needs support for H5 will be able to install it by specifying your fork as the value for the `platform` option.

samspencer5991, 2024-03-25:
Still a bit of work to do before I would consider it a useable experience by anyone else or even myself @valeros 

I've been able to compile code currently, but have not uploaded the build artefacts at all. Initial board bring-up and testing has been done with STM32CubeIDE successfully.

I'm running into some roadblocks now trying to upload/debug on the H5. I have downloaded the stm32h5x.cfg file from the STM32 OpenOCD github repo (https://github.com/STMicroelectronics/OpenOCD) and added that into the following folder: `\.platformio\packages\tool-openocd\openocd\scripts\target`

When trying to upload using a stlink (confirmed working in stm32cubeide, the following error appears:

`CURRENT: upload_protocol = stlink
Uploading .pio\build\bridge8-v2-x-0\bridge8_v2.0.0.0-beta.49.elf
xPack Open On-Chip Debugger 0.12.0-01004-g9ea7f3d64-dirty (2023-01-30-15:04)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

hla_swd
Error: The selected transport doesn't support this target
C:\Users\samsp\.platformio\packages\tool-openocd/openocd/scripts/target/stm32h5x.cfg:55: Error:
in procedure 'script' 
at file "embedded:startup.tcl", line 28
at file "C:\Users\samsp\.platformio\packages\tool-openocd/openocd/scripts/target/stm32h5x.cfg", line 55
*** [upload] Error 1
`

ypearson-bdai, 2024-10-02:
@samspencer5991 Did you get this working? I'm interested in the H563ZI support also

closed 2025-04-27 by samspencer5991