platformio/platform-ststm32 #761: nucleo F411RE zephyr not supported

albmef, 2024-03-15:
hi 

i get the following error while trying to configure a project using an ST nucleo F411RE board and zephyr RTOS framework. 
PIO core 6.1.13 home 3.4.4
same behavior with ubuntu 22.04 and zindozs 12

in short:
CMake Error at /home/alban/.platformio/packages/framework-zephyr/cmake/modules/extensions.cmake:2629 (message):\n No prj.conf file(s) was found in the\n /home/alban/Documents/PlatformIO/Projects/zephyr_F411_blinky/zephyr\n folder(s)

tested OK with zephyr and another board "nRF52840-DK". no need for prj.conf for this one.
tested OK with arduino and the ST nucleo F411RE board.
the F411RE + zephyr combo fails.

below the full error report
"""
PIO Core Call Error: "Resolving nucleo_f411re dependencies...\nAlready up-to-date.\nUpdating metadata for the vscode IDE...\n\n\nUserSideException: Processing nucleo_f411re (platform: ststm32; board: nucleo_f411re; framework: zephyr)\n--------------------------------------------------------------------------------\nVerbose mode can be enabled via `-v, --verbose` option\nCONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_f411re.html\nPLATFORM: ST STM32 (17.3.0) > ST Nucleo F411RE\nHARDWARE: STM32F411RET6 100MHz, 128KB RAM, 512KB Flash\nDEBUG: Current (stlink) On-board (stlink) External (blackmagic, cmsis-dap, jlink)\nPACKAGES: \n - framework-zephyr @ 2.30600.0 (3.6.0) \n - tool-cmake @ 3.21.3 \n - tool-dtc @ 1.4.7 \n - tool-gperf @ 3.0.4 \n - tool-ninja @ 1.7.1 \n - toolchain-gccarmnoneeabi @ 1.120301.0 (12.3.1)\nReading CMake configuration\n-- Application: /home/alban/Documents/PlatformIO/Projects/zephyr_F411_blinky/zephyr\n-- CMake version: 3.21.3\n-- Found Python3: /home/alban/.platformio/penv/.zephyr-3.6.0/bin/python (found suitable version \"3.10.12\", minimum required is \"3.8\") found components: Interpreter \n-- Cache files will be written to: /home/alban/.cache/zephyr\n-- Zephyr version: 3.6.0 (/home/alban/.platformio/packages/framework-zephyr), build: zephyr-v30600\n-- Board: nucleo_f411re\n-- Configuring incomplete, errors occurred!\n\nCMake Warning at /home/alban/.platformio/packages/framework-zephyr/cmake/app/boilerplate.cmake:20 (message):\n Loading of Zephyr boilerplate.cmake directly is deprecated, please use\n 'find_package(Zephyr)'\nCall Stack (most recent call first):\n CMakeLists.txt:2 (include)\n\n\nLoading Zephyr default modules (Zephyr base).\nCMake Deprecation Warning at /home/alban/.platformio/packages/framework-zephyr/cmake/modules/FindDeprecated.cmake:94 (message):\n 'PYTHON_PREFER' variable is deprecated. Please use Python3_EXECUTABLE\n instead.\nCall Stack (most recent call first):\n /home/alban/.platformio/packages/framework-zephyr/cmake/modules/python.cmake:16 (find_package)\n /home/alban/.platformio/packages/framework-zephyr/cmake/modules/zephyr_default.cmake:129 (include)\n /home/alban/.platformio/packages/framework-zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)\n /home/alban/.platformio/packages/framework-zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)\n /home/alban/.platformio/packages/framework-zephyr/cmake/app/boilerplate.cmake:25 (find_package)\n CMakeLists.txt:2 (include)\n\n\nCMake Error at /home/alban/.platformio/packages/framework-zephyr/cmake/modules/extensions.cmake:2629 (message):\n No prj.conf file(s) was found in the\n /home/alban/Documents/PlatformIO/Projects/zephyr_F411_blinky/zephyr\n folder(s), please read the Zephyr documentation on application development.\nCall Stack (most recent call first):\n /home/alban/.platformio/packages/framework-zephyr/cmake/modules/configuration_files.cmake:47 (zephyr_file)\n /home/alban/.platformio/packages/framework-zephyr/cmake/modules/zephyr_default.cmake:129 (include)\n /home/alban/.platformio/packages/framework-zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)\n /home/alban/.platformio/packages/framework-zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)\n /home/alban/.platformio/packages/framework-zephyr/cmake/app/boilerplate.cmake:25 (find_package)\n CMakeLists.txt:2 (include)\n\n\n========================== [FAILED] Took 0.62 seconds =========================="
"""

best regards,
AM

valeros, 2024-03-15:
Hi @albmef, nRF52840-DK and ST Nucleo F411RE use different versions of the Zephyr framework. In a nutshell, `prj.conf` files are mandatory in projects from Zephyr v3.4.0.

closed 2024-03-15 by valeros

albmef, 2024-03-15:
excuse me but,
why do the zephyr installer not put a default or empty prj.conf file into the zephyr directory instead of throwing an error ?

albmef, 2024-03-15:
ok i had to:
- put an empty prj.conf file into the zephyr project subdirectory
- replace #include <zephyr.h> by  #include <zephyr/kernel.h> in the main.c file
the code compiles and runs
importing the blinky from ~/.platformio/packages/framework-zephyr/samples works too

albmef, 2024-03-18:
sure i did
on ubuntu:
- pio pkg update -g -p ststm32 -> OK
- create new project with nucleo F411RE board + zephyr -> OK
- copy paste blinky code into main.c, build and upload, debug -> OK

on windows:
- pio pkg update -g -p ststm32 -> OK
- create new project with nucleo F411RE board + zephyr -> OK
- copy paste blinky code into main.c, build and upload, debug -> OK

thanks for your support