platformio/platform-ststm32 #768: stm32f030 flash issues

Oliver0804, 2024-04-15:
I downloaded using the Arduino IDE, and it successfully made the LED blink, just to confirm that my hardware is fine. However, when I use the Arduino framework under PlatformIO, it doesn't work properly, and I get logs like this.

Could it be that I'm missing some settings in the .ini file?

```
Processing demo_f030f4 (platform: ststm32; board: demo_f030f4; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/demo_f030f4.html
PLATFORM: ST STM32 (17.2.0) > Demo F030F4
HARDWARE: STM32F030F4P6 48MHz, 4KB RAM, 16KB Flash
DEBUG: Current (stlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 @ 4.20701.0 (2.7.1) 
 - framework-cmsis @ 2.50900.0 (5.9.0) 
 - tool-dfuutil @ 1.11.0 
 - tool-dfuutil-arduino @ 1.11.0 
 - tool-openocd @ 3.1200.0 (12.0) 
 - tool-stm32duino @ 1.0.1 
 - toolchain-gccarmnoneeabi @ 1.120301.0 (12.3.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 13 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio/build/demo_f030f4/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  29.6% (used 1212 bytes from 4096 bytes)
Flash: [=======   ]  67.2% (used 11008 bytes from 16384 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, jlink, serial, stlink
CURRENT: upload_protocol = stlink
Uploading .pio/build/demo_f030f4/firmware.bin
xPack Open On-Chip Debugger 0.12.0-01004-g9ea7f3d64-dirty (2023-01-30-17:05)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

hla_swd
[stm32f0x.cpu] halted due to debug-request, current mode: Thread 
xPSR: 0xc1000000 pc: 0x08001f98 msp: 0x20001000
** Programming Started **
Warn : Adding extra erase range, 0x08002bd8 .. 0x08002bff
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
shutdown command invoked
```

[image]
[image]
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:demo_f030f4]
platform = ststm32
board = demo_f030f4
framework = arduino
debug_tool = stlink

upload_protocol = stlink
board_upload.offset_address =0x08000000
```

[image]

valeros, 2024-04-15:
Hi @Oliver0804,

I see you're using an stlink probe to upload the firmware. Could you please try to lanuch a debugging session to figure out if the firmware evens starts.?

> board_upload.offset_address =0x08000000

I believe this line is unnecessary, this address is used by default.

Oliver0804, 2024-04-15:
Hi , @valeros Thank you for your response. 

Attached is the result of compiling with the settings I added. 

In fact, I didn't include them initially, but the LED still couldn't blink properly.

[image]

> Hi @Oliver0804,
> 
> I see you're using an stlink probe to upload the firmware. Could you please try to lanuch a debugging session to figure out if the firmware evens starts.?
> 
> > board_upload.offset_address =0x08000000
> 
[image]

> I believe this line is unnecessary, this address is used by default.

valeros, 2024-04-15:
So what about debugging, have you tried lanching a debugging session.

Oliver0804, 2024-04-15:
After entering Debug mode, I noticed that it stopped at:
[image]

```
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK) {
    Error_Handler();  <--stop here
  }
  ```
[image]

valeros, 2024-04-15:
In the Arduino IDE you selected a different board, so what's your board exactly? Is it STM32F030F4 Demo Board?

Oliver0804, 2024-04-15:
> In the Arduino IDE you selected a different board, so what's your board exactly? Is it STM32F030F4 Demo Board?

In the Arduino IDE, I chose the "Generic F030F4px" board, but in PlatformIO, the closest option available is the "demo_f030f4" board.

Oliver0804, 2024-04-15:
I noticed that the difference might be whether to use an external oscillator, but currently, the hardware I'm successfully running on Arduino IDE doesn't have an external oscillator.

valeros, 2024-04-15:
Unfortunately, the Generic F0 targets are not supported at the moment and the closest board doesn't mean it will work. As you correctly mentioned, there is probably a discrepancy in the clock configuration, as STM32F030F4 Demo Board indeed has an external oscillatior.

Oliver0804, 2024-04-15:
@valeros Thank you for your reply. 
However, I'm wondering if it's possible to work with the platform in ArduinoIDE. So, is the board manager used by PlatformIO different from STM32duino's?
https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json

Oliver0804, 2024-04-16:
I see, thanks, I'll close this issue first, thanks for your help!

closed 2024-04-16 by Oliver0804