platformio/platform-ststm32 #771: Compilation errors while using STM32Ethernet for genericSTM32F407VGT6 board

HareshPrajapati, 2024-04-24:
I am using latest `ststm32` and latest [STM32Ethernet](https://github.com/stm32duino/STM32Ethernet) for `genericSTM32F407VGT6 `
board.

below is `platformio.ini` file
```
[env:genericSTM32F407VGT6]
platform = ststm32
board = genericSTM32F407VGT6
monitor_port = COM12
monitor_speed = 115200
upload_protocol = stlink
build_flags = -DENABLE_HWSERIAL2
framework = arduino
lib_deps = https://github.com/stm32duino/STM32Ethernet.git
           https://github.com/stm32duino/LwIP.git
```
I also added `hal_conf_extra.h` file with `HAL_ETH_LEGACY_MODULE_ENABLED` define as suggested [here](https://github.com/stm32duino/Arduino_Core_STM32/issues/2003)

Still unable to compile project.

I am getting below errors while compiling 
```
Compiling .pio\build\genericSTM32F407VGT6\lib949\STM32duino LwIP\api\netifapi.c.o
Compiling .pio\build\genericSTM32F407VGT6\lib949\STM32duino LwIP\api\sockets.c.o
.pio\libdeps\genericSTM32F407VGT6\STM32Ethernet\src\utility\ethernetif.cpp: In function 'void low_level_init(netif*)':
.pio\libdeps\genericSTM32F407VGT6\STM32Ethernet\src\utility\ethernetif.cpp:170:18: error: 'struct ETH_InitTypeDef' has no member named 'AutoNegotiation'
  170 |   EthHandle.Init.AutoNegotiation = ETH_AUTONEGOTIATION_ENABLE;
      |                  ^~~~~~~~~~~~~~~
.pio\libdeps\genericSTM32F407VGT6\STM32Ethernet\src\utility\ethernetif.cpp:171:18: error: 'struct ETH_InitTypeDef' has no member named 'Speed'
  171 |   EthHandle.Init.Speed = ETH_SPEED_100M;
      |                  ^~~~~
.pio\libdeps\genericSTM32F407VGT6\STM32Ethernet\src\utility\ethernetif.cpp:172:18: error: 'struct ETH_InitTypeDef' has no member named 'DuplexMode'
  172 |   EthHandle.Init.DuplexMode = ETH_MODE_FULLDUPLEX;
      |                  ^~~~~~~~~~
.pio\libdeps\genericSTM32F407VGT6\STM32Ethernet\src\utility\ethernetif.cpp:172:31: error: 'ETH_MODE_FULLDUPLEX' was not declared in this scope
  172 |   EthHandle.Init.DuplexMode = ETH_MODE_FULLDUPLEX;
      |                               ^~~~~~~~~~~~~~~~~~~
In file included from C:\Users\Mikrotroniks\.platformio\packages\framework-arduinoststm32\system\Drivers\CMSIS\Device\ST\STM32F4xx\Include/stm32f4xx.h:132,
                 from C:\Users\Mikrotroniks\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/stm32_def.h:36,
                 from .pio\libdeps\genericSTM32F407VGT6\STM32Ethernet\src\utility\ethernetif.cpp:48:
C:\Users\Mikrotroniks\.platformio\packages\framework-arduinoststm32\system\Drivers\CMSIS\Device\ST\STM32F4xx\Include/stm32f407xx.h:11633:53: error: invalid conversion from 'long unsigned int' to 'ETH_MediaInterfaceTypeDef' [-fpermissive]
11633 | #define SYSCFG_PMC_MII_RMII_SEL_Msk          (0x1UL << SYSCFG_PMC_MII_RMII_SEL_Pos) /*!< 0x00800000 */
C:\Users\Mikrotroniks\.platformio\packages\framework-arduinoststm32\system\Drivers\CMSIS\Device\ST\STM32F4xx\Include/stm32f407xx.h:11634:46: note: in expansion of macro 'SYSCFG_PMC_MII_RMII_SEL_Msk'    
11634 | #define SYSCFG_PMC_MII_RMII_SEL              SYSCFG_PMC_MII_RMII_SEL_Msk       /*!<Ethernet PHY interface selection */
      |                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Mikrotroniks\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_hal_eth.h:1346:40: note: in expansion of macro 'SYSCFG_PMC_MII_RMII_SEL'
 1346 | #define ETH_MEDIA_INTERFACE_RMII      (SYSCFG_PMC_MII_RMII_SEL)
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\genericSTM32F407VGT6\STM32Ethernet\src\utility\ethernetif.cpp:174:35: note: in expansion of macro 'ETH_MEDIA_INTERFACE_RMII'
  174 |   EthHandle.Init.MediaInterface = ETH_MEDIA_INTERFACE_RMII;
      |                                   ^~~~~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\genericSTM32F407VGT6\STM32Ethernet\src\utility\ethernetif.cpp:181:18: error: 'struct ETH_InitTypeDef' has no member named 'RxMode'
  181 |   EthHandle.Init.RxMode = ETH_RXPOLLING_MODE;
      |                  ^~~~~~
.pio\libdeps\genericSTM32F407VGT6\STM32Ethernet\src\utility\ethernetif.cpp:183:18: error: 'struct ETH_InitTypeDef' has no member named 'ChecksumMode'
  183 |   EthHandle.Init.ChecksumMode = ETH_CHECKSUM_BY_HARDWARE;
      |                  ^~~~~~~~~~~~
.pio\libdeps\genericSTM32F407VGT6\STM32Ethernet\src\utility\ethernetif.cpp:184:18: error: 'struct ETH_InitTypeDef' has no member named 'PhyAddress'
  184 |   EthHandle.Init.PhyAddress = LAN8742A_PHY_ADDRESS;
      |                  ^~~~~~~~~~
C:\Users\Mikrotroniks\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_hal_eth.h:1306:35: error: expected identifier before numeric constant
 1306 | #define HAL_ETH_STATE_RESET       0x00000000U    /*!< Peripheral not yet Initialized or disabled */
      |                                   ^~~~~~~~~~~
C:\Users\Mikrotroniks\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/Legacy/stm32f4xx_hal_eth_legacy.h:354:3: note: in expansion of macro 'HAL_ETH_STATE_RESET'
  354 |   HAL_ETH_STATE_RESET             = 0x00U,    /*!< Peripheral not yet Initialized or disabled         */
      |   ^~~~~~~~~~~~~~~~~~~
C:\Users\Mikrotroniks\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_hal_eth.h:1306:35: error: expected '}' before numeric constant
 1306 | #define HAL_ETH_STATE_RESET       0x00000000U    /*!< Peripheral not yet Initialized or disabled */
      |                                   ^~~~~~~~~~~
C:\Users\Mikrotroniks\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/Legacy/stm32f4xx_hal_eth_legacy.h:354:3: note: in expansion of macro 'HAL_ETH_STATE_RESET'
[... 198 lines cut ...]
.pio\libdeps\genericSTM32F407VGT6\STM32Ethernet\src\utility\ethernetif.cpp:564:24: error: 'struct ETH_InitTypeDef' has no member named 'Speed'
  564 |         EthHandle.Init.Speed = ETH_SPEED_100M;
      |                        ^~~~~
.pio\libdeps\genericSTM32F407VGT6\STM32Ethernet\src\utility\ethernetif.cpp:573:80: error: 'struct ETH_InitTypeDef' has no member named 'DuplexMode'
  573 |       HAL_ETH_WritePHYRegister(&EthHandle, PHY_BCR, ((uint16_t)(EthHandle.Init.DuplexMode >> 3) |
      |                                                                                ^~~~~~~~~~
.pio\libdeps\genericSTM32F407VGT6\STM32Ethernet\src\utility\ethernetif.cpp:574:80: error: 'struct ETH_InitTypeDef' has no member named 'Speed'
  574 |                                                      (uint16_t)(EthHandle.Init.Speed >> 1)));
      |                                                                                ^~~~~
.pio\libdeps\genericSTM32F407VGT6\STM32Ethernet\src\utility\ethernetif.cpp:578:36: error: 'ETH_MACInitTypeDef' was not declared in this scope; did you mean 'ETH_InitTypeDef'?
  578 |     HAL_ETH_ConfigMAC(&EthHandle, (ETH_MACInitTypeDef *) NULL);
      |                                    ^~~~~~~~~~~~~~~~~~
      |                                    ETH_InitTypeDef
.pio\libdeps\genericSTM32F407VGT6\STM32Ethernet\src\utility\ethernetif.cpp:578:56: error: expected primary-expression before ')' token
  578 |     HAL_ETH_ConfigMAC(&EthHandle, (ETH_MACInitTypeDef *) NULL);
      |                                                        ^
.pio\libdeps\genericSTM32F407VGT6\STM32Ethernet\src\utility\ethernetif.cpp:578:5: error: 'HAL_ETH_ConfigMAC' was not declared in this scope
  578 |     HAL_ETH_ConfigMAC(&EthHandle, (ETH_MACInitTypeDef *) NULL);
      |     ^~~~~~~~~~~~~~~~~
Compiling .pio\build\genericSTM32F407VGT6\lib949\STM32duino LwIP\core\altcp.c.o
*** [.pio\build\genericSTM32F407VGT6\src\main.cpp.o] Error 1
*** [.pio\build\genericSTM32F407VGT6\lib9b1\STM32Ethernet\utility\ethernetif.cpp.o] Error 1
```

Please Help...

valeros, 2024-04-24:
Hi @HareshPrajapati, could you please check if it works in Arduino IDE?

HareshPrajapati, 2024-04-24:
@valeros 
It is completely compiled  on an Arduino.

HareshPrajapati, 2024-04-24:
I Already did it.
see
[image]

AirLongDian, 2024-04-27:
i have the same problem

HareshPrajapati, 2024-04-30:
Thank you very much @valeros 

It Works !

HareshPrajapati, 2024-04-30:
> Thank you very much @valeros
> 
> It Works !

> What happens if you set those defines globally in the `platformio.ini` file? For example:
> 
> ```ini
> ...
> build_flags =
>   -DENABLE_HWSERIAL2
>   -DHAL_ETH_LEGACY_MODULE_ENABLED
>   -DHAL_ETH_MODULE_DISABLED
> ```

Solution..

closed 2024-04-30 by HareshPrajapati

fpistm, 2024-05-06:
Just a quick thought, the `hal_conf_extra.h` file should work in pio environment.
https://github.com/stm32duino/Arduino_Core_STM32/blob/f31d070d1f2059494c6369ab52808729381f9750/system/STM32F4xx/stm32f4xx_hal_conf.h#L10-L12
If not (it seems the case), the way pio builds and includes files is not correct and would hide other issue compare to Arduino IDE env.