platformio/platform-ststm32 #772: JSON file for NUCLEO-F446ZE incorrect, not showing support for Arduino via STM32diuno core

pmantoine, 2024-04-25:
Running PlatformIO, with it telling me:

PLATFORM: ST STM32 (17.3.0) > ST Nucleo F446ZE
HARDWARE: STM32F446ZET6 180MHz, 128KB RAM, 512KB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, cmsis-dap, jlink)
PACKAGES:

framework-arduinoststm32 @ 4.20701.0 (2.7.1)
framework-cmsis @ 2.50900.0 (5.9.0)
toolchain-gccarmnoneeabi @ 1.120301.0 (12.3.1)
Converting CommandStation-EX.ino
Error: This board doesn't support arduino framework!
This despite supposedly having v2.7.1 of STM32duino core.

When I look in .platformio/platforms/ststm32/boards, the nucleo_f446ze.json file there contains:

{
"build": {
"cpu": "cortex-m4",
"extra_flags": "-DSTM32F4 -DSTM32F446xx",
"f_cpu": "180000000L",
"mcu": "stm32f446zet6",
"product_line": "STM32F446xx"
},
"connectivity": [
"can"
],
"debug": {
"default_tools": [
"stlink"
],
"jlink_device": "STM32F446ZE",
"onboard_tools": [
"stlink"
],
"openocd_board": "st_nucleo_f4",
"openocd_target": "stm32f4x",
"svd_path": "STM32F446x.svd"
},
"frameworks": [
"cmsis",
"mbed",
"stm32cube",
"libopencm3",
"zephyr"
],
"name": "ST Nucleo F446ZE",
"upload": {
"maximum_ram_size": 131072,
"maximum_size": 524288,
"protocol": "stlink",
"protocols": [
"jlink",
"cmsis-dap",
"stlink",
"blackmagic",
"mbed"
]
},
"url": "https://developer.mbed.org/platforms/ST-Nucleo-F446ZE/",
"vendor": "ST"
}

Which clearly lacks the Arduino core mentions.

I have been told by the STM32duino core maintainer that this is a PlatformIO issue. Hand editing the file does indeed show build support works just fine as the necessary variant files etc are present. Is there some issue with the automatic generation of this file?

This is my hand-edited file that works when I put in place instead:

{
  "build": {
    "cpu": "cortex-m4",
    "extra_flags": "-DSTM32F4 -DSTM32F446xx",
    "f_cpu": "180000000L",
    "mcu": "stm32f446zet6",
    "product_line": "STM32F446xx",
    "variant": "STM32F4XX/F446Z(C-E)(H-J-T)"
  },
  "connectivity": [
    "can"
  ],
  "debug": {
    "default_tools": [
      "stlink"
    ],
    "jlink_device": "STM32F446ZE",
    "onboard_tools": [
      "stlink"
    ],
    "openocd_board": "st_nucleo_f4",
    "openocd_target": "stm32f4x",
    "svd_path": "STM32F446x.svd"
  },
  "frameworks": [
    "arduino",
    "cmsis",
    "mbed",
    "stm32cube",
    "libopencm3",
    "zephyr"
  ],
  "name": "ST Nucleo F446ZE",
  "upload": {
    "maximum_ram_size": 131072,
    "maximum_size": 524288,
    "protocol": "stlink",
    "protocols": [
      "jlink",
      "cmsis-dap",
      "stlink",
      "blackmagic",
      "mbed"
    ]
  },
  "url": "https://developer.mbed.org/platforms/ST-Nucleo-F446ZE/",
  "vendor": "ST"
}

Let me know if I can be of further assistance!

valeros, 2024-04-25:
Hi @pmantoine, feel free to submit a PR with proper changes that enable Arduino framework for this board.

pmantoine, 2024-05-07:
Missed a single line for "core", and have pushed it.