platformio/platform-ststm32 #786: elf has a LOAD segment with RWX permissions

martonpetroczki, 2024-07-04:
Using:
```
[env:devebox_h743vitx]
platform = ststm32
board = devebox_h743vitx
framework = arduino
upload_protocol = stlink
```

with empty sketch:
```
#include <Arduino.h>
void setup() {
}
void loop() {
}
```

Processing devebox_h743vitx (platform: ststm32; board: devebox_h743vitx; framework: arduino)
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/devebox_h743vitx.html
PLATFORM: ST STM32 (17.3.0) > DevEBox H743VITX
HARDWARE: STM32H743VIT6 480MHz, 512KB RAM, 2MB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 @ 4.20701.0 (2.7.1)
 - framework-cmsis @ 2.50900.0 (5.9.0)
 - toolchain-gccarmnoneeabi @ 1.120301.0 (12.3.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 12 compatible libraries
Scanning dependencies...
No dependencies
Building in debug mode
Compiling .pio\build\devebox_h743vitx\src\main.cpp.o
Linking .pio\build\devebox_h743vitx\firmware.elf

**`c:/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld.exe: warning: .pio/build/devebox_h743vitx/firmware.elf has a LOAD segment with RWX permissions`**

Checking size .pio\build\devebox_h743vitx\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.2% (used 1248 bytes from 524288 bytes)
Flash: [          ]   0.9% (used 18204 bytes from 2097152 bytes)
Building .pio\build\devebox_h743vitx\firmware.bin
================================================================================================== [SUCCESS] Took 4.20 seconds ==================================================================================================
 *  Terminal will be reused by tasks, press any key to close it. 

Is this normal or am I doing something wrong?
Please delete if the question is not relevant.

brightproject, 2024-07-06:
Thanks @martonpetroczki for issue, I have same problem
My `*ini`

> [env:genericSTM32F411CE]
> platform = ststm32
> board = blackpill_f411ce
> board_build.mcu = stm32f411ceu6
> board_build.f_cpu = 72000000L
> ; board_build.f_cpu = 100000000L
> framework = arduino
> monitor_speed = 115200
> monitor_port = COM[3]

```
Linking .pio\build\genericSTM32F411CE\firmware.elf
.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld.exe: warning: .pio/build/genericSTM32F411CE/firmware.elf has a LOAD segment with RWX permissions
Checking size .pio\build\genericSTM32F411CE\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   1.3% (used 1732 bytes from 131072 bytes)
Flash: [=         ]   5.4% (used 28112 bytes from 524288 bytes)
Building .pio\build\genericSTM32F411CE\firmware.bin
```
I found file in folder

> .platformio\packages\framework-arduinoststm32\system\Drivers\CMSIS\Device\ST\STM32F4xx\Source\Templates\gcc\startup_stm32f411xe.s

But I don't sure that is helps.
https://stackoverflow.com/questions/73429929/gnu-linker-elf-has-a-load-segment-with-rwx-permissions-embedded-arm-project

valeros, 2024-07-08:
That's expected warning with the latest toolchain and linker scripts from the Arduino framework.

closed 2024-07-08 by valeros