platformio/platform-ststm32 #793: STM32Duino: USB serial library not automatically included

maxgerhardt, 2024-08-01:
From [the docs](https://docs.platformio.org/en/latest/platforms/ststm32.html#stm32duino-configuration-system), consider the `platformio.ini`

```
[env:nucleo_f401re]
platform = ststm32
framework = arduino
board = nucleo_f401re
build_flags =
  -D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC
```
and `src/main.cpp` as 

```cpp
#include <Arduino.h>

void setup() {
    Serial.begin(115200);
}

void loop() { }
```
Then this will result in a compile failure

```
Linking .pio\build\nucleo_f401re\firmware.elf
c:/users/max/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld.exe: warning: .pio/build/nucleo_f401re/firmware.elf has a LOAD segment with RWX permissions
c:/users/max/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/nucleo_f401re/SrcWrapper/src/stm32/hw_config.c.o: in function `hw_config_init':
hw_config.c:(.text.hw_config_init+0x20): undefined reference to `USBD_CDC_init'
c:/users/max/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/nucleo_f401re/src/main.cpp.o: in function `setup':
main.cpp:(.text.setup+0xc): undefined reference to `SerialUSB'
c:/users/max/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld.exe: main.cpp:(.text.setup+0x6): undefined reference to `USBSerial::begin(unsigned long)'
c:/users/max/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/nucleo_f401re/libFrameworkArduino.a(WSerial.cpp.o): in function `serialEventRun()':
WSerial.cpp:(.text._Z14serialEventRunv+0x18): undefined reference to `USBSerial::available()'
c:/users/max/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld.exe: WSerial.cpp:(.text._Z14serialEventRunv+0x30): undefined reference to `SerialUSB'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\nucleo_f401re\firmware.elf] Error 1
===================[FAILED] Took 12.16 seconds ===================
```
And it's especially suspicious that the dependencies are empty:
```
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_f401re.html
PLATFORM: ST STM32 (17.5.0) > ST Nucleo F401RE
HARDWARE: STM32F401RET6 84MHz, 96KB RAM, 512KB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, cmsis-dap, jlink)
PACKAGES:
 - framework-arduinoststm32 @ 4.20801.0 (2.8.1)
 - framework-cmsis @ 2.50900.0 (5.9.0)
 - toolchain-gccarmnoneeabi @ 1.120301.0 (12.3.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 14 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
```

The functions which it's missing in the linker error are supplied by the USBDevice library. When the code is changed to

```
#include <Arduino.h>
#include <USBDevice.h>

void setup() {
    Serial.begin(115200);
}

void loop() {

}
```

the dependency is recognized and the build goes through.

```
Dependency Graph
|-- USBDevice @ 1.0.0
Building in release mode
Compiling .pio\build\nucleo_f401re\src\main.cpp.o
[..]
Linking .pio\build\nucleo_f401re\firmware.elf
c:/users/max/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld.exe: warning: .pio/build/nucleo_f401re/firmware.elf has a LOAD segment with RWX permissions
Checking size .pio\build\nucleo_f401re\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   5.0% (used 4872 bytes from 98304 bytes)
Flash: [          ]   4.0% (used 20956 bytes from 524288 bytes)
Building .pio\build\nucleo_f401re\firmware.bin
================[SUCCESS] Took 6.67 seconds ================
```

This is a regression from 17.4.0 to 17.5.0: Setting `platform = ststm32@17.4.0` makes the original `main.cpp` without `#include <USBDevice.h>` compile without problems.

MostHated, 2024-08-01:
I was banging my head against a wall with this yesterday and thought I goofed up my build flags or something and discovered going back to 17.4.0 fixed it.

valeros, 2024-08-02:
Thanks for pointing it out, I've just pushed a new package with a fix. To pull the latest changes run `pio pkg update -g -p ststm32`.

closed 2024-08-02 by valeros

MostHated, 2024-08-14:
That is quite strange, as I had done all that back a week or two ago when this initial issue came about when you mentioned the fix/update was out. I then removed the ststm32@17.4.0 version definition and just used platform = ststm32 and have been doing so since, so I wonder what caused it to fritz out. I just did it again, removing the cache, then doing the above update command, and now it works again. Weird.