platformio/platform-ststm32 #794: Posible error with dfu-util UPLOADERFLAGS "No DFU capable USB device available"

pmmarquez, 2024-08-08:
With the Arduino Nicla Vision i was having this issue:
```
dfu-util: No DFU capable USB device available
*** [upload] Error 74
```
I track it down to this line in **builder/main.py**:
```python
_upload_flags = [
    "-d", ",".join(["%s:%s" % (hwid[0], hwid[1]) for hwid in hwids]),
    "-a", "0", "-s",
    "%s:leave" % board.get("upload.offset_address", "0x08000000"), "-D"
]
```
Changing the "," by " " to join the hwids was the solution:
```python
_upload_flags = [
    "-d", " ".join(["%s:%s" % (hwid[0], hwid[1]) for hwid in hwids]),
    "-a", "0", "-s",
    "%s:leave" % board.get("upload.offset_address", "0x08000000"), "-D"
]
```
I don't know if this is a general issue or is just related with my system, I tested in both Windows and Linux(WSL), I also don't have any other boards to test at the moment, but hopefully this helps other people with the same issue.

valeros, 2024-08-08:
Hi @pmmarquez,

How did you come up with the idea of using `" "` instead of `","`? The official [manual page](https://dfu-util.sourceforge.net/dfu-util.1.html) for dfu-util suggests using `,` for separating several device IDs.

pmmarquez, 2024-08-08:
> Hi @pmmarquez,
> 
> How did you come up with the idea of using `" "` instead of `","`? The official [manual page](https://dfu-util.sourceforge.net/dfu-util.1.html) for dfu-util suggests using `,` for separating several device IDs.

I tried different variants directly in the command line and only with space worked out, later I changed in builder/main.py and that solved the issue for me

denblock, 2024-08-08:
I had the same problem on Windows with the Arduino GIGA.

Using " " instead of "," as the seperator also worked for me.

cheddarwhizzy, 2024-08-17:
Trying to run an example sketch from the Arduino IDE via PlatformIO using Arduino Giga Display Shield + R1 Giga Wifi on OSX yielded the same result. Had to remove all except the first array under `hwids` from `~/.platformio/platforms/ststm32/boards/giga_r1_m7.json` as per [this comment](https://community.platformio.org/t/giga-dfu-issues/41982/12#:~:text=replace%20that%20hwids%20section%20with%20just) - after doing that, I was able to `pio run -t upload` successfully. Works fine in Arduino IDE as-is, but PIO didn't work w/o this change for me.

zulu80, 2024-09-04:
I have the same problem with the Arduino Giga. I am using macos, platformio 6.1.15 / 3.4.4 with CLion.
I have reproduced the issue with the command line. The comma is not the cause, but a symptom.  The upload always fails if more than one device ID is specified.
The suggestion from @pmmarquez, causes only one device ID to be transferred and masks the actual problem.

bu5hm4nn, 2024-09-14:
I ran into the same issue as @zulu80 with my Arduino Giga. I ended up editing `giga_r1_m7.json` and removing all device id's but the first. Then it worked. Please fix.

EvanRobinson, 2024-10-23:
Me3 on the same problem as @zulu80 and @bu5hm4nn with the Giga R1 WiFi working on the Mac (M2 Max MacBookPro, 14", 32GB , Sonoma 14.4.1), PIO 6.1.16 / 3.4.4.

Edited giga_r1_m7.json as well, problem appears solved.

Please fix.

commit 62a48b6c (2024-10-23): Remove redundant HWIDs for Arduino Giga & Nicla boards

Fixes #794
closed 2024-10-23 by valeros