platformio/platform-ststm32 #795: Files deleted all of a sudden in \.platformio\packages\framework-arduinoststm32\variants\Generic_F103Vx

fkahhaleh, 2024-08-10:
Hi,

I am working on a custom board for Marlin. It was defined as BARD_CUSTOM and was compiling/flashing fine.
After that phase was done, I wanted to get my bootloader working so that I can flash via Serial instead of STLINK.

while changing some parameters in my custom.ini file I ran into some build errors So I wanted to go back to my working base line
and take slower path and smaller changes to see what breaks it.

My baseline no longer compiles.
All changes were made to the .ini definition file that's all.

the error I am getting is (multiple times): 
`/.platformio\packages\framework-arduinoststm32\system\STM32F1xx/stm32f1xx_hal_conf.h:4:10: fatal error: variant.h: No such file or directory`

I went to check on
`\.platformio\packages\framework-arduinoststm32\variants\Generic_F103Vx` and the folder is completely empty.
Other folders such as Generic_F103Zx for example has the pins and variant header files...etc.

How can I get those missing files back? 
I could not find a repair or a way to re-install the packages.

It would also be great to know how they got deleted in the first place!

Platform.ini:
`[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = custom
include_dir  = Marlin
extra_configs =
    Marlin/config.ini
    ini/avr.ini
    ini/due.ini
... REMOVED to keep it short
    ini/stm32-common.ini
    ini/stm32f0.ini
    ini/stm32f1-maple.ini
    ini/stm32f1.ini
... REMOVED to keep it short
    ini/custom.ini `

Custom.ini
`[env:custom]
extends           = common_stm32
board             = genericSTM32F103VE
build_flags       = ${common_stm32.build_flags}                    
                    -DHAL_SRAM_MODULE_ENABLED
upload_protocol   = stlink`

fkahhaleh, 2024-08-10:
Ok So after I posted this, I found that I had multiple ST STM32 platforms installed (I thought  I only had the latest 17.x)
I uninstalled all of them, recompiled the project and PlatformIO installed the missing packages and now I can see the variants folder back and the firmware compiles successfully.

I'd still like to know what was going on before and if there is a better way to resolve it.
Thank you!

valeros, 2024-08-12:
Hi @fkahhaleh, I believe PlatformIO was interrupted in some way while switching between platform versions and hence it didn't have a change to properly install a new Arduino core package and back up the old one. Next time you have a corrupted package, just remove the entire folder in `~/.platformio/packages/<package_name>` and PlatformIO will redownload that package according to your platform restrictions.