platformio/platform-ststm32 #799: [Board Request] Add Support for WeActStudio STM32H562RGT6 to PlatformIO

Zazins, 2024-09-20:
Description of Problem
I would like to request the inclusion of the WeActStudio STM32H562RGT6 board in the PlatformIO board list. This board is based on the STM32H562RGT6 microcontroller, which runs at 250 MHz and includes peripheral support such as CAN, USB, and SDMMC.

The board is commonly used in high-performance applications and having direct support in PlatformIO would streamline the development process for users. I am happy to provide specifications and configurations to assist with the integration.

Steps to Reproduce:
Add WeActStudio STM32H562RGT6 to the PlatformIO supported board list.
Provide support for STM32H562RGT6 (250 MHz clock, 1024 KB flash, 640KB RAM, and peripheral setup).
Ensure board definitions are available for frameworks like STM32Cube and Arduino.
Actual Results:
The WeActStudio STM32H562RGT6 board is not currently supported in PlatformIO.

Expected Results:
WeActStudio STM32H562RGT6 board should be listed in PlatformIO's board list with full support for building, uploading, and development using the STM32H562RGT6 microcontroller.

Additional Info:
The board is popular among hobbyists and developers, and it would be useful to have official support in PlatformIO. Please refer to the following links for further details:

[WeActStudio STM32H562RGT6 Board GitHub][(https://github.com/WeActTC/MiniSTM32H7xx)](https://github.com/WeActStudio/WeActStudio.STM32H5_64Pin_CoreBoard)

Thank you for considering this request.

KristianJerpetjon, 2024-10-01:
How would one go about making a PR to solve this ?

valeros, 2024-10-02:
Hi @KristianJerpetjon,

The main issue here is that the H5 family is not supported in the latest stable OpenOCD which makes it hard to use with PlatformIO.

ypearson-bdai, 2024-10-02:
Please add support to `nucleo_h563zi` which is support by `zephyr`

Getting error:
`: Unknown board ID 'nucleo_h563zi'`

ypearson-bdai, 2024-10-02:
> Hi @KristianJerpetjon,
> 
> The main issue here is that the H5 family is not supported in the latest stable OpenOCD which makes it hard to use with PlatformIO.

There are methods to program the H5 other than OpenOCD