platformio/platform-ststm32 #801: Compilation for blackpill_f411ce ends in error openr.c:(.text._open_r+0x10): undefined reference to `_open'

brightproject, 2024-09-24:
I am compiling the example code from

> https://github.com/avpripri/RTIMULib2/tree/master/Arduino

Under `esp32` the code is compiling and working perfectly.

> [env:ESP32]
> platform = 
> 	espressif32@^3.5.0 
> 	; espressif32@^6.7.0 
> 	; espressif32@^6.8.1 
> 	; https://github.com/pioarduino/platform-espressif32
> board = esp32-s3-devkitc-1-n16r8
> framework = arduino
> monitor_speed = 115200
> monitor_port = COM6
> upload_speed = 921600
> upload_port = COM6
> monitor_filters = esp32_exception_decoder

For `stm32` everything is much more interesting and confusing.

> Processing genericSTM32F411CE (platform: 
> platformio/ststm32@^14.1.0; board: blackpill_f411ce; framework: arduino)
> --------------------------------Verbose mode can be enabled via `-v, --verbose` option
> CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/blackpill_f411ce.html
> PLATFORM: ST STM32 (14.2.0) > WeAct Studio BlackPill V2.0 (STM32F411CE)
> HARDWARE: STM32F411CEU6 100MHz, 128KB RAM, 512KB Flash
> DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
> PACKAGES:
>  - framework-arduinoststm32 @ 4.20000.210603 (2.0.0)
>  - framework-cmsis @ 2.50700.210515 (5.7.0)
>  - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
> Converting RTArduLinkIMU.ino

> [platformio]
> ; default_envs = genericSTM32F103CB
> default_envs = genericSTM32F411CE
> 
> [env:genericSTM32F103CB]
> platform = 
>     ; ststm32@^6.1.1
>     ; ststm32@^8.0.0
>     ststm32@^9.0.0
> board = genericSTM32F103CB
> framework = arduino
> upload_protocol = serial
> 
> [env:genericSTM32F411CE]
> platform = 
>     ; platformio/ststm32@^6.1.1
>     ; platformio/ststm32@^8.0.0
> 	; platformio/ststm32@^14.1.0
> 	platformio/ststm32@^17.5.0
> ; platform_packages = 
> ; 	platformio/toolchain-gccarmnoneeabi@1.100301.220327
> board = blackpill_f411ce
> framework = arduino
> 
> monitor_speed = 115200
> ; monitor_speed = 230400
> monitor_port = COM6
> 
> ; monitor_filters = default, time, log2file
> monitor_filters = time, log2file
> 
> upload_protocol = 
>     ; serial
>     stlink
> build_flags =
>     ; -Q
>     ; -lrt
>     -specs=nosys.specs
>     ; -D SOME_MACRO=1
>     ; -D CONFIG_STD_CPP17=1
>     ; -D CONFIG_NEWLIB_LIBC=1
>     ; -D NEWLIB=1
> 	; -v
> 	; -w
> 	; -Werror
> 

If I select board `genericSTM32F103CB`, and version` platformio/ststm32@^6.1.1`, then the code is compiled and everything is `OK`.
With newer versions the code does not compile.
If I select board `genericSTM32F411CE`, then the code is not compiled, neither with version `17.5.0` nor with the newest one.
The error is always like this:
```
.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld.exe:

.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard\libc_nano.a(libc_a-openr.o): in function `_open_r':
openr.c:(.text._open_r+0x10): undefined reference to `_open'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\genericSTM32F411CE\firmware.elf] Error 1
```
note that I try to use flags:

> build_flags = -specs=nosys.specs
But this does not lead to a positive result.
I'm also curious what the flag does `-Q` и how to output the compilation log if I want to output information with the `-v` or `-Werror` flag I just don't have enough terminal buffer.
Specifying in the `*ini` file

> monitor_filters = default, time, log2file

Doesn't give any results.

I will hope for help from the community.

valeros, 2024-09-25:
Hi @brightproject, the issue tracker in this repository is not the right place to ask project-specific questions. Anyway, v6.1.1 is 4 years old, there have been plenty of changes in both - the underlying frameworks and the `ststm32` platform itself. First of all, I'd recommend using the latest version v17.x. Secondly, I doubt that `build_flags = -specs=nosys.specs` is the correct way of setting special linker flags, it should probably be passed to the linker via an extra script, see https://docs.platformio.org/en/latest/scripting/examples/extra_linker_flags.html

brightproject, 2024-09-25:
> Anyway, v6.1.1 is 4 years old

Thanks @valeros  for the kind words and explanations.
That's the problem, with the latest version of the framework, I get similar compilation errors.
And I can't even figure out what the reason for this compiler behavior is.
If only I could find at least one clue, I would refactor the code for the new framework.

brightproject, 2024-09-26:
The build for `.pio` is OK, the solution to the problem is here:

> https://github.com/orgs/stm32duino/discussions/2523

[image]