platformio/platform-ststm32 #806: Outdated stm32cube framework version (including HAL and LL)

AndBondStyle, 2024-10-18:
This issue was mentioned multiple times already (#368, #439, #555). Current `stm32cubeXX` platformio packages are out of date for about 3 years ([example](https://registry.platformio.org/tools/platformio/framework-stm32cubeg4)). Are there any plans to automate the update process to keep platformio packages in sync? Or is there any potential compatibility problems?

---

As a temporary fix, I'm using this method. Maybe it can be useful to someone:
1. Clone the official `STM32CubeXX` repo with a specific version:
```
git clone https://github.com/STMicroelectronics/STM32CubeG4 -b v1.6.0 --depth 1 <path>
```
2. Add `package.json` file into the repository:
```
{
    "description": "STM32Cube is a set of tools and embedded software bricks available free of charge to enable fast and easy development on the STM32 platform (STM32CubeG4 MCU Firmware Package)",
    "keywords": ["framework", "hal", "stm32", "st"],
    "name": "framework-stm32cubeg4",
    "version": "1.6.0",
    "homepage": "https://github.com/STMicroelectronics/STM32CubeG4"
  }
```
3. Include this directory as a platform dependency of your projects (using [local folder](https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_install.html#local-folder) aka "file://" spec):
```diff
[env:genericSTM32G431CB]
platform = ststm32
board = genericSTM32G431CB
framework = stm32cube
+ platform_packages =
+    framework-stm32cubeg4 @ file:///path/to/repo
```
4. Optionally, you could push this repo to github and use [git spec](https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_install.html#repository-git-hg-svn) instead

AndBondStyle, 2024-10-18:
More possibly related issues: #766, #691, #686, #682, #666, #642, #636, #632

insolace, 2024-12-28:
Any update on this? Adding platform_packages doesn't remove the out of date platformio framework which is making it difficult to use the latest STM32G0 drivers.

valeros, 2025-01-29:
Resolved in https://github.com/platformio/platform-ststm32/commit/4d46eda1f48982b3a92a415400ef7531a636cb1d

AndBondStyle, 2025-01-29:
@valeros just to clarify: this is another manual update of the packages, or some process is implemented to get future versions automatically (or at least more frequently than once in 3 years)?

valeros, 2025-01-30:
@AndBondStyle It's a manual update, but more frequent updates are planned.

RudolphRiedel, 2025-02-15:
What about support for C0, U0, U5 and H5?

valeros, 2025-02-17:
@RudolphRiedel Support for new targets is not in plan at the moment.

crespum, 2026-08-24:
What's the roadblock for supporting new ST families? Is it lack of resources or is there something in the architecture itself which makes it difficult to integrate?