platformio/platform-ststm32 #808: cppcheck: Bad CPU type in executable on macOS arm64

loozhengyuan, 2024-10-25:
Similar to #436, `cppcheck` is not correctly installed on macOS arm64.

```console
$ pio pkg uninstall -g -t platformio/tool-cppcheck             
Tool Manager: Removing tool-cppcheck @ 1.21100.230717
Tool Manager: tool-cppcheck@1.21100.230717 has been removed!
$ pio pkg install -g -t platformio/tool-cppcheck          
Tool Manager: Installing platformio/tool-cppcheck
Unpacking  [####################################]  100%
Tool Manager: tool-cppcheck@1.21100.230717 has been installed!
$ file ~/.platformio/packages/tool-cppcheck/cppcheck 
~/.platformio/packages/tool-cppcheck/cppcheck: Mach-O 64-bit executable x86_64
```

This results in the following error when the `pio check` command is ran:

```console
$ pio check
...
OSError: [Errno 86] Bad CPU type in executable: '~/.platformio/packages/tool-cppcheck/cppcheck'
```

valeros, 2024-10-30:
`tool-cppcheck` package for MacOS ARM64 has been added to the registry.

closed 2024-10-30 by valeros

loozhengyuan, 2024-11-10:
@valeros Thanks for the quick fix, also verified on my end that it is working. Appreciate it!