platformio/platform-ststm32 #809: Add support for the STM32H7RS

chintal, 2024-11-02:
ST has recently launched a series of scalable bootflash MCUs - [ STM32H7Rx/7Sx]( https://www.st.com/resource/en/application_note/an6062-introduction-to-stm32h7rx7sx-system-architecture-and-performance-stmicroelectronics.pdf.
). They've also released Nucleo boards for one of the variants : [NUCLEO-H7S3L8](https://www.st.com/en/evaluation-tools/nucleo-h7s3l8.html).

Would it be possible to use platformIO with these chips?

The HAL library provided by ST for these parts is `STM32H7RSxx_HAL_Driver`, and not the `STM32H7xx_HAL_Driver` used by all the older STM32H7 parts. 

These are bootflash chips having a small amount of on-chip flash (64k), intended to be used by a Boot firmware. The Application firmware is intended to run off of external flash, either with XIP or load-and-run. CubeMX manages this by actually creating 2 projects (or 3, if you also enable the ExtMemLoader application). 

I have not yet worked out how the second program into the external flash is programmed by CubeMX or otherwise. I intend to just shoehorn everything I need presently into the boot application, since what I want from this chip is the raw clock speed. The approach to deal with this issue may be similar to what is done on the dual core STM32H7s, which I have not yet used.

chintal, 2024-11-08:
Similar to what I have done for #798, I've put together a minimal setup to workaround this issue and get baremetal support for the STM32H7RSxx, specifically the STM32H7S3L8 on the  [NUCLEO-H7S3L8](https://www.st.com/en/evaluation-tools/nucleo-h7s3l8.html). I have tested this setup with code running from the on-chip flash only ("Boot", per ST CubeMX nomenclature). I have not tried to run any code on the external flash ("Application"). I expect building and debugging might work more or less as is, but writing the application to the external flash will probably need additional changes. 

Since I haven't tried any of the frameworks, and I don't actually expect them to work out of the box, I'm not creating a PR. I have also not tested with any debugger other than the Nucleo on-board STLink. 

I expect the following will minimally be needed to make a PR here viable: 
  - Update STM32CubeMX to something much more recent (actual minimum version required : unknown)
  - Update openocd to a recent version (actual minimum version required : unknown, could even be STM's fork for now)

For the moment, the following steps allow baremetal development for the chip  when targeting the on-chip flash using PlatformIO:

## Board file for Nucleo H7S3L8

The `nucleo_h7s3l8.json` file, based on one of the existing Nucleo boards, is created in ``~/.platformio/platforms/ststm32/boards`` with the following content:

```
{
  "build": {
    "core": "stm32",
    "cpu": "cortex-m7",
    "extra_flags": "-DSTM32H7RS -DSTM32H7S3xx -DSTM32H7S3L8xx",
    "f_cpu": "600000000L",
    "mcu": "stm32h7s3l8h6",
    "product_line": "STM32H7S3xx",
    "variant": "STM32H7S3xx/H7S3L8"
  },
  "debug": {
    "default_tools": [
      "stlink"
    ],
    "jlink_device": "STM32H7S3L8Hx",
    "onboard_tools": [
      "stlink"
    ],
    "openocd_target": "stm32h7x",
    "openocd_board": "stm32h7s3l8_generic",
    "svd_path": "STM32H7S3L8.svd"
  },
  "frameworks": [
    "arduino",
    "cmsis",
    "libopencm3",
    "stm32cube",
    "zephyr"
  ],
  "name": "Nucleo H7S3L8",
  "upload": {
    "maximum_ram_size": 262144,
    "maximum_size": 65536,
    "protocol": "stlink",
    "protocols": [
      "stlink",
      "jlink",
      "cmsis-dap",
      "blackmagic",
      "mbed"
    ]
  },
  "url": "https://www.st.com/en/evaluation-tools/nucleo-h7s3l8.html",
  "vendor": "ST"
}
```

## OpenOCD

The OpenOCD version provided by PlatformIO does not presently support the STM32H7RS. Based on some combination of external references and STMCubeMX generated code, I've put together an OpenOCD package for PlatformIO which works, available at https://github.com/ebs-universe/pio-tool-openocd-stm32. The openocd binary used is the binary which is shipped with STM32CubeIDE.

I have not published it to the PlatformIO repository since I do not intend to maintain it once the default openOCD version supports this chip.

## STM32Cube

Since the default ``STM32Cube`` versions provided by PlatformIO do not support the H7RS, I include the following files generated by CubeMX directly into my source tree. 

```
❯ tree src/cubemx 
src/cubemx
├── CMSIS
│   ├── Device
│   │   └── ST
│   │       └── STM32H7RSxx
│   │           ├── Include
│   │           │   ├── stm32h7rsxx.h
│   │           │   ├── stm32h7s3xx.h
│   │           │   └── system_stm32h7rsxx.h
│   │           ├── LICENSE.txt
│   │           └── Source
│   │               └── Templates
│   ├── Include
│   │   ├──  ...
│   └── LICENSE.txt
├── startup_stm32h7s3l8hx.s
├── STM32H7RSxx_HAL_Driver
│   ├── Inc
│   │   ├── Legacy
│   │   │   └── ...
│   │   ├── ...
│   ├── LICENSE.txt
│   └── Src
│       ├── ...
├── STM32H7S3L8HX_FLASH.ld
├── STM32H7S3L8HX_RAM.ld
├── syscalls.c
├── sysmem.c
└── system_stm32h7rsxx.c

13 directories, 149 files
```

I have also placed the ``stm32u0xx_hal_conf.h`` generated by CubeMX in the platformIO project ``include`` folder.

## platformio.ini

Successful build, program, and debug using the Nucleo on-board stlink requires the following project configuration: 

```
[env]
platform = ststm32
platform_packages =
  toolchain-gccarmnoneeabi@>1.120000.0
  tool-openocd@https://github.com/ebs-universe/pio-tool-openocd-stm32/archive/refs/heads/main.zip
board = nucleo_h7s3l8
upload_protocol = stlink
debug_tool = stlink
debug_port = localhost:3333
build_flags = 
    -I include
    -I src/cubemx/CMSIS/Device/ST/STM32H7RSxx/Include
    -I src/cubemx/CMSIS/Include
    -I src/cubemx/STM32H7RSxx_HAL_Driver/Inc
    -mthumb -mfpu=fpv5-d16
    -std=gnu++11
board_build.ldscript = src/cubemx/STM32H7S3L8HX_FLASH.ld

```

f-io, 2026-08-30:
Any progress on support for the this family?