platformio/platform-ststm32 #810: Zephyr project no longer builds

ypearson-bdai, 2024-11-03:
### Description of problem

Erased older platform IO, installed latest platform IO and broke my Zephyr build. The older version used Zephyr 3.6 and the latest uses Zephyr 3.7 now my code no longer compiles

PlatformIO Core, version 6.1.16
Ubuntu 22 LTS

### Actual Results
```
.pio/libdeps/nucleo_f767zi/zenoh-pico/src/system/zephyr/network.c:467:17: warning: implicit declaration of function 'net_if_ipv6_maddr_leave'; did you mean 'net_if_ipv6_maddr_lookup'? [-Wimplicit-function-declaration]
  467 |                 net_if_ipv6_maddr_leave(ifa, mcast);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~
      |                 net_if_ipv6_maddr_lookup
Compiling .pio/build/nucleo_f767zi/lib850/zenoh-pico/utils/pointers.o
Compiling .pio/build/nucleo_f767zi/lib850/zenoh-pico/utils/string.o
Compiling .pio/build/nucleo_f767zi/lib850/zenoh-pico/utils/uuid.o
Generating linker script .pio/build/nucleo_f767zi/zephyr/linker.cmd
.pio/libdeps/nucleo_f767zi/zenoh-pico/src/system/zephyr/network.c:471:17: warning: implicit declaration of function 'net_if_ipv6_maddr_rm'; did you mean 'net_ipv6_addr_cmp'? [-Wimplicit-function-declaration]
  471 |                 net_if_ipv6_maddr_rm(ifa, &((struct sockaddr_in6 *)rep._iptcp->ai_addr)->sin6_addr);
      |                 ^~~~~~~~~~~~~~~~~~~~
      |                 net_ipv6_addr_cmp
Archiving .pio/build/nucleo_f767zi/lib850/libzenoh-pico.a
Indexing .pio/build/nucleo_f767zi/lib850/libzenoh-pico.a
Linking .pio/build/nucleo_f767zi/zephyr/firmware-pre0.elf
/home/ypearson/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld.bfd: GNU-style constructors required but STATIC_INIT_GNU not enabled
/home/ypearson/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld.bfd: warning: .pio/build/nucleo_f767zi/zephyr/firmware-pre0.elf has a LOAD segment with RWX permissions
/home/ypearson/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld.bfd: /home/ypearson/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/lib/thumb/v7e-m/nofp/libc.a(libc_a-fini.o): in function `__libc_fini_array':
(.text.__libc_fini_array+0x1c): undefined reference to `_fini'

```

### Expected Results
Build success

```
[env:nucleo_f767zi]
platform = ststm32
board = nucleo_f767zi
framework = zephyr
monitor_speed = 115200
debug_tool = stlink

lib_deps = https://github.com/eclipse-zenoh/zenoh-pico#release/1.0.0.6

build_flags =
    -DZ_BATCH_UNICAST_SIZE=2048
    -DZ_FRAG_MAX_SIZE=4096
    -DINTERFACE=$INTERFACE
```

```
CONFIG_GPIO=y
CONFIG_I2C=y
CONFIG_I2C_CALLBACK=y
CONFIG_I2C_RTIO=y

CONFIG_SERIAL=y
CONFIG_CONSOLE=y
CONFIG_UART_CONSOLE=y
CONFIG_LOG=y

# Crashes zenoh
# CONFIG_LOG_MODE_IMMEDIATE=y

CONFIG_CBPRINTF_FP_SUPPORT=y

# Zenoh Pico options

# Kernel options
CONFIG_MAIN_STACK_SIZE=5120
CONFIG_HEAP_MEM_POOL_SIZE=150000
CONFIG_ENTROPY_GENERATOR=y
CONFIG_TEST_RANDOM_GENERATOR=y
CONFIG_INIT_STACKS=y

# Generic library options
CONFIG_NEWLIB_LIBC=y
CONFIG_NEWLIB_LIBC_NANO=n
CONFIG_POSIX_API=y

# Generic networking options
CONFIG_SERIAL=y
CONFIG_NETWORKING=y

CONFIG_NET_L2_ETHERNET=y
CONFIG_ETH_NATIVE_POSIX=y
CONFIG_ETH_NATIVE_POSIX_RANDOM_MAC=y

CONFIG_NET_IPV4=y
CONFIG_NET_IPV6=y
CONFIG_NET_TCP=y
CONFIG_NET_ARP=y
CONFIG_NET_UDP=y
CONFIG_NET_DHCPV4=n
CONFIG_NET_SHELL=y
CONFIG_NET_MGMT=y
CONFIG_NET_MGMT_EVENT=y
CONFIG_DNS_RESOLVER=y

# Sockets
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y
CONFIG_NET_SOCKETS_POLL_MAX=4

# Network driver config
CONFIG_TEST_RANDOM_GENERATOR=y
CONFIG_INIT_STACKS=y

# Network buffers
CONFIG_NET_PKT_RX_COUNT=16
CONFIG_NET_PKT_TX_COUNT=16
CONFIG_NET_BUF_RX_COUNT=80
CONFIG_NET_BUF_TX_COUNT=80
CONFIG_NET_CONTEXT_NET_PKT_POOL=y

# Network address config
CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NET_CONFIG_NEED_IPV4=y
CONFIG_NET_IPV4_IGMP=y
CONFIG_NET_CONFIG_NEED_IPV6=y
CONFIG_NET_IPV6_MLD=y
CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.168.1.2"
CONFIG_NET_CONFIG_MY_IPV4_NETMASK="255.255.255.0"
CONFIG_NET_CONFIG_MY_IPV4_GW="192.168.1.1"
#CONFIG_NET_CONFIG_PEER_IPV4_ADDR="192.168.10.10"
CONFIG_NET_CONFIG_MY_IPV6_ADDR="2001:db8::2"
CONFIG_NET_CONFIG_PEER_IPV6_ADDR="2001:db8::1"

# IP address options
CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=3
CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=4
CONFIG_NET_MAX_CONTEXTS=10

# Logging
CONFIG_NET_LOG=y
CONFIG_LOG=y
CONFIG_NET_STATISTICS=y
CONFIG_PRINTK=y
```

#### Steps to Reproduce

1. Use supplied `prj.conf` and `platformio.ini`
2. Verify zephyr is 3.7
3. Verify build fails

valeros, 2024-11-04:
Hi @ypearson-bdai, I just compiled your project without any issues:

[image]

Please clean your project and compile it in verbose mode, then attach the logs here. Thanks.

ypearson-bdai, 2024-11-04:
log attached 
[log.txt](https://github.com/user-attachments/files/17622605/log.txt)

This wasn't happening until I upgraded

ypearson-bdai, 2024-11-04:
I think I found it. My code (which came from a zephyr example) had a `exit(-1)` call. This is was fine from previous version. What do you think?

valeros, 2024-11-05:
> Any way to target zephyr 3.6 instead of 3.7?

You can downgrade the framework package via the [platform_packages](https://docs.platformio.org/en/latest/projectconf/sections/env/options/platform/platform_packages.html) option, for example:
```ini
[env:nucleo_f767zi]
platform = ststm32
board = nucleo_f767zi
framework = zephyr
...
platform_packages =
  framework-zephyr @ ~2.30600.0

```
> I think I found it. My code (which came from a zephyr example) had a exit(-1) call if an error occurred. This is was fine from previous version. What do you think?

I can look into it, but I need a simple example to reproduce the issue.

ypearson-bdai, 2024-11-05:
Ok, I'll provide that example soon.

I'll try what you mentioned and report back

ypearson-bdai, 2024-11-05:
Here is the sample project, I think its easy to reproduce now. Let me know what you think.
[ztest2.zip](https://github.com/user-attachments/files/17635259/ztest2.zip)

WilliamGFish, 2024-11-05:
I've similar issues with zenoh-pico and Zephyr. I am using Zephyr 4.0:

Changing the exit(-1) made no difference.

> [472/477] Linking CXX executable zephyr\zephyr_pre0.elf
> FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map C:/Users/BillyBob/zephyrGoodX/zephyrNew/sensor_zenoh/build/zephyr/zephyr_pre0.map
> cmd.exe /C "cd . && C:\gnu_arm_embedded_13\bin\arm-none-eabi-g++.exe  -gdwarf-4 -gdwarf-4 @CMakeFiles\zephyr_pre0.rsp -o zephyr\zephyr_pre0.elf -L"C:/gnu_arm_embedded_13/bin/../lib/gcc/arm-none-eabi/13.3.1/thumb/v7e-m/nofp" -lm -lc -lgcc -lc && cmd.exe /C "cd /D C:\Users\BillyBob\zephyrGoodX\zephyrNew\sensor_zenoh\build\zephyr && "C:\Program Files\CMake\bin\cmake.exe" -E true""
> C:/gnu_arm_embedded_13/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.bfd.exe: warning: zephyr\zephyr_pre0.elf has a LOAD segment with RWX permissions
> C:/gnu_arm_embedded_13/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.bfd.exe: modules/zenoh-pico/lib..__external__zenoh-pico__zephyr.a(system.c.obj): in function `_z_task_init':
> C:/Users/BillyBob/zephyrGoodX/external/zenoh-pico/src/system/zephyr/system.c:83:(.text._z_task_init+0x3c): undefined reference to `_z_report_system_error'
> C:/gnu_arm_embedded_13/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.bfd.exe: modules/zenoh-pico/lib..__external__zenoh-pico__zephyr.a(system.c.obj): in function `_z_task_join':
> C:/Users/BillyBob/zephyrGoodX/external/zenoh-pico/src/system/zephyr/system.c:86:(.text._z_task_join+0xc): undefined reference to `_z_report_system_error'
> C:/gnu_arm_embedded_13/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.bfd.exe: modules/zenoh-pico/lib..__external__zenoh-pico__zephyr.a(system.c.obj): in function `_z_mutex_init':
> C:/Users/BillyBob/zephyrGoodX/external/zenoh-pico/src/system/zephyr/system.c:95:(.text._z_mutex_init+0xa): undefined reference to `_z_report_system_error'
> C:/gnu_arm_embedded_13/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.bfd.exe: modules/zenoh-pico/lib..__external__zenoh-pico__zephyr.a(system.c.obj): in function `_z_mutex_drop':
> C:/Users/BillyBob/zephyrGoodX/external/zenoh-pico/src/system/zephyr/system.c:101:(.text._z_mutex_drop+0x10): undefined reference to `_z_report_system_error'
> C:/gnu_arm_embedded_13/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.bfd.exe: modules/zenoh-pico/lib..__external__zenoh-pico__zephyr.a(system.c.obj): in function `_z_mutex_lock':
> C:/Users/BillyBob/zephyrGoodX/external/zenoh-pico/src/system/zephyr/system.c:104:(.text._z_mutex_lock+0x8): undefined reference to `_z_report_system_error'
> C:/gnu_arm_embedded_13/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.bfd.exe: modules/zenoh-pico/lib..__external__zenoh-pico__zephyr.a(system.c.obj):C:/Users/BillyBob/zephyrGoodX/external/zenoh-pico/src/system/zephyr/system.c:106: more undefined references to `_z_report_system_error' follow
> collect2.exe: error: ld returned 1 exit status
> ninja: build stopped: subcommand failed.

WilliamGFish, 2024-11-06:
OK, found the issue, it relates to PR#586 ( https://github.com/eclipse-zenoh/zenoh-pico/pull/586) File: platform_common.h

They are calling a function within a Marco which causes a compilation problem:
```
void _z_report_system_error(int errcode);

#define _Z_CHECK_SYS_ERR(expr)             \
    do {                                   \
        int __res = expr;                  \
        if (__res != 0) {                  \
            _z_report_system_error(__res); \
            return _Z_ERR_SYSTEM_GENERIC;  \
        }                                  \
        return _Z_RES_OK;                  \
    } while (false)
```

Unfortunately, #define is a pre-processor directive used in C programs to define macros so the function has not been defined. Removing the function call and some logic amends it could be:

```
#define _Z_CHECK_SYS_ERR(expr)            \
    do {                                  \
        int __res = expr;                 \
        if (__res > 0) {                  \
            return _Z_ERR_SYSTEM_GENERIC; \
        }                                 \
        return __res;                     \
    } while (false)
```

I will raise with zenoh team.

Billy..

valeros, 2024-11-06:
@WilliamGFish Thanks for confirming!

I'm closing the issue as it seems that the culprit is a third-party dependency that has nothing to do with PlatformIO.

closed 2024-11-06 by valeros

WilliamGFish, 2024-11-06:
@valeros 
It is very much an issue with zenoh...a fix has been developed and had some initial testing.