platformio/platform-ststm32 #821: Add support for Blues Cygnet (STM32L433)

rdlauer, 2024-12-10:
This board was part of the 2.9.0 release of Arduino Core for STM32. I had thought it would automatically be part of the 18.0.0 STM32 platform release for PlatformIO, but apparently there is a manual process? Happy to create a PR if that is what is needed. Thanks!

maxgerhardt, 2024-12-10:
FYI, I have the a work in progress board definition, but it won't be uploading via USB (expects ST-Link by default). Also, PlatformIO doesn't yet have a package for the STM32CubeProgrammer, so if that's needed for the DFU upload, there is more base work to be done.

<details>
<summary>boards/blues_cygnet.json</summary>

```json
{
    "build": {
      "cpu": "cortex-m4",
      "extra_flags": "-DSTM32L4 -DSTM32L433xx",
      "f_cpu": "80000000L",
      "framework_extra_flags": {
        "arduino": "-DCUSTOM_PERIPHERAL_PINS -DARDUINO_CYGNET"
      },
      "mcu": "stm32l433cct6",
      "product_line": "STM32L433xx",
      "variant": "STM32L4xx/L433C(B-C)(T-U)_L443CC(T-U)"
    },
    "connectivity": [
      "can"
    ],
    "debug": {
      "default_tools": [
        "stlink"
      ],
      "jlink_device": "STM32L433CC",
      "openocd_target": "stm32l4x",
      "svd_path": "STM32L4x3.svd"
    },
    "frameworks": [
      "arduino",
      "cmsis",
      "stm32cube",
      "libopencm3"
    ],
    "name": "Cygnet",
    "upload": {
      "maximum_ram_size": 65536,
      "maximum_size": 262144,
      "protocol": "stlink",
      "protocols": [
        "jlink",
        "cmsis-dap",
        "stlink",
        "blackmagic",
        "mbed"
      ]
    },
    "url": "https://blues.com/feather-mcu/",
    "vendor": "Blues"
  }
```
</details>

zfields, 2024-12-10:
> The "Buy" link for the Cygnet doesn't even work...

It's a brand new board, and we were hoping to have it functional in PlatformIO before selling it, because PIO provides one of the best development experiences.

> Where's the board's documentation, schematic, etc.?

It is ready but not published while we await support. The main MCU is the `STM32L433CCT6`.

> Do you have the board on hand?

Yes, and we are happy to test anything you come up with.

> So is it using the "STM32CubeProgrammer (DFU)" upload method in the Arduino IDE... ?

Yes, that is correct.

> PlatformIO doesn't yet have a package for the STM32CubeProgrammer, so if that's needed for the DFU upload, there is more base work to be done.

It would be AMAZING if you guys could get it in there. However, this has been true for the "Swan R5" for over a year now. While it would be very nice to have, it doesn't appear to be mandatory.

maxgerhardt, 2024-12-10:
Is the USB-DFU upload protocol compatible with the `dfu-util` or does it need STM32CubeProg? What VID/PID does the board show up under when in DFU mode? How does it enter DFU mode? Via a 1200bps touch, or a DTR toggle?

zfields, 2024-12-10:
> Is the USB-DFU upload protocol compatible with the dfu-util or does it need STM32CubeProg?

I think it is compatible with dfu-util, but I don't know for sure. It should work similar to the way you've implemented the `Swan R5`.

> What VID/PID does the board show up under when in DFU mode?

```
Bus 001 Device 027: ID 0483:df11 STMicroelectronics STM Device in DFU Mode
```

> How does it enter DFU mode?

Manually, via BOOT and RESET buttons.

maxgerhardt, 2024-12-10:
When you download + open https://github.com/maxgerhardt/pio-cygnet-test in VSCode and use the "Verbose Upload" project task, does it upload?

rdlauer, 2024-12-10:
It does upload, yes:

```
Executing task: platformio run --verbose --target upload --environment blues_cygnet 

Processing blues_cygnet (platform: ststm32; board: blues_cygnet; framework: arduino)
------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/blues_cygnet.html
PLATFORM: ST STM32 (18.0.0) > Cygnet
HARDWARE: STM32L433CCT6 80MHz, 64KB RAM, 256KB Flash
DEBUG: Current (stlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 @ 4.20900.0 (2.9.0) 
 - framework-cmsis @ 2.50900.0 (5.9.0) 
 - tool-dfuutil @ 1.11.241029 
 - tool-openocd @ 3.1200.0 (12.0) 
 - tool-stm32duino @ 1.0.1 
 - toolchain-gccarmnoneeabi @ 1.120301.0 (12.3.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 14 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
MethodWrapper(["checkprogsize"], [".pio/build/blues_cygnet/firmware.elf"])
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   1.9% (used 1244 bytes from 65536 bytes)
Flash: [=         ]   6.8% (used 17764 bytes from 262144 bytes)
.pio/build/blues_cygnet/firmware.elf  :
section              size        addr
.isr_vector           396   134217728
.text               16756   134218124
.rodata               852   134234880
.ARM.extab              0   134235732
.ARM                    8   134235732
.preinit_array          0   134235740
.init_array            16   134235740
.fini_array             8   134235756
.data                 156   536870912
.bss                 1088   536871068
.noinit                 0   536872156
._user_heap_stack    1540   536872156
.ARM.attributes        48           0
.comment               56           0
Total               20924
<lambda>(["upload"], [".pio/build/blues_cygnet/firmware.bin"])
AVAILABLE: blackmagic, cmsis-dap, dfu, jlink, mbed, stlink
CURRENT: upload_protocol = dfu
"/Users/roblauer/.platformio/packages/tool-dfuutil/bin/dfu-util" -d 0x0483:0xDF11 -a 0 -s 0x08000000:leave -D "/Users/roblauer/Downloads/pio-cygnet-test-main/.pio/build/blues_cygnet/firmware.bin"
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Opening DFU capable USB device...
Device ID 0483:df11
Device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(10) = dfuERROR, status(10) = Device's firmware is corrupt. It cannot return to run-time (non-DFU) operations
Clearing status
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading element to address = 0x08000000, size = 18192

Erase           [                         ]   0%            0 bytes
Erase           [==                       ]  11%         2048 bytes
Erase           [=====                    ]  22%         4096 bytes
Erase           [========                 ]  33%         6144 bytes
Erase           [===========              ]  45%         8192 bytes
Erase           [==============           ]  56%        10240 bytes
Erase           [================         ]  67%        12288 bytes
Erase           [===================      ]  78%        14336 bytes
Erase           [======================   ]  90%        16384 bytes
Erase           [=========================] 100%        18192 bytes
Erase    done.

Download        [                         ]   0%            0 bytes
Download        [==                       ]  11%         2048 bytes
Download        [=====                    ]  22%         4096 bytes
Download        [========                 ]  33%         6144 bytes
Download        [===========              ]  45%         8192 bytes
Download        [==============           ]  56%        10240 bytes
Download        [================         ]  67%        12288 bytes
Download        [===================      ]  78%        14336 bytes
Download        [======================   ]  90%        16384 bytes
Download        [=========================] 100%        18192 bytes
Download done.
File downloaded successfully
Submitting leave request...
Transitioning to dfuMANIFEST state
======================== [SUCCESS] Took 6.49 seconds ========================
 *  Terminal will be reused by tasks, press any key to close it. 
```

rdlauer, 2024-12-10:
Looks good!

```
---- Opened the serial port /dev/tty.usbmodem2091348936301 ----
Blinky!
Blinky!
Blinky!
```

maxgerhardt, 2024-12-10:
See PR above.