platformio/platform-ststm32 #839: Can't upload to Arduino Portenta in DFU mode on Windows

unlimitedbacon, 2025-03-11:
Uploading during normal operation works fine, but upload fails in the bootloader (double click reset button).

```
Configuring upload protocol...
AVAILABLE: cmsis-dap, dfu, jlink, mbed, stlink
CURRENT: upload_protocol = dfu
Looking for upload port...
Auto-detected: COM8
*** [upload] Cannot configure port, something went wrong. Original message: PermissionError(13, 'A device attached to the system is not functioning.', None, 31)
```

This is happening with PlatformIO only. The upload works fine with the Arduino IDE.

valeros, 2025-03-11:
Hi @unlimitedbacon, does it work if you add `use_1200bps_touch=False` to your `platformio.ini`, for example:
```ini
[env:portenta_h7_m7]
platform = ststm32
framework = arduino
board = portenta_h7_m7
board_upload.use_1200bps_touch = False
```

unlimitedbacon, 2025-03-11:
Nope, I get the same thing

```
Configuring upload protocol...
AVAILABLE: cmsis-dap, dfu, jlink, mbed, stlink
CURRENT: upload_protocol = dfu
Looking for upload port...
Using manually specified: \\.\COM8
*** [upload] Cannot configure port, something went wrong. Original message: PermissionError(13, 'A device attached to the system is not functioning.', None, 31)
```

valeros, 2025-03-13:
Can you share here the logs of a successful upload from the Arduino IDE?

jz4396, 2025-08-13:
I am having the same issue still. Just wanted to bump this issue again. Here is a log of a successful upload from Arduino:
`Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Opening DFU capable USB device...
Device ID 2341:035b
Device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 011a
Device returned transfer size 4096
DfuSe interface name: "Internal Flash  2MB   "
Downloading element to address = 0x08040000, size = 156964

Erase   	[                         ]   0%            0 bytes
Erase   	[                         ]   0%            0 bytes
Erase   	[=                        ]   5%         8192 bytes
Erase   	[==                       ]  10%        16384 bytes
Erase   	[===                      ]  13%        20480 bytes
Erase   	[====                     ]  18%        28672 bytes
Erase   	[=====                    ]  20%        32768 bytes
Erase   	[======                   ]  26%        40960 bytes
Erase   	[=======                  ]  28%        45056 bytes
Erase   	[========                 ]  33%        53248 bytes
Erase   	[=========                ]  36%        57344 bytes
Erase   	[==========               ]  41%        65536 bytes
Erase   	[===========              ]  44%        69632 bytes
Erase   	[============             ]  49%        77824 bytes
Erase   	[=============            ]  52%        81920 bytes
Erase   	[==============           ]  57%        90112 bytes
Erase   	[===============          ]  60%        94208 bytes
Erase   	[================         ]  65%       102400 bytes
Erase   	[=================        ]  70%       110592 bytes
Erase   	[==================       ]  73%       114688 bytes
Erase   	[===================      ]  78%       122880 bytes
Erase   	[====================     ]  80%       126976 bytes
Erase   	[====================     ]  83%       131072 bytes
Erase   	[=====================    ]  86%       135168 bytes
Erase   	[======================   ]  88%       139264 bytes
Erase   	[=======================  ]  93%       147456 bytes
Erase   	[======================== ]  96%       151552 bytes
Erase   	[=========================] 100%       156964 bytes
Erase    done.

Download	[                         ]   0%            0 bytes
Download	[=                        ]   5%         8192 bytes
Download	[==                       ]  10%        16384 bytes
Download	[===                      ]  13%        20480 bytes
Download	[====                     ]  18%        28672 bytes
Download	[=====                    ]  20%        32768 bytes
Download	[======                   ]  26%        40960 bytes
Download	[=======                  ]  28%        45056 bytes
Download	[========                 ]  33%        53248 bytes
Download	[=========                ]  36%        57344 bytes
Download	[==========               ]  41%        65536 bytes
Download	[===========              ]  44%        69632 bytes
Download	[============             ]  49%        77824 bytes
Download	[=============            ]  52%        81920 bytes
Download	[==============           ]  57%        90112 bytes
Download	[===============          ]  60%        94208 bytes
Download	[================         ]  65%       102400 bytes
Download	[=================        ]  70%       110592 bytes
Download	[==================       ]  73%       114688 bytes
Download	[===================      ]  78%       122880 bytes
Download	[====================     ]  80%       126976 bytes
Download	[=====================    ]  86%       135168 bytes
Download	[======================   ]  88%       139264 bytes
Download	[=======================  ]  93%       147456 bytes
Download	[======================== ]  96%       151552 bytesWarning: Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release

Download	[=========================] 100%       156964 bytes
Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state`