platformio/platform-ststm32 #841: STM32 CMSIS project creation error

Rockwell1799, 2025-03-15:
Hello
I'm trying to add my first project for an STM32F401CCU6 using CMSIS on the platformio, but I'm getting this error:

`MissingPackageManifestError: Could not find one of 'package.json' manifest files in the package`

My platformio.ini file : 
```
[env:genericSTM32F401CC]
platform = https://github.com/platformio/platform-ststm32.git

board = genericSTM32F401CC

framework = cmsis

board_build.mcu = stm32f401ccu6

board_build.f_cpu = 84000000L

upload_protocol = stlink
```
Can someone please help me get this resolved?
TIA

valeros, 2025-03-20:
Hi @Rockwell1799, it seems you have a broken package, try deleting the directory with PlatformIO packages `%UserProfile%\.platformio\packages` or `~/.platformio/packages`.

Rockwell1799, 2025-03-27:
Thanks for your help.problem solved by removing the package folder.

closed 2025-03-28 by valeros