platformio/platform-ststm32 #843: Debugger broken in versions 18.0.0+

Zachary-Norris, 2025-03-28:
Hello, I noticed the debugger gets stuck in `Infinite_Loop` before even reaching the `main()` function in versions 18.0.0+ (my testing shows the debugger working in prior versions such as 17.6.0). The issue persists when tested with the default platformio project.

[image]
Pausing the debugger shows that the code gets stuck in this default handler `Infinite_Loop`.

[image]
Stepping through the code from the beginning I found this call to be the culprit:

[image]
As mentioned, this issue persists even with the above default project.

Here is the configuration of my platformio.ini file:
```
[env:stm32]
platform = ststm32@18.0.0
board = genericSTM32H750VB
framework = arduino
upload_protocol = stlink
debug_tool = stlink
```

I am using stlink version 3 with the latest firmware. I'm not sure if regularly uploaded code (non-debugger) is also broken or just when using the debugger as I haven't gotten that far.

valeros, 2025-03-31:
Hi @Zachary-Norris, thanks for reaching out. Maybe the size of RAM is set incorrectly for this board. Could you try adjusting it to `524288`? For example:
```ini
[env:stm32]
platform = ststm32@18.0.0
board = genericSTM32H750VB
framework = arduino
upload_protocol = stlink
debug_tool = stlink

; Fix  the RAM size
board_upload.maximum_ram_size = 524288
```
Then clean and debug your project again.

Zachary-Norris, 2025-03-31:
Hi @valeros - yeah, that worked.

For my clarity, why does RAM need to be set to 512MB? The datasheet calls out 1MB of RAM for this microcontroller. 

[image]

valeros, 2025-04-01:
The 1MB RAM in the STM32H750VB is not a single continuous block but is split into multiple memory regions. In our case we need a 512KB chunk of general-purpose AXI-SRAM. The rest of the RAM is divided among DTCM-RAM, ITCM-RAM, etc, each with different sizes and intended uses.

commit d831a63d (2025-04-01): Fix default RAM size for genericSTM32H750VB board

Resolves #843
closed 2025-04-01 by valeros