platformio/platform-ststm32 #849: Please support the Nucleo C071RB

DavidEGrayson, 2025-05-17:
I'd like to be able to program the [Nucleo C071RB](https://www.st.com/en/evaluation-tools/nucleo-c071rb.html) in PlatformIO.

The Nucleo C071RB is similar to the Nucleo G071RB, and it's already supported by the STM32Duino Arduino core, and it's even listed on your site here:

https://registry.platformio.org/tools/platformio/framework-arduinoststm32

However, I noticed that it's not in the [big list of boards](https://docs.platformio.org/en/latest/boards/index.html) in PlatformIO's documentation, and the UI in the VSCode extension doesn't find it.

Having support for generic STM32C071 boards would also be nice.

It would also be helpful if someone could tell me how I go about compiling for this board before you add support for it.  If I write "board = nucleo_c071rb" in my platformio.ini, it doesn't work, but maybe there's a way to grab some sort of board definition from `~/.platformio/packages/framework-arduinoststm32` instead?

DavidEGrayson, 2025-05-19:
Thank you for moving this issue to the appropriate repository.

I found PlatformIO's [Custom Embedded Boards](https://docs.platformio.org/en/latest/platforms/creating_board.html) documentation and found the [nucleo_g071rb.json](https://github.com/platformio/platform-ststm32/blob/develop/boards/nucleo_g071rb.json) in this repository.   Here is the board JSON I came up with and stored in `~/.platformio/boards/nucleo_c071rb.json`:

```json
{
  "build": {
    "core": "stm32",
    "cpu": "cortex-m0plus",
    "extra_flags": "-DSTM32C0 -DSTM32C0xx -DSTM32C071xx",
    "f_cpu": "48000000L",
    "framework_extra_flags": {
      "arduino": "-D__CORTEX_SC=0"
    },
    "mcu": "stm32c071rbt6",
    "product_line": "STM32C071xx",
    "variant": "STM32C0xx/C071R(8-B)T"
  },
  "debug": {
    "default_tools": [
      "stlink"
    ],
    "jlink_device": "STM32C071RB",
    "onboard_tools": [
      "stlink"
    ],
    "openocd_target": "stm32c0x",
    "svd_path": "STM32C071.svd"
  },
  "frameworks": [
    "arduino",
    "cmsis",
    "stm32cube"
  ],
  "name": "Nucleo C071RB",
  "upload": {
    "maximum_ram_size": 24576,
    "maximum_size": 131072,
    "protocol": "stlink",
    "protocols": [
      "stlink",
      "jlink",
      "cmsis-dap",
      "blackmagic",
      "mbed"
    ]
  },
  "url": "https://www.st.com/en/evaluation-tools/nucleo-c071rb.html",
  "vendor": "ST"
}
```

Then I used this `platform.ini` file in my project:

```text
[env:nucleo_c071rb]
platform = ststm32
board = nucleo_c071rb
framework = arduino
```

I was able to successfully build, but when I tried to upload to the board, I see that OpenOCD printed this error:

    embedded:startup.tcl:28: Error: Can't find target/stm32c0x.cfg

That file was [added](https://review.openocd.org/c/openocd/+/6874) in 2023, and the OpenOCD maintainers also [merged my patch](https://review.openocd.org/c/openocd/+/8525) in January 2025 to add support for the STM32C071xx.  OpenOCD hasn't actually tagged a new release since v0.12.0 in 2023.

Can we update PlatformIO to use the latest OpenOCD version from its master branch?

charlescao460, 2025-06-27:
Hi @valeros , can you reveal more details about the plan of STM32 MCU series? Will the support be eventually discontinued?

DavidEGrayson, 2025-06-30:
I think if PlatformIO just updates OpenOCD, that would go a long way towards allowing users to use the later STM32 chips, without needing official support.

DavidEGrayson, 2026-11-04:
@valeros ended up adding Nucleo C071RB support in version 20.0.0.  Thank you!

closed 2026-11-04 by DavidEGrayson