platformio/platform-ststm32 #855: Can't (Maple) DFU upload to STM32F103 anymore

maxgerhardt, 2025-06-13:
Commit https://github.com/platformio/platform-ststm32/commit/4b5ff01c8f91f1a2fd441c593aedfb81aa465f0f removed the usage of the `tool-stm32duino` package in favor of the `tool-stm32flash` package, but the builder script is still expecting the `maple_upload` command to exist.

https://github.com/platformio/platform-ststm32/blob/c8a228c871f5144d63dd78e61bcf548438cbce57/builder/main.py#L238-L248

So, when having a configuration like
```ini
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = arduino
upload_protocol = dfu
```

This results in just 
```
> pio run -t upload
Processing bluepill_f103c8 (platform: ststm32; board: bluepill_f103c8; framework: arduino)
...
Looking for upload port...
Auto-detected: /dev/ttyACM0
Uploading .pio/build/bluepill_f103c8/firmware.bin
sh: line 1: maple_upload: command not found
*** [upload] Error 127
```

valeros, 2025-06-17:
Thanks @maxgerhardt for pointing it out, should be fixed in the dev branch.

closed 2025-06-17 by valeros

FELG14, 2025-07-16:
This problem continues to occur. I already tried it with the new commit and it is still the same. Please check it.

FELG14, 2025-07-16:
```
C:\Users\Usuario\.platformio\penv\Scripts\pio.exe run -t upload -e bluepill_f103c8
[0m[0mProcessing bluepill_f103c8 (platform: https://github.com/platformio/platform-ststm32.git; board: bluepill_f103c8; framework
: arduino)
---------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/bluepill_f103c8.html
PLATFORM: ST STM32 (19.2.0+sha.0459d32) > BluePill F103C8
HARDWARE: STM32F103C8T6 72MHz, 20KB RAM, 64KB Flash
DEBUG: Current (stlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 @ 4.21001.250617 (2.10.1)
 - framework-cmsis @ 2.50900.0 (5.9.0)
 - tool-dfuutil @ 1.11.0
 - tool-dfuutil-arduino @ 1.11.0
 - tool-openocd @ 3.1200.0 (12.0)
 - tool-stm32duino @ 1.0.2
 - tool-stm32flash @ 0.7.0
 - toolchain-gccarmnoneeabi @ 1.120301.0 (12.3.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 14 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\bluepill_f103c8\FrameworkArduinoVariant\PeripheralPins.c.o
Compiling .pio\build\bluepill_f103c8\FrameworkArduinoVariant\PeripheralPins_MALYANM200_F103CB.c.o
Compiling .pio\build\bluepill_f103c8\FrameworkArduinoVariant\generic_clock.c.o
Compiling .pio\build\bluepill_f103c8\FrameworkArduinoVariant\startup_M200_f103xb.S.o
Compiling .pio\build\bluepill_f103c8\FrameworkArduinoVariant\variant_AFROFLIGHT_F103CB.cpp.o
Compiling .pio\build\bluepill_f103c8\FrameworkArduinoVariant\variant_MALYANM200_F103CB.cpp.o
Compiling .pio\build\bluepill_f103c8\FrameworkArduinoVariant\variant_MAPLEMINI_F103CB.cpp.o
Compiling .pio\build\bluepill_f103c8\FrameworkArduinoVariant\variant_PILL_F103Cx.cpp.o
Compiling .pio\build\bluepill_f103c8\FrameworkArduinoVariant\variant_generic.cpp.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_adc.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_adc_ex.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_can.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_ccb.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_cec.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_comp.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_comp_ex.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_cordic.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_cortex.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_crc.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_crc_ex.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_cryp.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_cryp_ex.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_dac.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_dac_ex.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_dcache.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_dcmi.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_dcmi_ex.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_dfsdm.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_dfsdm_ex.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_dma.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_dma2d.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_dma_ex.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_dsi.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_dts.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_eth.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_eth_ex.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_exti.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_fdcan.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_firewall.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_flash.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_flash_ex.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_flash_ramfunc.c.o
Compiling .pio\build\bluepill_f103c8\SrcWrapper\src\HAL\stm32yyxx_hal_fmac.c.o
[... 158 lines cut ...]
Compiling .pio\build\bluepill_f103c8\FrameworkArduino\wiring_analog.c.o
Compiling .pio\build\bluepill_f103c8\FrameworkArduino\wiring_digital.c.o
Compiling .pio\build\bluepill_f103c8\FrameworkArduino\wiring_pulse.cpp.o
Compiling .pio\build\bluepill_f103c8\FrameworkArduino\wiring_shift.c.o
Compiling .pio\build\bluepill_f103c8\FrameworkArduino\wiring_time.c.o
Archiving .pio\build\bluepill_f103c8\libFrameworkArduino.a
Linking .pio\build\bluepill_f103c8\firmware.elf
Checking size .pio\build\bluepill_f103c8\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   5.3% (used 1076 bytes from 20480 bytes)
[... cut ...]
Building .pio\build\bluepill_f103c8\firmware.bin
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, dfu, jlink, mbed, stlink
CURRENT: upload_protocol = dfu
Looking for upload port...
Auto-detected: COM1
Uploading .pio\build\bluepill_f103c8\firmware.bin
"maple_upload" no se reconoce como un comando interno o externo,
programa o archivo por lotes ejecutable.
*** [upload] Error 1
=============================================== [FAILED] Took 15.08 seconds ===============================================

``
```

valeros, 2025-07-17:
@FELG14 Thanks for reporting, please retest with the latest version from this repo.

FELG14, 2025-07-17:
> [@FELG14](https://github.com/FELG14) Thanks for reporting, please retest with the latest version from this repo.

Thank you very much, it works now without any problems.

commit 0459d323 (2025-06-17): Revert "stm32duino" package dependency

It's still required for DFU upload for stm32f103 targets

Resolves #855